MarlinFirmware/Marlin #6191: UBL & Mesh Editing requirement?

Tannoo, 2017-04-01:
If `UBL_G26_MESH_EDITING` is required to run `G29`, then why not a sanity check?

I loaded the latest RCBugFix tonight and had `AUTO_BED_LEVELING_UBL` enabled. `G29` simply does nothing but report back `ok`.
```
      #if PLANNER_LEVELING && !ENABLED(AUTO_BED_LEVELING_UBL) || ENABLED(AUTO_BED_LEVELING_UBL) && ENABLED(UBL_G26_MESH_EDITING)
        case 29: // G29 Detailed Z probe, probes the bed at 3 or more points,
                 // or provides access to the UBL System if enabled.
          gcode_G29();
          break;
```

Tannoo, 2017-04-01:
Something else is going on because I saw that I do have the mesh edit enabled.

closed 2017-04-01 by Tannoo

thinkyhead, 2017-04-02:
Yep! Somehow in my sleepless state I forgot to remove that condition from the UBL_G29 precursor. It's patched now.