MarlinFirmware/Marlin #6192: CoreXY Safe Homing/Probing Issue

firejackal, 2017-04-01:
So I'm having an issue with "Auto Home" (G28), and bed leveling.
I'm running the latest bugfix. This is a corexy printer I built, I have min and max endstops, everything reports fine and interacts fine.
When I use the LCD to control the corexy motions, everything is good.

But if I use G28 to home, it moves it to the min endstops correct. But after that when it's suppose to go to the safe homing position (middle of the bed) it does not, instead it tries to force itself into the minimum endstops. I tried commented out the max endstop code to make sure it wasn't that, still acted the same.

Also when doing the auto leveling function, it will start to move to do the probes, but it won't cover the entire defined area because it is getting jammed up and skipping steps, almost like it's trying to move the wrong directions for the corexy setup.

I have not touched the advanced config file.
Here is my current config file:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/887546/Configuration.txt)

ruggb, 2017-04-01:
?? why is X&Y_MIN_POS = 20 ?? That may be confusing it. But then I really do not understand why it is an option. Why would you need to limit movement to 20,20 instead of the real home 0,0?

ruggb, 2017-04-01:
Out of curiosity, what do max endstops buy you? I have a coreXY self-built also and I haven't figured that out yet.

Dreamloom, 2017-04-01:
I have max endstops on my custom corexy for Mr. Murphy's benefit.

firejackal, 2017-04-01:
ruggb:
> ? why is X&Y_MIN_POS = 20 ?? That may be confusing it. But then I really do not understand why it is an option. Why would you need to limit movement to 20,20 instead of the real home 0,0?

It was just a test, I was trying different values to see what happened, 0, -20, 20, etc, they all didn't seem to have a change.
On my anet a8 prusa clone printer, it used a negative value to indicate that the minimum endstops was beyond the bed size.

> Out of curiosity, what do max endstops buy you? I have a coreXY self-built also and I haven't figured that out yet. I also don't have bed leveling and maybe that is what it is for???

I originally installed the max endstops to check for maximum bed movement, but I then realized that marlin did not support this (like an auto size function), just something I need to take off since it doesn't seem needed.

firejackal, 2017-04-01:
also, according to the logs, it thinks it moves to the desired safe home position even through it does not.
READ:   current_position=(0.00, 0.00, 10.00) : sync_plan_position
READ:   destination=(210.00, 240.00, 10.00) : Z_SAFE_HOMING
READ: >>> do_blocking_move_to(210.00, 240.00, 10.00)
READ: <<< do_blocking_move_to
READ: >>> homeaxis(Z)
READ:   current_position=(210.00, 240.00, 10.00) : set_probe_deployed

thinkyhead, 2017-04-02:
Your configuration looks mostly good. But if you're not actually using the max endstops, comment them out in the block:
```cpp
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
```

ruggb, 2017-04-02:
@firejackal I have 2 extruders and use the right one as E0, so I have a negative value for E1 offset (-30,0). Otherwise, I don't think a neg value is valid for other user config values.
The fact that your bed size is smaller than your endstop size can be accounted for in 2 places. In Slic3r and Repetier-Host. If you are using something else, I am sure it is similar.
I have a total movement size (and endstops @ 0) of 305,335,295.
The actual build size is 305,305,295 - The E1 is accommodated by the extra 30 in the 335 movement.
I also have a hotbed in the center so I have configured that in both Repetier and Slic3r as separate configurations. I just select the ones I am working with. 0,0 is actually 50,70 with the hotbed config. and the config shows the hotbed overlaying the actual movement size.
Does that help?

Bob-the-Kuhn, 2017-04-13:
@firejackal - do you still have the G28 problem?

Did commenting out the xx_MAX endstops fix it?

I just ran into a user that claims the endstop checking software is screwed up.  At least on RCBugFix it looks like the expected endstops are being checked but so are a couple of the MAX endstops.

What version of Marlin are you running?

----

I suggest keeping your MAX endstops enabled.  Worst case is they do nothing.  They just might keep you from trying to run off the rails.

firejackal, 2017-04-14:
@Bob-the-Kuhn 
Yeah I still have the problem, I'm going to try redoing my frame and locations of the belts to make sure it isn't an alignment problem before a CoreXY problem with G28 homing.

I am running the latest bugfix (I flashed some earlier revisions to check them too and same reaction with my motors motor (or axis) moving the wrong direction at times.)
But on the latest bugfix, I did try disabling my max endstops and nothing changed.

With my prusa anet setup the "trying to run of the rails" didn't result in any major issues, but with the corexy setup, it will affect both axis (as it tries to move off to the left as in it's doing in my case, it will move the y axis too, since that's how corexy works.)
I think during printing with marlin there is a setting that tells the printer to e-stop if a endstop is hit, shouldn't happen, but since my corexy setup is kind of buggy at this moment so that will be helpful.

Bob-the-Kuhn, 2017-04-14:
About an hour ago fixes for CORExx endstop issues were merged into RCBugFixes.  Before that it would check endstops it shouldn't.  If you had endstops always enabled then it could have given you some strange results.  If Z was at the max then maybe a X movement wouldn't happen.

If you do a M119 and none of the endstops are triggered and you have strange movements then the movement issues are not due to endstops.

If it were an endstop issue when NOT doing a G28 then Marlin will report which endstop was hit.

Motors funning in the wrong direction SOMETIMES.  

Lets do a sanity check on your hardware.  Using this [diagram](http://corexy.com/theory.html), identify which is your A and B steppers.  Turn them by hand and verify that you get the right movement.  Then make sure the A stepper is connected to the X axis driver on your controller and the B stepper to the Y driver.  Command a pure X move and verify both steppers rotate properly.  if not then you can reverse via hardware or software.

If you turn them by hand and get the wrong movement then perhaps you actually have a COREYX.  The equations for the two flavors are:
```
COREXY    dX = (dA + dB)/2     dY = (dA - dB)/2     dA = dX + dY     dB = dX - dY
COREYX    dX = (dA - dB)/2     dY = (dA + dB)/2     dA = dX - dY     dB = dX + dY
```
where "d" stands for delta or difference.

firejackal, 2017-04-17:
Thanks for the reply @Bob-the-Kuhn, I'm going try the new bugfixes but at the moment I have been redoing my frame to make sure everything is correctly leveled and spaced.

But on the sanity check, using the corexy's diagram, the two A and B steppers move the correct directions by hand, and in software. I had also tried out the coreyx when I had started having these issues.

But after I get this frame squared away I'll do an update here if the problem persists.
And sorry for my late responses, my job has me working 12 hour shifts which takes away a lot of time to mess with this printer.

Bob-the-Kuhn, 2017-04-17:
12 hour days - I remember those.  NO THANKS

Take care of the important stuff and play with the printer when things calm down.

ruggb, 2017-04-17:
someone has to work time & 1/2 for time pay in order to get the US productivity figures up.

closed 2018-02-06 by thinkyhead