MarlinFirmware/Marlin #6193: Adding real jerk control, it will greatly improve print quality

kunok, 2017-04-01:
Why not adding real jerk (as the derivative of the acceleration, in mm/s^3) control to Marlin?, there is already code out there, in TinyG and G2 core have it implemented and (at least TinyG) is Open source
Look here, specially the graphics:
https://github.com/synthetos/g2/wiki/Jerk-Controlled-Motion-Explained
From this link:
-Better surface finish - speeds change gradually without discontinuities that result in temporary micro-positional errors
-Less machine vibration and resonance, resulting in less chatter and skip
-Less wear on mechanical parts, as machine "impacts" are greatly reduced
-"Steadier hand" for rapid operations such as pick and place, pipetting, etc.
-Motors run smoother and cooler

I've seen jerk control in machine tools and it really improves movements, and made everything smoother.
It will greatly improve the performance of marlin based 3d printers, will make that the steppers suffer less, and the printers will lose less steps.

Even a constant jerk will be a good improvement from the constant acceleration that we have currently.

TinyG original code were implemented in an 8 bit Xmega, so it should be doable.
I know is not that easy, but both TinyG and marlin is based in Grlb, so it should have some grade of compatibility, and I really think that it would be a very big improvement, even if it needs to be implemented from zero, it will be worth doing. Unfortunately I only know how to code in G-code, but I offer myself as a tester.

kunok, 2017-04-01:
More info
http://www.mmsonline.com/articles/understanding-jerk-control
http://research.engr.oregonstate.edu/mpcl/resources/Feed-optimization-for-%EF%AC%81ve-axis-CNC-machine-tools-with-drive-constraints.pdf

Sebastianv650, 2017-04-02:
It's also in my opinion that Marlin needs a better planner system (jerk based) some day, I have the feeling that the development of Marlin is mostly adding new features and not a development of the core functionality itself.
But there is also a reason for that: 3D printing needs a lot of extra features a CNC mil never needs (heated parts, special g codes, bed leveling only to name a few). This things also need some processing power, at least a little bit.

And that's the problem, Marlin runs on 16MHz ATMegas, where the systems you linked to runs on ARM or at minimum an XMega which can run at 32MHz as far as I found out. The current Marlin system has nearly no free power left, and I doubt we can get this S shaped ramp up/down thing "for free".

There is another problem with 3D printing that limits the possibility of speed-jump free movement: The XYZE speed is always continuous as long as the ratio between mm³ plastic layed down per mm XYZ movement is constant, but that's not always the case. As soon as for example the extrusion width changes (gap fill), this ratio changes from one movement block to the next. Therefore, the E speed jumps at the block boundaries. The only way to get a jerk free movement in this case would be to come to a full stop and accelerate again.

This leads to another question I thinking about from time to time: What we want to get from the acceleration limits? Limiting the acceleration per stepper, to prevent lost steps? Or to limit the absolute acceleration of the print head, to ensure a good print quality due to vibration and other effects. In most cases we will say the second one is true. But in this case, Marlin would have to know which axes are connected to a moving object like the print head.
For example say the max. acceleration is 1000mm/s². If we have a printer where the bed is only moved along Z and the nozzle is moved along X and Y, wo have to limit the X and Y acceleration to stay within the wanted limit together. This means max acceleration = 1000mm/s² <= SQRT(accelX^2+accelY^2).
If we have a printer with the bed moving in Z and Y, the nozzle only along X, X and Y axis could accelerate at 1000mm/s² on there own because the moving mass (nozzle, bed) is not linked to each other. Therefore the absolute XY acceleration can be as high as SQRT(1000^2+1000^2) = 1414mm/s² without loweing the print quality.
Therefore I think Marlin needs an option to specify which axis are linked together, changing the way how acceleration or jerk limits are applied..

thinkyhead, 2017-04-02:
> I have the feeling that the development of Marlin is mostly adding new features and not a development of the core functionality itself.

That is the phase we are in. However, the core is constantly being attended-to as new optimizations and improvements become evident. For example, mprovements to the stepper/planner —specifically a jerk update— have been incorporated from the Prusa MK2 branch. However, Marlin developers simply won't have time and space to integrate all the goodies from the latest grbl, TinyG, etc. until after the 1.1 release. That work will begin in the new 1.2 branch at that time.

kunok, 2017-04-02:
>3D printing needs a lot of extra features a CNC mil never needs (heated parts, special g codes, bed leveling only to name a few). This things also need some processing power, at least a little bit.

The joke is that jerk control isn't really needed in slow sturdy mill CNC, is needed in fast wobbling cnc, and 3d printers are one of the best example of this type of cnc. 
> There is another problem with 3D printing that limits the possibility of speed-jump free movement: The XYZE speed is always continuous as long as the ratio between mm³ plastic layed down per mm XYZ [...]

Well in that case don't use the jerk control in the extruder, the other axis will greatly benefit from this. Or just use a very high value.
In the reality the jerk isn't infinite anyway, you just can't physically do it, and the steppers and the step controllers are suffering from trying to do.

>That is the phase we are in. However, the core is constantly being attended-to as new optimizations and improvements become evident. For example, improvements to the stepper/planner —specifically a jerk update— have been incorporated from the Prusa MK2 branch. However, Marlin developers simply won't have time and space to integrate all the goodies from the latest grbl, TinyG, etc. until after the 1.1 release. That work will begin in the new 1.2 branch at that time.

I understand that we need a 1.1 release, but I really think that it will improve the print quality far better than for example autoleveling.
I may try to look for the motion code and try to do something myself. Any advices for a newbie?

kunok, 2017-05-14:
Now that we have a released 1.1 version can I respectfully suggest this to be looked again?

thinkyhead, 2018-02-06:
Perhaps the motion geeks will revisit this after 2.0 is released.

boelle, 2019-02-18:
was this not added in 2.0 late 2018?

boelle, 2019-03-12:
@thinkyhead i think we can close this one

if not much mistaken it was added 4/7-2018, if nothing else it has the wrong label

boelle, 2019-06-22:
@thinkyhead *bump*

CCS86, 2020-02-22:
I'm curious about this too, and how the new "Junction Deviation" system compares.