MarlinFirmware/Marlin #6194: (1.1.0 Bugfix) Delta Printer adjusts to Z-Offset after G29

NilsRoe, 2017-04-01:
I have migrated my ancient Kossel Mini Delta to the newest 1.1.0 RC8 Firmware yesterday. I was experiencing some issues with the autolevel and decided to try 1.1.0 Bugfix today (01. Apr 2017 ~12:00)

I am experiencing a strange issue with the autolevel - my allen key probe is configured such that it reaches around -6.9mm **below** the extruder. This has worked out more or less for the last two years. 

Now, after a G28/G29, the nozzle touches the bed when Z goes to 6.9mm (or the respectively set Z offset):

G28: Nozzle barely touches the bed at Z0.0 (paper test)
G28, then G29: Nozzle touches the bed at Z6.9

Setting the Z offset in the configuration.h and in the LCD menu caused the allen key to be dragged around the bed and ruining the tape layer. I have tried numerous settings with the probe wandering off or the nozzle slamming into the bed. Changing around the M206 parameter didn't help (after G29).

My Delta rod settings etc should be correct. The probe is deployed manually and then retracted automaically by pushing it on a screw head outside the printing area.

My config files are attached. If I made a trivial mistake, please feel free to point it out!

Nils

NilsRoe, 2017-04-01:
I have since observed another issue: Apparently the value for #define Y_PROBE_OFFSET_FROM_EXTRUDER is completely ignored (at least partially) when setting up the points for probing. The probe starts at the front, about 2cm away from the brim, and finally ends at the back with just a few mm of space left. It seems the nozzle is moved to the appropriate points, but not the probe.

wookie666, 2017-04-02:
Regarding your Probe offset problem, My bltouch is offset -20. And when it does the bed levelling it works fine. Make sure that your auto_bed_levelling options are taking into account the probe offset. Mine was something along the lines of set Delta_probably_radius Delta_printable_radius-40. So it would account for the offset. then my left right front and back probe points were -delta_probably_radius and delta probable radius respectively.

closed 2017-10-31 by thinkyhead