MarlinFirmware/Marlin #6195: Storing EEPROM values from Babystepping

matrix3dmachine, 2017-04-01:
Hi,
Using babystepping on the Z-axis in Marlin RC8BUGFIX March 31.

After I make the changes using z-axis babystepping during a print, I then the values using a "STORE EERPOM" from the LCD menu.  
The changes are not being saved to the EEPROM or not being retrieved for the next print.

Any idea how to fix?

Mike

NilsRoe, 2017-04-01:
AFAIK this is not a value that can be saved. I have used an older build before, but I have never been able to save/retrieve a babystep correction value.

NiklasOlofsson, 2017-04-02:
Use babystepping for Z until satisfied. Then transfer the changes to the Z Offset value and then store it.

matrix3dmachine, 2017-04-02:
Yes, It used to store that baby stepping value to EEPROM in Marlin versions previously, but that does not work anymore. 
I was still wanting a way to adjust z-offset using the LCD. I don't use probes, so the z offset option is removed from the LCD. 

Any way to get z offset in the LCD display without enabling it using probes? Using a standard proximity switch for z axis. 

Mike

NiklasOlofsson, 2017-04-03:
Ah. I'm to snowed in on having automatic bed leveling. ;-)

Bob-the-Kuhn, 2017-04-03:
On the Z offset, have you tried enabling FIX_MOUNTED_PROBE?

matrix3dmachine, 2017-04-03:
Actually yes, enabling FIX MOUNTED PROBE, seems to give me back z offset. 

Need to test it first to see if it works. 

Mike

matrix3dmachine, 2017-04-03:
What does that mean?
Snowed in???

NiklasOlofsson, 2017-04-03:
It is a common used phrase here in Sweden ("insnöad"). You are so in to something that you fail to see anything else. ;)

matrix3dmachine, 2017-04-03:
This worked perfect. Thx to all. Can some one tell me how to close a github issue?

CONSULitAS, 2017-04-11:
You are welcome.

Just hit "Close issue"

http://stackoverflow.com/a/34010719/7792450

closed 2017-04-11 by Roxy-3D

thinkyhead, 2017-04-13:
> Yes, It used to store that baby stepping value to EEPROM in Marlin versions previously

Only very recently has this been submitted to the Marlin main branch. and here it is: #6245

SuperCables, 2020-08-18:
in Configuration_adv
uncomment
#define BABYSTEP_ZPROBE_OFFSET
This will combine baby-stepping and probe offset.

doug-scott, 2020-08-31:
> in Configuration_adv
> uncomment
> #define BABYSTEP_ZPROBE_OFFSET
> This will combine baby-stepping and probe offset.
> (I know, this was closed 3 years ago, but for everyone not using a fixed mount (ex bl touch) this is my solution)

Does this mean that if I comment #define BABYSTEP_ZPROBE_OFFSET out (//#define BABYSTEP_ZPROBE_OFFSET for ex) that any babystep changes I make will only apply to the currently printing job, and will not update/change my m851 setting?  Sorry if this seems obvious, I have read too much on this topic and am at the terminally confused part now.  What I want is to be able to adjust z-offset while printing the skirt, and have the print continue with that z-offset to the end, and then forget it.  Next print job will/may require me to adjust it again.

SuperCables, 2020-08-31:
BABYSTEP_ZPROBE_OFFSET means that baby stepping will be permanent so long as you save eeprom before power off. This is because baby stepping normally doesn't save, but BABYSTEP_ZPROBE_OFFSET makes baby stepping set the permanent z probe offset instead.