MarlinFirmware/Marlin #6198: Printer Comes To Screeching Halt When Finishing Prints?

tjay31584, 2017-04-01:
Hi all,

I am new to all this, but I have installed Marlin RC8 on a Velleman K8200 running a Mega 2560, Ramps 1.4 board, and RepRap Discount Full Graphics Controller.  For the most part, I have gotten everything working I need.  I spent time in the configuration files and got all the motors working properly, end stops, LCD encoder working properly, PID tuned the heater for the E3D V6, etc.

The printer currently prints via USB 99% of the time using Repetier 1.6.2, but I have had a problem occur probably 80% of the time I print where the printer will come to a screeching halt when the print job finishes.  SOMTIMES, it will complete and go where it's supposed to, but not consistently.  It literally stops with the nozzle still on the part and sometimes sounds like the motors are being cut off unexpectedly because sometimes it's almost as if they are trying to rapid home but skip steps or something?  I almost thought it was a power supply problem (like not being able to support all the rapids at once causing a brownout), so I hooked my DMM up to the power supply monitoring min/max readings at the PS, but nothing abnormal is observed when it happens.

I get the message about printer kill being called and to reset it on the LCD, but in Repetier, I see no errors other than it says the printer reset was detected, then the boot up sequence is displayed in the log for the printer after I reset it.  Once I reset the printer, I can home it, call the park position, etc with no issues.  The printer will complete prints that take 10 minutes or 4 hours with no issues DURING the prints; it's only at the very end after it completes.  I have no idea what it is I need to even look at to figure this out, so any help would be appreciated!

tjay31584, 2017-04-01:
I just finished a print and here's an example of what I get in the log:

>16:36:45.688 : N13475 G91*21
16:36:45.688 : N13476 G1 E-1*86
16:36:45.688 : N13477 G92 E0*65
16:36:45.766 : N13478 G90*25
16:36:45.766 : N13479 M107*45
16:36:45.782 : Printjob finished at 4/1/2017 4:36 PM
16:36:45.782 : Printing Time: 28m:0s
16:36:45.782 : Lines Send: 12363
16:36:45.788 : N13480 M84*17
16:36:45.788 : N13481 G1 X0 Y200 F4800*78
16:36:45.788 : N13482 M84*19
16:36:45.860 : X:92.45 Y:79.21 Z:31.50 E:0.00 Count X: 5987 Y:5217 Z:80939
16:36:46.712 : echo:enqueueing "M84 X Y Z E"
16:36:46.729 : Error:Printer halted. kill() called!

And attached is a photo of the LCD screen.
[image]

tjay31584, 2017-04-02:
Here's another, only this time I got garbage on the LCD also.  I've never seen this before, so I thought I'd post it too.

>17:39:05.926 : N5590 M107*28
17:39:05.926 : N5591 G1 E587.1964 F2400*11
17:39:06.070 : N5592 G0 X91.39 Y109.05 Z25 F4560*48
17:39:06.086 : N5593 G91*43
17:39:06.086 : N5594 G1 E-1*108
17:39:06.086 : N5595 G92 E0*123
17:39:06.213 : N5596 G90*47
17:39:06.213 : N5597 M107*27
17:39:06.229 : Printjob finished at 4/1/2017 5:39 PM
17:39:06.229 : Printing Time: 16m:41s
17:39:06.229 : Lines Send: 4964
17:39:06.234 : N5598 M84*46
17:39:06.234 : N5599 G1 X0 Y200 F4800*113
17:39:06.234 : N5600 M84*44
17:39:06.344 : X:91.39 Y:109.05 Z:25.00 E:0.00 Count X: 6047 Y:7006 Z:64175
17:39:07.046 : Error:Printer halted. kill() called!

Here's the LCD:
[image]

thinkyhead, 2017-04-02:
RC8 is getting a little long-in-the-tooth. Please test the `RCBugFix` branch and see if the problem still exists.

tjay31584, 2017-04-02:
Thanks for the reply.  I just installed it to see what happens.  I noticed the LCD menu is SO much faster, and the printer starts basically immediately after I hit print, no more hanging like it's thinking.  I assume those are both changes?  It seemed like before it had to wait to start printing until after it had transferred enough of the program or something.

I will let you know what happens after this calibration cube finishes in the next 15 minutes or so!

tjay31584, 2017-04-02:
Ok, finished two cubes of various times, and they both completed normally with the printer moving to the dedicated park position with no other issues that are apparent.  It seems this is just an issue with my setup and RC8 possibly, since there was no way it would have done this before twice in a row.

Thanks to everyone for all the hard work on this, I'm having a lot of fun with this and learning a lot.  You guys rock!

thinkyhead, 2017-04-02:
I'm glad it's working for you. I suggest that you re-download `RCBugFix` periodically, because we're in the final phase of clearing up bugs and are freezing all but the most basic features to get ready for the 1.1.0 release. Just today I patched a few bugs, and I'm sure we'll find a few more before the release.

thinkyhead, 2017-04-02:
> I'm having a lot of fun with this and learning a lot.

Yeah, this technology rocks.

closed 2017-04-02 by thinkyhead