MarlinFirmware/Marlin #6200: Can we make the delay time for BLTouch user configurable?

mperdue, 2017-04-02:
In marlin_main.cpp if BLTouch is defined you do 

safe_delay(375);

Can this be set up with a define in configuration.h so it is easily changed by the user? I keep having to search for that when I do an update because my brain resets when I sleep.

Tannoo, 2017-04-02:
In `configuration.h`:
```
// The BLTouch probe emulates a servo probe.
// The default connector is SERVO 0. Set Z_ENDSTOP_SERVO_NR below to override.
//#define BLTOUCH
//#define BLTOUCH_DELAY 375
```
and in `marlin_main.cpp`:
```
  #if ENABLED(BLTOUCH)
    void bltouch_command(int angle) {
      servo[Z_ENDSTOP_SERVO_NR].move(angle);  // Give the BL-Touch the command and wait
      safe_delay(BLTOUCH_DELAY);
    }
```

Will this help?

mperdue, 2017-04-02:
Yeah, that's about what i've been doing but, of course, it keeps going away when a new release comes along.

Tannoo, 2017-04-02:
Yes, unless a PR is put in with those changes and it get's merged, you will have to continue to do it if you want that convenience.

Tannoo, 2017-04-02:
I have a couple of PR's out that have not been merged yet.

I have several other "mods" that I have not made PR's for.

I do have to "merge" those with every update to the main code, if I want them active. I also try to keep them up-to-date to see if there are any issues with other newer additions.

For me, it seems that I am being put to an endurance test. lol
Maybe that I'll give up and go away.

Bob-the-Kuhn, 2017-04-02:
I was going to suggest using  **SERVO_DELAY** but the current BLTouch setup code always changes it to 50mS.

Maybe the solution is to change the init code so uses 50mS or SERVO_DELAY, whichever is bigger.

Roxy-3D, 2017-04-02:
I like the idea suggested by Tannoo.   Perhaps it could default to 50ms but if BLTOUCH_DELAY is specified, that gets used instead.   It will be easiest for users if anything affecting the operation of the BL-Touch is  kept together.

mperdue, 2017-04-02:
Defaulting to 50 would be great for me since that's what I keep changing it to. :-)

Roxy-3D, 2017-04-02:
Oh!   I mis-understood.   When I saw that 50ms number I thought "That seems awful fast.   But I guess it works..."     If we add the control for BL-Touch probes delay times, I would have mine up at 250.

thinkyhead, 2017-04-02:
Here you go: #6214

closed 2017-04-02 by thinkyhead