MarlinFirmware/Marlin #6203: Issue with CoreYX

corydlent, 2017-04-02:
I am still new to using Marlin so there might be something that I am doing wrong with configuring my system.

Marlin version: 1.1.0

My hardware configuration is minimal as I am still assembling the system, two stepper motors and stepper drivers is all that is on the RAMPS 1.4 board.

Sending code using Repetier or manually using PuTTY:
M17
G0 X100 Y100 F2000
G0 X0 Y0 F2000

The above move fails on the third line, will not move back to origin.
It will move back when move is done in two steps, but not together.
Failure is repeatable so I do not think it is a hardware issue.
Error reported is endstop hit at whatever X position is used in line 2.

Configuration file attached.
Can anybody help me diagnose this?

Thank you,

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/888194/Configuration.txt)

Bob-the-Kuhn, 2017-04-02:
The M17 command isn't needed.  The steppers are automatically powered up by the G0 command.

> will move back when move is done in two steps

I'm not sure what that means.  Please give the command sequence that allows the head to return to the home position and the command sequence that fails.

Maybe you have an endstop problem.  If you're only moving in the +X direction (or the +Y direction), it usually means your endstop logic is inverted.

Issue M119 before and after some of the working and failing moves.  I expect that you'll see either X or Y or both always say TRIGGERED except at the home position.  If that's the case then change the X_MIN_ENDSTOP_INVERTING and/or Y_MIN_ENDSTOP_INVERTING setting(s).

I see you have software endstops enabled.  If you're **NOT** using hardware endstops then these could get in the way of movements.  When your system powers up or is reset it will assume that it's at location X0, Y0 (per your X_MIN_POS and Y_MIN_POS settings).  **#define min_software_endstops true** means that you won't be able to move in the - direction past 0,0.

thinkyhead, 2017-04-02:
Is this after a `G28`?

corydlent, 2017-04-03:
Thank you Bob-the-Kuhn and thinkyhead or your replies.

Bob-the-Kuhn:

Thank you for the info on G0, I did not know this.
Sorry for the confusion on: "Move back when move is done in two steps".
The original sequence does not allow reverse movement but this sequence does:

M17
G0 X100 Y100 F2000
G0 X0 Y100 F2000 ;Move X direction
G0 X0 Y0 F2000 ;Move Y direction

No error thrown with end stops for this move. It seems to be related to moves that involve both directions.

Issuing M119 before and after a failed move both show TRIGGERED.
It does the same thing if not moving back to the home position, for instance, this would fail:

M17
G0 X200 Y200 F2000
G0 X100 Y100 F2000

I changed the  #define min_software_endstops to false and it still does the same thing.

corydlent, 2017-04-03:
thinkyhead:

No G28 is issued, I have no endstop switches attached at the moment.
I just want to test out the system so nothing is hooked up.

Is a G28 required in order to operate the system normally?
In other words, is homing to hardstop not permitted?
That is essentially what I am doing at the moment.

Thank you both for your input it is greatly appreciated.

corydlent, 2017-04-03:
Another piece of information that might be helpful.

I changed the settings for [XY]_HOME_DIR to 1 (MAX) and the behavior inverts.
Forward motion is not permitted, and it says endstops are reached when attempting the same procedure.
Reverse direction is possible, i.e. G0 X-100 Y-100 F2000

Software limits still disabled.

Thanks.

Bob-the-Kuhn, 2017-04-03:
G28 just tells the machine where it is at when you have hardware endstops.

The alternative is to manually home it by moving the head to the X=0, y=0 z=0 position and issue a G92 command.

It won't take long before you realize it's less work to install the endstops than it is to manually home the machine.

----

I'd also like to see your Configuration.h and Configuration_adv.h files.  Just zip them up and point me to them.  Or zip them up and add .TXT to the end of the file name and drop it on your reply.

Please do the following and include the log in the ZIP file.  This assumes you're using the USB port.

M119
M114
G0 X100 Y100 F2000
M119
M114
G0 X0 Y100 F2000 ;Move X direction
M119
M114
G0 X0 Y0 F2000 ;Move Y direction
M119
M114

and

M119
M114
G0 X200 Y200 F2000
M119
M114
G0 X100 Y100 F2000
M119
M114

corydlent, 2017-04-04:
Bob-the-Kuhn:

I greatly appreciate all of the help you have given me.
I believe I have resolved my issue.

I wanted to bypass / skip the homing sequence all together for now until I get the switches wired up.

I ended up inverting the endstop states for all motors and this worked for me.

For anybody who is curious this is the final configuration file.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/892099/Configuration.txt)

closed 2017-04-04 by corydlent