MarlinFirmware/Marlin #6204: ABL_BILINEAR_SUBDIVISION

apballard, 2017-04-02:
Hi All

I thought I'll try out ABL_BILINEAR_SUBDIVISION.

So I enabled it.

I perform a G28 then a G29, but once the G29 is complete the printer no longer responds to G1/G0 ZXXX

Meaning it no does a physical move of the Z-AXis after the G29.  Everything works fine before that.

Furthermore, the Z value does update.  G1 Z10 will then display Z=10 without a physical move.

Am I doing something wrong? Maybe I don't understand the requirements properly?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/888198/Configuration.zip)

thinkyhead, 2017-04-02:
When you do the build, how much Program and Data memory usage does it report?

apballard, 2017-04-02:
Thanks @thinkyhead 

I was running a RCBugFix a wee or so old.  But now it seems to be working.  Just downloaded the latest version.

My compile was returning:
Program size: 94 376 bytes (used 37% of a 253 952 byte maximum) (59,51 secs)
Minimum Memory Usage: 6858 bytes (84% of a 8192 byte maximum)

M100 was returning 
Found 1019 bytes free at 0x1CF2

But for now, going to run some tests.

tximy, 2017-04-02:
I am testing the last week RCBugFix and at uncomment the #ABL_BILINEAR_SUBDIVISION over G29 the Z axis did not respond too.

thinkyhead, 2017-04-03:
Let's do some logging, and if needed I can add some more logging to isolate the cause.

BillyBob54, 2017-04-03:
If it helps- I am experiencing the same problem (RCBugFix downloaded 25May).  I have G29 in my start up gcode, homing works OK, Bi Linear Bed Leveling routine works OK, but when the print starts, head stays at ~3mm (position at finish of leveling).  I found that I can "drive” the nozzle to the bed with Baby stepping, but the Z axis does not move up while printing.

apballard, 2017-04-03:
Ok, my issues with M851 and G29 not working seems to be a separate problem.  Created issue #6233 for it.

Still having issues with ABL subdivide.  But will look into it when i can solve M851 issue with G29.

pflannery, 2017-04-04:
@apballard 
When you change the z-offset after a G29 Marlin will auto update the mesh data by the z-offset. 
How much do you change the z offset after you call to G29?

apballard, 2017-04-04:
Ok, so i've sorted out the cause for the offsets.

Back to the original problem for this issue, after a G29 with Subdivide enabled, Z does not move, however its value does change.

X and Y movements still work fine after a G29.

I'm not sure if i'm doing it right but M100 F says over 800 bytes free.

Any suggestions?

apballard, 2017-04-08:
@pflannery I just issue a single m851 correcting no more than 0.06

apballard, 2017-04-11:
Can't seem to get workin. Decided no need, normal ABL is working brilliantly

closed 2017-04-11 by apballard

thinkyhead, 2017-04-11:
> M100 F says over 800 bytes free.

Probably not enough free memory. The stack is likely to crash into the heap at some point.

Roxy-3D, 2017-04-11:
>> M100 F says over 800 bytes free.

> Probably not enough free memory. The stack is likely to crash into the heap at some point. But, we know this about bilinear subdivision — it is a memory hog.

I think I disagree.   If the Bi-Linear code is active and set up...  And has handled a move...  That 800 bytes free really is free.   Marlin doesn't have any memory leaks.   If you invoke some other GCode that is a big memory pig, it is possible to use a little bit more stack memory from the bottom side of the memory pool.

But even so...  It is only during that one GCode that you had less than 800 bytes of free memory.

thinkyhead, 2017-04-13:
Depending on active features, it seems to tempt fate if you go beyond the point where the compiler warns about "instability" due to low RAM. The stack is sometimes "abused" to allocate buffers, and there are a lot of concurrent functions often doing things with strings, and so on.

Just as with CPU usage, we should find the biggest culprits and see if we can design them better. And maybe there are things we can do organizationally and systematically to improve overall efficiency.

We'll just have to science the shit out of this thing.