MarlinFirmware/Marlin #6206: BLtouch wont deploy unless Doubleprobe is turned on

wookie666, 2017-04-02:
As the title says. My bltouch will not deploy unless i have Double probe turned on.
DELTA_PRINTABLE_RADIUS

With my probe being x-40 and Y-20 from the nozzle. will this account for the offset correctly. as i did a test before and lifted one side of the bed up 2mm, and it seemed as though the print was not adjusting correction on the first layer.
I have set my offsets correctly in marlin.

my version is 

FIRMWARE_NAME:Marlin 1.1.0-RC8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Anycubic plus EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

Bob-the-Kuhn, 2017-04-03:
Please see if [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix ) solves this problem.  There's been a ton of corrections since RC8.

wookie666, 2017-04-03:
Do i use RCBugFix or do i use 1.1.0-RC8 "Woozy Wookiee" which is the latest?

wookie666, 2017-04-03:
> Please see if RCBugFix solves this problem. There's been a ton of corrections since RC8.

I have noticed these are the changes

Roxy-3D, 2017-04-03:
> I have noticed these are the changes since Dec 6. I know my firmware is later than that, so are you sure that RCBUGFIX is later than my Marlin 1.1.0-RC8

Are you looking here???  https://github.com/MarlinFirmware/Marlin/tree/RCBugFix   Because I merged in a spelling fix 5 hours ago and that is not in your list.

wookie666, 2017-04-03:
> RCBugFix Because I merged in a spelling fix 5 hours ago and that is not

​Ill give it a shot then. Thanks mate

I just wasn't sure on how to check that i had the absolute latest
​

wookie666, 2017-04-03:
just tested the latest bugfix and the bltouch probe wont deploy with probe_double_touch turned off

wookie666, 2017-04-03:
Ok. after turning on double touch, this happens. There is a noticeable delay before dropping the probe. 

Didn't happen on the last firmware.

Also, i have disabled the bltouch delay, and i have servo delay set to 300 which is standard.

Any ideas 
https://youtu.be/A9HXHkkHKtk

psavva, 2017-04-03:
@wookie666 , the delay was introduced in this PR 
https://github.com/MarlinFirmware/Marlin/pull/6114 by @Bob-the-Kuhn 
And the delay is configurable via this PR
https://github.com/MarlinFirmware/Marlin/pull/6214

wookie666, 2017-04-03:
Yes I understand. I recompiled the firmware and commented out the bltouch servo delay and also commented out the servo delay that is down the bottom of configuration.h and my probe wouldn't deploy. So I put the standard servo delay back on but left bltouch servo delay commented out and it still had that delay. So which is it?

psavva, 2017-04-03:
You should comment out servo delay, but set BLTouch delay with a logical value.
Try 375 milliseconds as it is by default, and if your able to configure it lower, all good.

Btw, if you comment out the BLTouch Delay, it will dynamically be defined to the default 375ms value.

Do you see that the delay is too long now?
It is required on some BLTouch modules.

wookie666, 2017-04-03:
I'll test it in about 10hours and see. Thanks

closed 2017-10-31 by thinkyhead