MarlinFirmware/Marlin #6207: Is code_seen() able to accept decimal (float) numbers?

kcheeeung, 2017-04-02:
Hi guys. I wanted to make sure. Is code_seen() able to accept decimal (float) numbers?

Ideally, I'd want to M433 P2.4 R1 so it'd calculate the duty cycle need to drive the pressure regulator I have.

```
inline void gcode_M433(){
  if(!code_seen('P')) return; //You must pick a PSI
  float airPSI = code_value_int();
  if(airPSI < 0 || airPSI > 30) return; //PREVENT ACCIDENTAL PSI OVERLOAD TO VALVES

  if(!code_seen('R')) return; //You must pick a REG_#_PIN to set
  uint16_t PSIregulator = code_value_int();

  float REQ_VOLTAGE = (airPSI / PSI_TO_SIGNAL_RATIO);
  float REV_DUTYCYCLE_PSI = 255 - (255 * REQ_VOLTAGE/INPUT_HEATER_VOLTAGE);
  round(REV_DUTYCYCLE_PSI);

  if (PSIregulator == 1){
    pinMode(REG_1_PIN, OUTPUT);
    WRITE(REG_1_PIN, REV_DUTYCYCLE_PSI);
  }
  if (PSIregulator == 2){
    pinMode(REG_2_PIN, OUTPUT);
    WRITE(REG_2_PIN, REV_DUTYCYCLE_PSI);
  }
  if (PSIregulator == 3){
    pinMode(REG_3_PIN, OUTPUT);
    WRITE(REG_3_PIN, REV_DUTYCYCLE_PSI);
  }
}

```

thinkyhead, 2017-04-02:
`code_seen` only returns whether a GCode command has a particular parameter. After the call to `code_seen('X')` you then use `code_value_float()` to retrieve the value of the parameter.

kcheeeung, 2017-04-02:
Thanks for the help!

closed 2017-04-02 by kcheeeung