MarlinFirmware/Marlin #6211: Q: Why was LCD_BED_LEVELING moved outside of the MESH defines

WheresWaldo, 2017-04-02:
In previous versions of RCBF I could leave LCD_BED_LEVELING uncommented and switch between auto bilinear and MESH with just a change to two lines of code in Configuration.h. Because LCD_BED_LEVELING used to be within ENABLED(MESH_BED_LEVELING) it would not compile into the code and hence could be left enabled 100% of the time. Now I have to specifically comment out that line when switching to ABL or UBL. Seems unnecessary. Is there a reason I am not seeing for this change in behavior?

thinkyhead, 2017-04-02:
It is now an option for both `MESH_BED_LEVELING` and all forms of auto bed leveling when `PROBE_MANUALLY` is the selected probe type.

thinkyhead, 2017-04-02:
It is now an option for both `MESH_BED_LEVELING` and all forms of auto bed leveling when `PROBE_MANUALLY` is the selected probe type. So, yeah, we could just ignore it in other cases.

Meanwhile, comment out this line and maybe it will compile. (Maybe not.)
```cpp
#error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY."
```

closed 2017-04-02 by thinkyhead