MarlinFirmware/Marlin #6225: Delta levelling question

wookie666, 2017-04-03:
I am running rc8 on a delta and i have some questions.

I have a bltouch offset x-40 and y-20. with this offset, my levelling on my delta is printableradius - 60. 
This is producing a square that is 120 x 120. What happens with the remaining 60mm of the print radius when it comes to the bed mesh.

Also, i notice i can only use bilinear with a delta. How can i make sure that it is working as it should, and that it is taking into account the probe offsets correctly. I just do a regular G28 and then a regular g29. Is there anything else i should be doing for the bed levelling to work properly.

Can i do a G29. Then move the nozzle to each point on the bed and go to G1, will that take into account the bed level offset, or does it only work when printing?

wookie666, 2017-04-03:
Instead of opening another thread, does UBL work for deltas yet?

wookie666, 2017-04-03:
im still getting compile errors using the latest rcbugfix

psavva, 2017-04-03:
It's not yet ready for Deltas as far as i know.
@Roxy3d, can you confirm

Bob-the-Kuhn, 2017-04-03:
Correct - UBL isn't ready for Deltas yet.

Here's a write up on bilinear leveling that may answer some of your questions.
[Bilinear Auto Bed Leveling Details 0.4.zip](https://github.com/MarlinFirmware/Marlin/files/890364/Bilinear.Auto.Bed.Leveling.Details.0.4.zip)

Bob-the-Kuhn, 2017-04-03:
@wookie666 - I'll work on the compile issues if you zip up the two config files and drop them on your reply.

wookie666, 2017-04-03:
Sorry bon. The compile errors weren't actually errors but noted that ubl is not active for delta

psavva, 2017-04-03:
I think you can close this issue since deltas as ubl is not yet supported?

closed 2017-04-03 by wookie666