MarlinFirmware/Marlin #6226: FIlament change missing in RC8

PhilMaddox, 2017-04-03:
apologies if this has been covered previously but i cant seem to find the "filament change" option in the LCD menu and issuing a M600 doesn't do anything either. 
Ramps 1.4 running RC8 on cartesian ( Prusa Air2) 
the bilinear bed leveling has done wonders to my large object prints but missing the filament change feature.
While on the subject it would be great if the development could include a Gcode option to resume after the M600 command rather than ( or alongside) the LCD "click" button 
Thanks

Tannoo, 2017-04-03:
You have to enable it in configuration_adv.h#682.

PhilMaddox, 2017-04-03:
Thank you Tannoo . Much appreciatted

psavva, 2017-04-03:
@PhilMaddox, if all is ok, please close this ticket.

closed 2017-04-03 by PhilMaddox