MarlinFirmware/Marlin #6233: G29 and Offset

apballard, 2017-04-03:
Hi All

I thought I was going mad.. 

Maybe you can direct me in the right direction.

Yesterday downloaded the latest RCBugFix and now my G29 is offset by 2.  Any idea why?

Recv: Bilinear Leveling Grid:
Recv:      0     1     2     3     4
Recv:  0 -2.30 -2.27 -2.19 -2.30 -2.46
Recv:  1 -2.33 -2.21 -2.14 -2.28 -2.41
Recv:  2 -2.36 -2.22 -2.17 -2.28 -2.42
Recv:  3 -2.35 -2.40 -2.31 -2.40 -2.49
Recv:  4 -2.70 -2.68 -2.69 -2.78 -2.78

This is what it used to look like before:

Recv:      0     1     2     3     4
Recv:  0 -0.38 -0.30 -0.20 -0.29 -0.35
Recv:  1 -0.20 -0.29 -0.08 -0.21 -0.34
Recv:  2 -0.44 -0.34 -0.22 -0.21 -0.35
Recv:  3 -0.54 -0.59 -0.33 -0.52 -0.64
Recv:  4 -0.91 -0.77 -0.65 -0.73 -0.71

apballard, 2017-04-03:
so far I reverted the firmware to a previous version of RCBugFix and its working, so i'm trying to compare to see the difference.

apballard, 2017-04-04:
So it does not seem to be a configured issue, but the number of lines of code that have changed between the versions, not sure what I'm looking for.

Roxy-3D, 2017-04-04:
> not sure what I'm looking for.

You are looking for things that change z_offset, or current_position[Z_AXIS], or destination[Z_AXIS].    Something like that...    Something is setting the Z position in the wrong location.

apballard, 2017-04-04:
OK, thanks @Roxy-3D 

I think its related to #6110 .

Still trying to work out what the logic was intended for.  I'll have to run some tests later.

Roxy-3D, 2017-04-04:
> I think its related to #6110 .

If you can load and check the version immediately before, and immediately after #6110 that would be a huge help.   If the version before behaves correctly but after #6110 it fails, that tells us exactly where to look.   

That code is fairly simple.   If we know the answer above says #6110 is the guilty party...   I bet we can resolve this pretty quickly.

benlye, 2017-04-04:
Could the problem be caused by https://github.com/MarlinFirmware/Marlin/pull/6181/commits/a5e085cbea35805fd241ced3fd195517e1fbac37, which changes the value returned by run_probe to the probe Z position rather than the nozzle Z position, committed in #6181?

@apballard, what's your Z-offset set to (both with M851 and Z_PROBE_OFFSET_FROM_EXTRUDER, if you have it configured)?

I'll try this out myself when I get home a little later.

Roxy-3D, 2017-04-04:
> Could the problem be caused by a5e085c, which changes the value returned by run_probe to the probe Z position rather than the nozzle Z position, committed 

Very possibly!    For example, the distance here would be in error by 2x the Z_PROBE_OFFSET_FROM_EXTRUDER.
```cpp
     #endif
 -    return current_position[Z_AXIS];
 +    return current_position[Z_AXIS] + zprobe_zoffset;
```

>@apballard, what's your Z-offset set to (both with M851 and Z_PROBE_OFFSET_FROM_EXTRUDER, if you have it configured)?

If the error you are seeing is 2x the Z_PROBE_OFFSET_FROM_EXTRUDER (or M851 number)  This is likely to be the source of the problem.

apballard, 2017-04-04:
I'll put some logging in and test and see what happens.

apballard, 2017-04-04:
@Roxy-3D @benlye 
Thanks for your help.  Yes, it seems it is indeed that line that is doing it. By reverting it, it restores original way of working.

There must be a reason why the change was made.  In my case, maybe my method of determining my z probe offset is not the best?  What is the best accepted way to calculate the Z Probe Offset?

Until now I set it to something like -2.  I then perform G28 then G29 then I put a piece of paper, move z axis till its just touching and subtract that from the -2.  M851 to set it, M500 to store it and there you go.

Like it is now, I don't know how to use it.

benlye, 2017-04-04:
I confirm the same thing - the correction matrix is out by my Z-offset (as set with M851), which is -3.60.  The verbose output shows the expected(?) output, but the final correction matrix is not like it was before:

```
Send: G29 V3
Recv: G29 Auto Bed Leveling
Recv: Bed X: 15.000 Y: 15.000 Z: -0.052
Recv: Bed X: 45.000 Y: 15.000 Z: 0.033
Recv: Bed X: 75.000 Y: 15.000 Z: 0.078
Recv: Bed X: 105.000 Y: 15.000 Z: 0.178
Recv: Bed X: 135.000 Y: 15.000 Z: 0.253
...
Recv: Bilinear Leveling Grid:
Recv:      0     1     2     3     4     5     6
Recv:  0 -3.65 -3.57 -3.52 -3.42 -3.35 -3.28 -3.26
Recv:  1 -3.78 -3.69 -3.64 -3.56 -3.47 -3.42 -3.38
Recv:  2 -3.85 -3.74 -3.68 -3.59 -3.52 -3.47 -3.47
Recv:  3 -3.86 -3.75 -3.71 -3.62 -3.56 -3.53 -3.52
Recv:  4 -3.81 -3.71 -3.66 -3.58 -3.52 -3.50 -3.53
Recv:  5 -3.75 -3.67 -3.63 -3.56 -3.50 -3.49 -3.51
Recv:  6 -3.64 -3.56 -3.53 -3.47 -3.42 -3.41 -3.44
```

benlye, 2017-04-04:
Also to confirm, with the current RCBugFix code the actual position of the nozzle after G29 is out by the Z offset.  If I issue G1 Z3, my nozzle is barely above the bed.  G1 Z0 would drive the nozzle into the bed.

This is quite a dangerous issue.

Roxy-3D, 2017-04-04:
> There must be a reason why the change was made.

@apballard   I don't know why this made sense.  But this is an example of why testing code changes is necessary.   If you switch to the UBL system, I can assure you....  I TEST ALL CHANGES MADE TO IT on my very own printers.

apballard, 2017-04-05:
@Roxy-3D Lol, too true.
Ok, I'm going to try out/swap over to UBL. I've got a feeling you could use extra testers for that anyway. The more the merrier. Used it before and was very impressed. No idea why I'm sticking with ABL. 

To the detriment of my health I love bleeding edge.  Why use a stable release when you can stab yourself all over and use something that's beta, or even better....... alpha.  So I'm keen on testing anything new.

benlye, 2017-04-05:
@Roxy-3D, unfortunately I can't use UBL yet because it is too large to fit in the flash space on my 1284p board.

I'd create a PR to try to fix this myself, but without knowing the rationale for this change, and what is now dependent on it, I'm not sure if simply reverting the change is the right thing to do or if the ABL code needs to be changed to handle the new values.  As he made the change, perhaps @thinkyhead can let us know?

benlye, 2017-04-05:
Looking at the bilinear G29 code, it seems like this needs to change to avoid double-adding the z offset when using bilinear levelling with the new run_probe output.

From:
```
            #elif ENABLED(AUTO_BED_LEVELING_BILINEAR)

              bed_level_grid[xCount][yCount] = measured_z + zoffset;

            #endif
```
To:
```
            #elif ENABLED(AUTO_BED_LEVELING_BILINEAR)

              bed_level_grid[xCount][yCount] = measured_z;

            #endif
```
(Currently at [Marlin_main.cpp:4475](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L4475))

I can't test it myself until I get home from work.

benlye, 2017-04-05:
There were two places in the bilinear G29 code that needed to have the `bed_level_grid[xCount][yCount] = measured_z + zoffset;` changed to `bed_level_grid[xCount][yCount] = measured_z;`.  I created PR #6251.

benlye, 2017-04-05:
> How much do we need to shrink it so you can get it to fit into your 1284p board?

The last time I tried it was 13KB too large.  I think my configuration is fairly standard, with all the large stuff like SD support and levelling debugging disabled, and nothing else too exotic turned on.

Roxy-3D, 2017-04-05:
OK...   We can get 13 KB back.   At worst case...   And I don't think this happens...  We could make it so you can turn off G29 while using G26 and vice versa.   But I think we can avoid that.

benlye, 2017-04-07:
FYI, a better fix for G29 bilinear levelling was merged in #6258.  This should be resolved now.

apballard, 2017-04-11:
It is resolved! Thanks!

closed 2017-04-11 by apballard