MarlinFirmware/Marlin #6234: question RE accelerations

ruggb, 2017-04-03:
what is the diff between these settings
#define DEFAULT_MAX_ACCELERATION      { 9000, 9000, 100, 9000 }

and these settings
#define DEFAULT_ACCELERATION          9000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  9000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   9000    // X, Y, Z acceleration for travel (non printing) moves

Some say NON-PRINTING, but if it is reliable for non-printing, why not just have it the same .
Also for E it is the same since E isn't moving unless it is printing, and the max for Z is almost 1000X greater than default.

It just seems like the DEFAULT_MAX..... limits what the slicer/printer s/w is commanding (which is where one would reduce it if desired) and a max for printing or non-printing should be the same reliable number.
If you reduce printing accel to stop skipping steps non-printing must be reduced also.

All these variables just seem to add to the confusion - at least I am confused!

Sebastianv650, 2017-04-03:
The DEFAULT_MAX_ACCELERATION sets the maximum acceleration an axis will ever go to. Marlin checks the actual acceleration against this one per axis and will lower it if necessary. This allows you to set a upper mechanical limit your printer can handle. Think of it like a safe rpm limiter of your cars engine. It's there, even if most people will nearly never hit it.

DEFAULT_ACCELERATION is the used acceleration for print moves (moves that includes an e movement). This value is not linked to a specific axis, it's always in movement direction. The value of it is mostly driven by the print quality (surface quality) you want to get. If it's too high, defects like corner ringing can occur. As stated above, this value will only apply if no DEFAULT_MAX_ACCELERATION of any moved axis is reached.
DEFAULT_RETRACT_ACCELERATION gives the acceleration for retract and prime (e-only) moves. Again, this can never be higher than DEFAULT_MAX_ACCELERATION for the e axis.
DEFAULT_TRAVEL_ACCELERATION is the same as DEFAULT_ACCELERATION, but is taken for moves where no extruder move is necessary. This one can be higher than the acceleration for print moves because a travel move will most likely never influence printing quality. Again, it can't break the safe max. acceleration per axis stated in DEFAULT_MAX_ACCELERATION.

> ALSO on the LCD you have under control Accel: 5000, then there is A-retract & A-travel set separately, so what does Accel: refer to? Are they all the same thing?

Accel is the actual value overriding DEFAULT_ACCELERATION. Same is true for A-retract = DEFAULT_RETRACT_ACCELERATION and A-travel = DEFAULT_TRAVEL_ACCELERATION.

> ALSO, why would you want a different homing feed rate for the LCD than the internal. they are the same default.

What settings are you refering to with this?

> ALSO, if I set the MINIMUMFEEDRATE and MINTRAVELFEEDRATE to 10, does that mean my printer will always be moving? I don't understand the purpose of this.

Not always moving, but if it moves it will move with at elast a feed rate of 10. If you would need it, you would understand the purpose of it - therefore ignore it, leaving them at 0 ;)

ruggb, 2017-04-03:
I have lost that RE the FEEDRATE. Maybe it had to do with MANUAL_FEEDRATE vs HOMING_FEEDRATE or MAX_FEEDRATE. I guess they are all different function. They just look alike because the defaults are the same.

Goes explanations, need to be in the config doc.
thanks.

closed 2017-10-31 by thinkyhead