MarlinFirmware/Marlin #6235: Config error

tjp2579, 2017-04-03:
I am getting this error please help im having no luck about to give up cant get pronterface working can configure marlin F*** 3d printing in to stupid to figure this out 
[image]
[image]

tjp2579, 2017-04-03:
Also cant get my reprap LCD to work also just a blue screen and I turned it on in marlin I was able to compile the marlin config.h one or two times a hundred downloads of different pythone and other stuff people say to try thought this would be fun :(

tjp2579, 2017-04-03:
I have a reprap DIY kit from amazon i3 with [Sintron] Ultimate 3D Printer Full Complete Kit for DIY Reprap Prusa i3 + RAMPS 1.4, Mega 2560, MK8 Extruder, MK3 Heatbed, Stepper Motor and LCD Controller except I upgraed to the reprap graphic smart controlle

Dreamloom, 2017-04-04:
Have you tried emailing Sintron? They advertise pre-configured Marlin firmware to load on their kit printers: sales@sintron-hk.com  - sounds like that might cure your headache.

tjp2579, 2017-04-04:
Thanks, I have tried that. I don't know what's wrong I tried deleting all
Marlin files and starting over. I wish I could figure this out.

Best Regards,

Toby Protsman || Student Government || Computer Club || Admissions
Ambassador

Dreamloom, 2017-04-04:
You have an unmodified copy of the Marlin firmware files from Sintron? Ones you haven't changed? Can you post the configuration.h here as a .txt file so the real experts can look it over? Also: is your Ramps Extruder/Extruder/Bed, Extruder/Fan/Bed or what configuration? As for the LCD, do you have Reprap Discount Full Graphic Smart Controller?

tjp2579, 2017-04-04:
Fan/extruder/bed and I down loaded the copy of Marlin from The Marlin
website. Still waiting to hear from starion I emailed them again today.
Thank you for your time to help me much appreciated.

Best Regards,

Toby Protsman || Student Government || Computer Club || Admissions
Ambassador

tjp2579, 2017-04-04:
I'm sorry not at home to get the text file copy for you will be after 6pm
Eastern time

Best Regards,

Toby Protsman || Student Government || Computer Club || Admissions
Ambassador

tjp2579, 2017-04-04:
Sorry here is my controller

Best Regards,

Toby Protsman || Student Government || Computer Club || Admissions
Ambassador

teemuatlut, 2017-04-04:
Please post your configuration.h and configuration_adv.h here (rename the extensions to .txt).

Which version of Marlin are you running? The latest RCBugFix?

Tannoo, 2017-04-04:
>I am getting this error please help im having no luck about to give up cant get pronterface working can configure marlin F*** 3d printing in to stupid to figure this out

Calm down a type clearly will help us help you.
It looks as though you don't have the motherboard set correctly.

In `Configuration.h`, change the board name or number to the one you found in the `boards.h` file that matches your setup.
  #define MOTHERBOARD _BOARD_RAMPS_14_EFB_

tjp2579, 2017-04-04:
Ok just got home from work I just deleted any files I had downloaded for this let's start fresh what's first Marlin?

tjp2579, 2017-04-04:
Were should I save it on desktop or in default program file

Tannoo, 2017-04-04:
Desktop is fine for now.

tjp2579, 2017-04-04:
That is the first file I should start with or adrunio first I deleted everything

Tannoo, 2017-04-04:
>what's first Marlin?

Editing the `configuration.h` file.

Tannoo, 2017-04-04:
You need the Arduino IDE to compile and send the program to the controller.

You can edit the `configuration.h` file with whatever text editor you choose.

tjp2579, 2017-04-04:
And Python I had someone tell me to download a bunch of Python files

Tannoo, 2017-04-04:
Python has nothing to do with this. No need for it.

tjp2579, 2017-04-04:
Ya im kinda puter savy but this has me stumped

Tannoo, 2017-04-04:
Python is another programing language. Not Arduino.

Tannoo, 2017-04-04:
Arduino is roughly based on C++.

Tannoo, 2017-04-04:
I am headed to work now.... I will help when I can. (via phone).

tjp2579, 2017-04-04:
God I hope you get this going I'll love you forever​ lol I was so excited and now the spark is gone now it's fustration

tjp2579, 2017-04-04:
Oooo nooo

tjp2579, 2017-04-04:
Should I download Marlin for GitHub or there sight

Tannoo, 2017-04-04:
I'm not _quite_ done yet. lol

tjp2579, 2017-04-04:
Ok have Marlin and Andunio  downloaded no I have configuration.h open set to my board next is my LCD and that should do it or what is next

Tannoo, 2017-04-04:
Go here:
https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

And download as ZIP.

Tannoo, 2017-04-04:
That would be the lastest _bleeding_ edge code.

Tannoo, 2017-04-04:
Without your old config files, the process of setting up your printer correctly will take more time because things will have to be figured out.
Unless someone here might have the same printer as you do and could provide their config files for you to start with.

Tannoo, 2017-04-04:
If it has to be done the hard way, someone here smarter than me, might provide the directions to find the steps/mm for your printer.

Tannoo, 2017-04-04:
I gotta get. I'll be around. I'll help where I can.

tjp2579, 2017-04-04:
reading the config. file on marlin webpage ill fallow that

tjp2579, 2017-04-04:
It compiled no LCD display though whats next to get this thing moving?

tjp2579, 2017-04-04:
[config.txt](https://github.com/MarlinFirmware/Marlin/files/894693/config.txt) this is my config.h file I have a reprap Full Graphic Controller with just a blue screen

tjp2579, 2017-04-04:
Help please if someone is out there

tjp2579, 2017-04-04:
now I tryed changing some setting the recompiling it and now im getting a error exit status 1

tjp2579, 2017-04-05:
got it moving guy thanks for all your help still dont have a LCD ill work on that later

Tannoo, 2017-04-05:
Watch this video. He will give you a better understanding of Marlin's setup.

https://youtu.be/3gwWVFtdg-4

tjp2579, 2017-04-05:
Thanks watched it a hundred times

Bob-the-Kuhn, 2017-04-05:
He has a "smart adapter" to interface the two LCD cables to the RAMPs card.  Does anyone know if changes to the pins_RAMPS.h file are needed or is it plug & play?

tjp2579, 2017-04-05:
Im getting this error with prontoface 
[image]

Tannoo, 2017-04-05:
Before you jump that far... Get connected to your printer and verify endstops are good and motors move the correct direction, etc.

Do this before worrying about any slicer yet.

Tannoo, 2017-04-05:
I saw tjat you have pronterface's baudrate lower than Marlin's default.
Default speed is 250000.
The baudrate needs to be the same on both ends.
(i.e. Marlin <-> Arduino IDE, or Marlin <-> Pronterface)

tjp2579, 2017-04-05:
Ya I changed the baudrate on both I got a error said it was to high

Tannoo, 2017-04-05:
That's odd. I run 250000 on a 6' usb cable with no issues.

tjp2579, 2017-04-05:
You have the same setup as me

closed 2017-10-31 by thinkyhead