MarlinFirmware/Marlin #6236: Z probe homing issue

allmighty71, 2017-04-04:
I am in the process of building my own printer and have installed a PNP NO inductive probe for Z.  I have managed to get the probe to work and the bed stops when it reaches the probe, but then it backs off to do the second bump and stops.  It wont do the second bump and it now wont let me move closer to the probe.  When I move the bed away from the probe it acts as if I set a new zero and wont let it move back.  In short z wont bump and I can't move closer to the probe.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/892040/Configuration.txt)

Roxy-3D, 2017-04-04:
You have these settings in your Configuration.h file:
```cpp
#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.
```
I've never used an inductive probe...  But I think you are going to have problems if you have the min_software_endstops option set to true.   That might be what is keeping you from doing the double bump.    Give that a try and see what happens.  If that doesn't help you, there are a lot of people here using inductive probes...  somebody is going to be able to tell you what really needs to be changed.

Also...  your #define Z_PROBE_OFFSET_FROM_EXTRUDER 0   definition doesn't seem plausible.   Almost for sure this is a positive number.

allmighty71, 2017-04-04:
Thanks for the help, I found out I cant have the negative wire from the probe to the negative pin on Z MIN ENDSTOP.

closed 2017-04-04 by allmighty71