MarlinFirmware/Marlin #6237: extreme overextrusion after first printjob

azalanono, 2017-04-04:
I have a reproducable problem on my Marlin RC8 based 3D printer.
While the first printjob works fine, any succeeding job does normal moves, but does extreme over-extrusion in skirt. I can retry as often as I want, the behavior does not change.
I found out that restarting Marlin (a disconnect and re-connect forses a firmware-reboot) and then starting the same job makes it print (once) correctly.

I use OctoPi as print server.

[image]

Tannoo, 2017-04-04:
It may also be the extruder stepper driver losing control.
Could also be a bad connection with that driver.
Could also be a bad connection to the extruder motor.

To eliminate Marlin as the issue, download a fresh copy of RC8 or RCBugFix.

Sebastianv650, 2017-04-04:
For me it's very intresting that the massiv blobs mostly happens in the arcs, not the straight lines. Have you accidently enabled LIN_ADVANCE with a wired K value?

thinkyhead, 2017-04-04:
@azalanono Have you tested the current `RCBugFix` to see if the problem still exists?

knobik, 2017-04-13:
Im having the same problem, reproduced it 2x, will try RCBugFix today after work.

knobik, 2017-04-14:
RCBugFix fixed that problem for me.

closed 2017-11-01 by thinkyhead