MarlinFirmware/Marlin #6241: Not an issue, but a question about G29

wookie666, 2017-04-04:
Sorry, i cant figure out how to set something as a question

I am using the firmware from here, http://www.thingiverse.com/thing:1976680 which basically has just been modified to save G29 to eeprom, because i am using a manual fixed endstop right under the nozzle, which i place and remove so i dont have to do it every print.

my question is, is there a way that this can be done on the regular unmodified marlin, like rcbugfix? Is there a way i can do autolevel, then have it save those settings and not delete them when it does a G28, so that it can use the bed mesh for each print, until i either reset it using M502, or i autolevel again?

This so far is working the best as my endstop is right under the nozzle, and i get the most accurate way of probing.

The other question would be, what is the best way in this instance to set and adjust the z offset. I set the offset to say -19.3, then auto level, then remove the probe and find its wrong, i then adjust and probe again, then remove the probe and test Z0 and its fine. But then when i do a G28, it seems as though the M851 settings are not used anymore, and the manual_z_height is used. I can get around this by adjusting M206 to adjust my Z0, but is this the best way.

If this is hard to understand please ask. Im looking for a better way of probing on my delta, without having to have a servo probe, or probe every print.

Thanks.

Bob-the-Kuhn, 2017-04-05:
Marlin has been able to save to EEPROM & restore for quite some time.

After G29 issue a M500 command to save the G29 results to EEPROM.  They'll be automatically restored on power up and after a reset.  To make them active you'll need to do a G28 and then a M420 S1.

Within the last week or so they've just modified the M851 command so that a change in Z is immediately reflected in the bilinear settings.  Previously you needed to re-run G29 for the new setting to be used.

You can get the latest RCBugFix [here](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).

The hardest part will be updating your configuration files.  Names have changed and options have been added so it's best to grab the files that come with RCBugFix and copy all your machine specific items over.

wookie666, 2017-04-05:
Exactly what I wanted to hear. Excellent 
So m420 S1 is the recall bed levelling grid. I can add that after g28 in my start code. 

And then if using the latest rcbugfix I can use m851 to adjust my z correctly without having to do a g29 again or use m206. So as soon as I do m851. Them do g28 then move z to 0 that m851 is taken into account is that right?

Sorry for the questions I just want to get this right 

Also with the current version I am running. The g29 bilinear grid is shown in eeprom when typing m501. Does the latest rcbugfix do that too? Or is there an indication where I can see what the current bilinear grid is set to. I've noticed that others are able to update certain spots in the grid. How can I do that

Just to confirm I'm running a delta so I assume that some options might not be available to me at this stage

wookie666, 2017-04-05:
Or is there some documentation showing all the updated commands that can be used and what they do. Like the m420 command you just mentioned

Bob-the-Kuhn, 2017-04-05:
The only documentation is the comments at the beginning of the **gcode_M420()** routine.

No telling how long before we have the time to update the Wiki.

G28 needs to be run only once.  No need to run it again until reset or you bang up against a hard stop and loose steps.

M420 S1 turns on bed leveling.  G28 and M420 S0 turns it off.

Grids/meshes/compensation are either loaded from EEPROM during the powerup/reset initialization or created by G29.

UBL bed leveling can have multiple meshes saved in EEPROM.  To load a particular mesh then use the G29 Lxx or M420 Lxx commands.

UBL doesn't support delta printers yet.  It's in the works.  Unfortunately I can't give you a good estimate of the time line.

Roxy-3D, 2017-04-05:
> UBL doesn't support delta printers yet. It's in the works. Unfortunately I can't give you a good estimate of the time line.

If some people with Delta's want to load it up and tell us how far they get before things don't work...   I'll see if I can fix the problem.    Or...  Somebody else might be able to jump in with an easy answer.   I'm pretty sure there are going to be issues just because of how different Delta's are when they move.   But the UBL System was originally designed to fix the problems of a very low quality Geeetech Delta printer.

wookie666, 2017-04-05:
Ok so basically if my start code for every print co Sista of g28. Then m420
S1. It will load my bed level grid each print . As long as I do a g29 then
m500. Is that correct?

Bob-the-Kuhn, 2017-04-05:
That is correct.

Roxy-3D, 2017-04-05:
Probably...  (I can't pull up the source code now but all of the original Mesh commands were respected.   I don't use them.)   You always need a G28 because the machine needs to know where it is positioned.   But then G29 L1 will load UBL Mesh #1.   But if you saved the Mesh with UBL Active...   You don't even need the G29 L1.   When the system powers up and loads things from the EEPROM....   You will have the current mesh in its active or disabled state.   Just power up, do a G28 and then a G29 W.   It will tell you its view of the world.

Wookie, what are you trying to accomplish?   Are you trying to save a mesh?  Or do you just want things convenient every time the printer powers up?

wookie666, 2017-04-05:
Sorry mate. Now we are talking about ubl. I just want my Auto level mesh to
be saved and not reset until I do a g29 again.  Cause I Do an autolevelling
with a removable endstop I don't want to have to do this every print. So
what exactly do I need to do to have my mesh saved. And automatically used
every print. At this stagei am using bilinear because unlike doesn't work
on Delta's

wookie666, 2017-04-05:
Roxy, Bob.

Can someone tell me why when i have my z probe offset and home offset set to 0, that when i turn on auto levelling using M420 s1, my z height adjusts by 0.29?

This is what i am doing.

1. Set M851 z-19.3, as that is the offset from my probe to my hotend.
2. G28
3. G29
4. M500 to save the autolevel
5. G28 (M114 then shows Z @ 217.71, this height is perfect for my bed, 
6. M420 S1 to load the autolevel (M114 then shows Z @ 236.72)
This then adjusts my Z by 19.3mm as that is what my M851 was set to. This is wrong in my instance, because now, my printer always wants to print 19.3mm into the bed.

So i go ahead and set M851 to 0.

7. G28 to auto home again (screen shows Z @ 217.71)
8. M420 S1 to load auto level (screen then shows Z @ 217.42)
Somehow there is 0.29mm offset coming into play. My M206 is set to 0
Im not sure on what other offsets there are to work with here.

What am i doing wrong here. If i dont set M851 Z-19.3 before I do a G29, then the probe doesn't slow down before it hits the bed and it goes all weird as it probes. Almost as if it wasn't expecting to probe then, as its moving too fast (due to the fast homing still being active). If i set the M851 Z-19.3, the probe slows down approx 20mm above the bed, and then probes correctly.

After that though, i need to set M851 to 0, as its still taking into account the offset and puts my Z height wrong.

Please advise.

wookie666, 2017-04-05:
Also, using the latest RCBugFix

benlye, 2017-04-05:
@wookie666, if I read this correctly you are using bilinear bed levelling with the latest RCBugFix.  If that's true, you are quite likely running into the issue described in #6233, where the Z offset is being misapplied to the bilinear bed correction grid.  I don't know if this accounts for all the weirdness you're seeing, but it is likely to be at least part of the problem.

wookie666, 2017-04-05:
its very possible Benlye, I can get around it though by adjusting my Z offset using M851 to get my Z height the same after I turn on auto level with M420 S1. 
Although i ran some tests by manually putting my bed out of level by about 0.8 - 1mm and for some reason, When i then check different points on the bed after doing the auto level and saving it to the Eeprom, the points can vary by up to 0.3mm. Almost like the bed levelling isn't working. The hard part is, that because im running on a delta, all motors are always moving so i cant tell if the machine is actually offsetting for the autolevel grid.

I am probing the entire bed in a 9 x 9 grid, and coming in 20mm from the printable radius, So where it cant reach it wont. This way i probe the whole bed. I have also set subdivisions to 0, as i wasn't sure if it would overwork the Trigorilla board i am using. 

Will the G29 compensate for a delta machine not being fully calibrated for convex/concave printing due to the radius not being 100% for the calibrations.

I mean i am within 0.1 of a mm if i probe different spots just after doing an auto home, without using the auto leveling grid... Any if my bed is near perfectly level, i am still within about 0.1mm of testing. Although if i manually make the bed about 1mm out of level, then it doesn't account for that level properly.

if you need me to do more testing i can

Roxy-3D, 2017-04-05:
All of the options are in the front of the UBL files.  In UBL_G29.cpp it says:

*
   *   A     Activate   Activate the Unified Bed Leveling system.
   *
   
   *
   *   S #   Store      Store the current Mesh at the specified location in EEPROM. Activate this location
   *                    for subsequent Load and Store operations. It will also store the current state of
   *                    the Unified Bed Leveling system in the EEPROM.

If you do a G29 A   followed by a G29 S 1    The mesh will load from slot 1 every time the machine resets and it will be active.

wookie666, 2017-04-05:
But I can't use ubl as it won't load

oldmcg, 2017-04-07:
I am interested in trying UBL on my delta with bltouch.  I'm on rcbugfix as of a few days ago.

I have some questions about UBL and mesh with respect to delta.

1) Are there some existing known problems with UBL on rcbugfix that prevents working on delta, or is it just currently untested? 

2) With current code, should the specified rectangular grid to probe completely include the round bed, meaning it will simply skip unreachable corner points outside the DELTA_PROBEABLE_RADIUS, and hopefull not attempt to move to them?  Or does the specified probe grid have to be constrained to a square that fits within the round bed?

3) If probe while extruder head at X0,Y0 but Y_PROBE_OFFSET_FROM_EXTRUDER is -24, is the probed location recorded as X0,Y-24 or X0,Y0?  In other words, is the XY location of the physical probe recorded for the mesh, or the XY of the extruder head locaiton when the probe is extended, even though probe is offset from extruder?

4) Is the specified grid to probe (in configuration.h) relative to the extruder head or the probe XY offset?  In other words, if BACK_PROBE_BED_POSITION is 70, then will it try to probe with extruder at 70 or with physical probe at 70?  

5) Does the resulting mesh have to be rectangular points, or can they be hexagonally spaced?  I know the code doesn't create hex mesh currently -- just curious if I create code to do hex mesh if the leveling compensation will work correctly from that.  It seems more relevant to delta printers than square mesh points.

Thanks.

Bob-the-Kuhn, 2017-04-07:
> Are there some existing known problems with UBL on rcbugfix that prevents working on delta, or is it just currently untested?

Mostly untested.

> With current code, should the specified rectangular grid to probe completely include the round bed, meaning it will simply skip unreachable corner points outside the DELTA_PROBEABLE_RADIUS, and [...]

The grid is bounded by the min/max X & Y values as set by the DELTA_PRINTABLE_RADIUS minus the UBL_MESH_INSET.  The grid points that fall outside the DELTA_PRINTABLE_RADIUS are ignored.

When deciding if a point is probeable it uses DELTA_RADIUS and the probe offsets.

> If probe while extruder head at X0,Y0 but Y_PROBE_OFFSET_FROM_EXTRUDER is -24, is the probed location recorded as X0,Y-24 or X0,Y0? In other words, is the XY location of the physical probe recorded [...]

Let's assume that one of the points in the grid is 0,0.  When **probing** the PROBE will be at 0,0 and the nozzle will be at what ever location is needed to get the probe at the correct location.  When doing a **move** when UBL is active then the Z compensation at 0,0 is applied when the NOZZLE is at 0,0.

> Is the specified grid to probe (in configuration.h) relative to the extruder head or the probe XY offset? In other words, if BACK_PROBE_BED_POSITION is 70, then will it try to probe with extruder at 70 or with physical probe at 70?

It'll try to probe with the probe at 70.

> Does the resulting mesh have to be rectangular points, or can they be hexagonally spaced? I know the code doesn't create hex mesh currently -- just curious if I create code to do hex mesh if the [...]

Right now everything is based on rectangles.  Commanded moves (G0, G1 moves)  that cross grid/rectangle boundaries are split up into moves from one boundary to the next and a final move that completes the commanded move.

The "fun" part is determining when a boundary is being crossed and what the Z compensation is at that point.

Roxy-3D, 2017-04-07:
Most of the code should work.   But things like bounding the probe pattern to a circle have not been done.   As soon as all of the functionality is in UBL and things slow down, Delta's will be next.    AND...  it was designed to help Delta's Cup & Bowl problem    

But we are still a few weeks away from trying to get the extra logic and limits into it for Delta's.

oldmcg, 2017-04-07:
Yes, I just attempted UBL after commenting out a bunch of "#error not supported" and fixing some defines.  The first 3 probes worked fine, but on the fourth probe it attempted to move the head to a corner well outside the delta printable radius causing the geometry to bind until I hit the power.

I will stare a the code for a while to see where I can add some limits.  In the meantime I will try BILINEAR.

Also curious about the delta cup and bowl correction.  There are also tower angle corrections, delta radius corrections, etc, that can be inferred from the resulting mesh, not just compensating for uneven or sloped bed.  Is UBL intended to address that as well or is that a different beast entirely?

Thx.

wookie666, 2017-04-08:
Just tried Repetier and i really dont like it. Im keen to see auto levelling working properly on marlin, it works at the moment, but im sure it can be better.

I guess the biggest issue at the moment is i need to set the M851 Z-19.3 before i probe which is fine. Then i do a g29, then save to eeprom which is all good

But when i do a g28 to home, then turn on auto levelling, it copies that M851 information and offsets my z by 19.3, Which i suppose is what is is supposed to do. But the problem is, i dont always have the probe on, its a bracket i screw on manually to probe, then take off. and the endstop sits below the nozzle. Is there anyway this can be rectified?

Bob-the-Kuhn, 2017-04-08:
You have a removable Z probe attached to Z_MIN_PROBE_PIN and a Z_MIN endstop attached to Z_MIN_PIN.  And the problem is that the Z_MIN endstop is so low that the nozzle will hit the bed during G28.  Am I correct?  

Or is it that both the Z probe and the Z Min endstop is attached to the same Z_MIN_PIN?

Either way I think you have two choices:
- Move the Z_MIN endstop up so that it triggers when the nozzle is about 10mm above the bed.  That should result in the Z probe triggering before the Z min endstop triggers.
- Use a Z_MAX endstop and home in the +Z direction (by setting Z_HOME_DIR to +1).  You'll need to set Z_MAX_POS to a number small enough that the G28, G0 Z0 sequence doesn't crash the head into the bed.

I like using the M43 E1 command/mode to set endstop position.  You'll need to enable PINS_DEBUGGING in Configuration_adv.h in order to use it.  This command scans the endstops about 10 times a second and reports any changes via the USB port.  It also toggles the system LED when there is a change.  What I do when setting an endstop is to move the axis to where you want the endstop to trigger and then watch the LED as I move the endstop.  That'll get you close.  After moving the head away & back to the endstop you may want to do some fine tuning.

wookie666, 2017-04-08:
I have a delta with x y and z max . The z min is an endstop which is a
removable probe so what happens is it will home when it's removed I don't
have a zmin endstop plugged in. If I set the  offset to -19.3 to do g29 it
works fine. The issue then is when I remove the probe it still uses the z
offset. So what happens is it will home and z is then let's say 280. If I
turn on the autolevelling using m420 S1 it then changes the current z
position to 299.3. Which in turn makes my nozzle crash into the bed. So
what I do then is change my z offset to 0 and for some reason it's still
getting about 0.3mm from somewhere. So it will then change z to 280.3. But
I don't have any other offsets defined anywhere. So what I need to do is
adjust z offset to account and fine tune. At the moment my z offset is set
to -0.22 which this works perfect. So to recap. My manual z height is fine
if I don't enable autolevelling. When I enable autolevelling it takes into
account my z offset. If I don't set my z offset to -19.3 then when it
starts a g29 it doesn't slow before it gets to the bed and ends up crashing
and going out of place. .
And for some reason with z offset set to 0 and my home offsets set to 0 it
still gets 0.3mm offset from somewhere and I can't figure that out.

Bob-the-Kuhn, 2017-04-08:
What a nightmare!

I think you should download RCBugFix again.  One of the "improvements" a week or two ago ended up causing the Z offset to be included twice.  That was fixed 2-3 days ago.

I have no idea where the 0.3mm is coming from.

wookie666, 2017-04-08:
Ah ok I'll have to take a look if it was done a few days ago.

Roxy-3D, 2017-04-08:
And...   As it turns out....   You don't have to have a Z-Probe to use UBL.   It does cut out an iteration to automatically probe.   But you can zero the mesh and hop straight to a G26 Mesh Validation pattern.   At that point you are editing the mesh and you run a 2nd G26 to see what areas still need improvement.

It  will be 3 or 4 weeks until I can bring Delta's up.    Maybe somebody with a Delta will have fun bringing it up ahead of schedule?

oldmcg, 2017-04-08:
I'm working through the kinks on my delta to try to enable UBL.  Hard to balance between minimal code churn and making things "better".  

For example, position_is_reachable only tests XY for delta/scara but XYZ for cartesian.  

Should it check XYZ for all three or only XY for all three?  Or leave "as-is" with XY check for delta/scara and XYZ for cartesian?  Most call sites are only looking for XY.

Should I change the parameters from XYZ array to just X and Y since most callers create a stack array and stuff just two locals and a zero into it before calling the function?

Should I move it to header and inline it so delta results in just two multiplies and an add/compare and cartesian results in just four or six compares?

For delta I am also making it check whether both the head and the probe (for by_probe) are within the printable_radius.  Should I extend that logic to scara and cartesian as well?  I can't test them but logic says they ought to behave the same.

Also, curious why BILINEAR does not require EEPROM but UBL requires EEPROM.  Does BILINEAR without EEPROM use G29 results until next reboot?  If so, can't UBL do the same?

Thanks for any pointers in the right direction.

Roxy-3D, 2017-04-08:
> Also, curious why BILINEAR does not require EEPROM but UBL requires EEPROM. Does BILINEAR without EEPROM use G29 results until next reboot? If so, can't UBL do the same?

We do need to relax the contraint in UBL that you have the EEPROM enabled.   But that is because it uses the EEPROM to load the mesh at reset.   And you need a place to store a mesh and that is in EEPROM.   But now with the 3-Point and Grid leveling options, some people might choose to just use a mesh that is all 0.000's  (created with a G29 P0).    But for right now...  It loads a mesh out of EEPROM.

iNherjer, 2017-04-10:
Hi there 
Wehen i try to make a g29 safe with m500 reboot and do a m420 s1 the printer goes nuts! When I try to move down a few mm , the Z axis begins to rise , crash into the endstops and the belts begins to rattle until I send an emergency stop. I  have the rc8

Bob-the-Kuhn, 2017-04-10:
Please try [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).  It has a lot of fixes since RC8.

You'll need to transfer your machine specific settings over to the new config files.  More options and some name changes.

Roxy-3D, 2017-04-10:
> Wehen i try to make a g29 safe with m500 reboot and do a m420 s1 the printer goes nuts! When I try to > move down a few mm , the Z axis begins to rise , crash into the endstops and the belts begins to rattle until I send an emergency stop. I have the rc8

The setup for moves is different on Delta's.   If you look at how a G1 gets its moves accomplished on Delta's versus Cartesian machines, you may be able to shift things.    And if you do...   Please post your changes and I'll fold them into the UBL code.    But it will be 3 or 4 weeks before I start bringing up UBL on Delta's.

iNherjer, 2017-04-11:
I have tried to configure the RCbugFix but when it compiles it jumps to the sanityCheck and says : 

endif

/**
* Allow only one bed leveling option to be defined
*/
static_assert(1 >= 0
#if ENABLED(AUTO_BED_LEVELING_LINEAR)
+ 1
#endif
#if ENABLED(AUTO_BED_LEVELING_3POINT)
+ 1
#endif
#if ENABLED(AUTO_BED_LEVELING_BILINEAR)
+ 1
#endif
#if ENABLED(AUTO_BED_LEVELING_UBL)
+ 1
#endif
#if ENABLED(MESH_BED_LEVELING)
+ 1
#endif
, "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
);

I checked the configuration.h again and only the bilinear function is activated... can't find the problem

Bob-the-Kuhn, 2017-04-11:
Strange – I can’t get to this post on GitHub.

 

Please open a new issue, ZIP up both your config files and drop the ZIP there.  I’ll try to reproduce the problem.

 

   Bob

iNherjer, 2017-04-11:
Thanks - I solved the problem- it was my own stupidity... I opened the sketch in the old arduino software instead of the 1.8 ... sorry for trouble

Bob-the-Kuhn, 2017-04-11:
No trouble.

We really like to hear when things get solved.

Happy printing!

oldmcg, 2017-04-11:
I've been working on UBL for delta.  The current code (in lots of places) attempts to separately validate movements before calling either do_blocking_move_to or probe_pt.  

It would seem to make the code a lot cleaner, smaller, and safer if the call sites to these functions left the coordinate validation to the routine itself.  

probe_pt could return NAN or other fixed error value for out of range coordinates, which may or may not be useful to just store in the mesh data.  

do_blocking_move_to and its variants are currently void but could return a boolean success/failure.  How a caller chooses to act on a failed move or probe would still be up to the call site, including just ignoring the failure.  

Attempting to move outside of printable radius on delta can potentially damage the printer (maybe on other types of printers as well, I don't know).  For my printer, I would prefer that the firmware absolutely prevent that.  And the only way that seems safe in the code is to enforce it inside do_blocking_move_to.

But I have no idea how that would effect performance outside of probing.  For cartesian coordinate validation it is just 4 compares.  For delta it is two float multiplies, one add, and a compare.  From what I can see in the code, segments_per_second would not be affected because that would be inside blocking_move_to but outside prepare_move..

Thoughts?

Roxy-3D, 2017-04-11:
One thought would be this:

Is it possible if the nozzle is told to leave the printable radius to stop right at the edge of the printable radius.   And an ideal follow up would be this:   When the nozzle is told to move back within the printable radius, it is smart enough to follow the arc between its current position and where it is going to re-enter the printable radius.   And at that point, it continues on in a normal fashion.

oldmcg, 2017-04-12:
My concern is that would be both computationally taxing and conceptually misleading.  Say you did probe_pt( 100, 100 ) and got back a z value of 0.05.  But that probe really happened at (70.71, 70.71), not at (100, 100).  It would also require storing both the state of where the nozzle is "physically" located in addition to the state of where the nozzle is "virtually" located, and that would be in addition to the raw vs logical coordinates already in the code.  

I will ponder it some more.

Roxy-3D, 2017-04-12:
So...  I don't know that it makes sense to do this...  But if the nozzle attempts to leave the bed, we don't want that to happen.   But once it gets to the edge of the bed all movement gets clamped.  It quits moving.   But still... We need to track where the nozzle would be if it really was moving.

And when it does re-enter printable area, that is when the nozzle would be allowed to move again.  Something like this would probably best be done at a very high level.  Probably at the top of the G1 command.

oldmcg, 2017-04-24:
@Roxy-3D, I have made some changes to is_position_reachable logic with respect to delta probing and moves.  For now I'm just doing bilinear to learn the code before I look deeper at the UBL stuff.  I don't know if you're interested in these changes or not -- they prevent my delta from attempting to move outside the printable_radius and result in a valid bilinear bed map.  The unreachable part of a delta bed probe looks like a crescent moon, allowing the probe grid to be specified as containing the entire printable radius instead of probe grid corners fitting inside the printable radius.  It still requires interpolating the unreachable grid points, but that code already exists.

[image]

I moved the is_reachable stuff out to marlin.h so that ubl_g29.cpp could reference it as well.

I have no idea how to get my local branch changes where other people can look at them.  Is it as simple as git push?  I don't want to do anything bad to the respository.

If you're interested let me know, otherwise I'll keep my changes in my own fork.  Thx.

Roxy-3D, 2017-04-24:
@oldmcg   Do you have the nozzle moving around and doing some what expected behavior for some of the G29 commands?  If so...   We can probably make progress fast.   I will help you as much as I can!!!  

Now...  imagine a grid (or mesh) super-imposed on top of the crescent moon.   There will be mesh points off of the printable area that we are going to ignore.   But we will still need to fill in and edit the mesh points within the crescent moon.    That won't be hard to do if you have the nozzle moving around for some of the G29 sub-commands!!!

oldmcg, 2017-04-24:
Yes, I can try some other G29 commands.  Currently I’m using G29 V4 D T but for some reason P11 didn’t result in 11x11 grid until I changed in configuration.h.  So far I’ve avoided grid 15x15 because you mentioned memory corruption problem in another thread (which sounds like a lot of fun).

 

The bilinear routine (if not doing dry run) fills in all the missing grid points using interpolation, both inside the crescent and outside the printable_radius to the far corners of the grid.  I’m not convinced it is the best algorithm – I think linear regression or polynomial interpolating a direct line from the origin to each unprobed point using only actual probed data is better than neighbor-fitting from other interpolated points.

 

I know there is a way to manually fill in the probe-unreachable crescent points.  But right now I’m just trying to make it probe a round bed to the extent the probe can physically reach without going outside the geometrically possible radius, and that seems to be working at least in the bilinear case.

 

My bed is intentionally slanted to test the code.  Here is the raw data and a surface map without the interpolated points (excel just treats them like zeros):

 

	
-70

-56

-42

-28

-14

0

14

28

42

56

70

-70

					-0.548

					

-56

		-0.679

-0.519

-0.449

-0.299

-0.207

-0.1

-0.007

		

-42

	-0.621

-0.508

-0.357

-0.25

-0.133

-0.013

0.08

0.16

0.259

	

-28

	-0.458

-0.247

-0.142

-0.001

0.101

0.221

0.308

0.378

0.39

	

-14

	-0.179

-0.022

0.03

0.179

0.329

0.421

0.479

0.492

0.585

	

0

	-0.011

0.197

0.283

0.433

0.567

0.679

0.773

0.838

0.813

	

14

	0.222

0.39

0.521

0.648

0.731

0.827

0.899

0.979

0.988

	

28

		0.578

0.679

0.799

0.933

0.999

1.059

1.139

		

42

				1.091

1.083

1.16

				

56

											

70

											

 

 

 

 

And here is the surface map letting the existing bilinear interpolation fill in all the unprobed points…

 

 

 

 

You can see where the interpolation produces some rather sharp corners, but I’m not sure it matters because they are well out of printable radius.

Roxy-3D, 2017-04-24:
> Yes, I can try some other G29 commands.  Currently I’m using G29 V4 D T but for some reason P11 didn’t result in 11x11 grid until I changed in configuration.h.  So far I’ve avoided grid 15x15 because you mentioned memory corruption problem in another thread (which sounds like a lot of fun).

UBL's G29 is totally different from the previous versions.   'P' is used for Phase.    A typical sequence will start with a G29 P1      That is going to invalidate the entire mesh, and probe what it can.  If you can get that to probe some of the points...   We are off to the races!!!   If it won't even probe the first point, well, it is going to be some work to get things going.

> I know there is a way to manually fill in the probe-unreachable crescent points.  But right now I’m just trying to make it probe a round bed to the extent the probe can physically reach without [...]

See if you can get it to probe even a few points...  If so, it won't be hard to make the probing stay on the bed.

> My bed is intentionally slanted to test the code.  Here is the raw data and a surface map without the interpolated points (excel just treats them like zeros):

My suggestion is don't do that...  physically level your bed as well as you can.  Delta's movement is so horribly flawed, you are going to have plenty of funny data to work with even if your bed is perfectly level.

oldmcg, 2017-04-25:
Ok, first pass on UBL code changes for delta, G29 P1 probed all accessible points and then stopped.

```
>>> g29 v4 W
SENDING:G29 V4 W
ubl.eeprom_start=72
Unified Bed Leveling System Version 1.00 Inactive.
No Mesh Loaded.
planner.z_fade_height : 0.00
z_offset: 0.000000
X-Axis Mesh Points at: -70.0  -56.0  -42.0  -28.0  -14.0  0.0  14.0  28.0  42.0  56.0  70.0
Y-Axis Mesh Points at: -70.0  -56.0  -42.0  -28.0  -14.0  0.0  14.0  28.0  42.0  56.0  70.0
Kill pin on :41  state:1
ubl_state_at_invocation :0
ubl_state_recursion_chk :0
Free EEPROM space starts at: 0x0048
end of EEPROM              : 0x0FFF
sizeof(ubl) :  1
z_value[][] size: 484
EEPROM free for UBL: 0x0FB7
EEPROM can hold 8 meshes.
sizeof(ubl.state) : 6
GRID_MAX_POINTS_X  11
GRID_MAX_POINTS_Y  11
UBL_MESH_MIN_X         -70
UBL_MESH_MIN_Y         -70
UBL_MESH_MAX_X         70
UBL_MESH_MAX_Y         70
MESH_X_DIST        14.000000
MESH_Y_DIST        14.000000
Unified Bed Leveling sanity checks passed.
>>> g29 v4 O
SENDING:G29 V4 O
ubl.eeprom_start=72
Bed Topography Report:
(0,10)                                                                                                                (10,10)
(65466,65466)                                                                                                            (70,70)
    .        .        .        .        .        .        .        .        .        .        .

    .        .        .        .        .        .        .        .        .        .        .

    .        .        .        .      1.131    1.213    1.200      .        .        .        .

    .        .      0.718    0.649    0.839    0.903    0.999    1.059    1.089      .        .

    .      0.152    0.340    0.461    0.618    0.711    0.797    0.859    0.899    0.928      .

    .     -0.001    0.147    0.293    0.383  [ 0.467]   0.589    0.633    0.648    0.773      .

    .     -0.199   -0.072    0.080    0.179    0.279    0.371    0.459    0.522    0.545      .

    .     -0.398   -0.267   -0.142   -0.031    0.091    0.121    0.238    0.308    0.390      .

    .     -0.621   -0.498   -0.367   -0.250   -0.143   -0.063    0.030    0.130    0.189      .

    .        .     -0.659   -0.569   -0.459   -0.359   -0.267   -0.160   -0.087      .        .

    .        .        .        .        .     -0.528      .        .        .        .        .
(65466,65466)                                                                                                                (70,65466)
(0,0)                                                                                                                   (10,0)
```

Not sure yet how to do the manual P2 probes, just want to make sure I'm on the right track.

Roxy-3D, 2017-04-25:
Holy Cow!   I don't think you know how close Delta's are to being able to run UBL!
I will help you with any question you have!   But I have to focus on the 'Data Corruption Issue' which is turning out to be just a compiler over optimization problem.  And merging my code with the main branch.
Worst case is in two weeks...   I really dig in...

But check this out...   You have the classic "Cup & Bowl" problem.  Just look at your numbers.  And I never got there...  But the whole reason UBL was designed was to make my super low quality Geeetech Delta printer work.   The numbers you just posted are way way better than what I thought my Geeetech Delta needed.

oldmcg, 2017-04-25:
I know my bed is both tilted and has cup/bowl problem -- I'm not trying to print with it like that, just get the bed probing/leveling code working with it like that.

Now working on keeping G29 P2 from trying to manually probe off the circular bed.  I have it manually probing the probe-unreachable crescent now, but it still wants to probe the grid corners outside the round bed.  I'm afraid that if I provide a zero or fixed non-nan value for these truly unreachable grid locations during P2, one of them will later get used as a nearest neighbor to compute real level data for near-the-edge-of-the-circle.  It would seem to make more sense to linear-interpolate the unreachable corner grid points from the origin through the valid grid points inside the circle than to stuff them with a fixed value.  It wouldn't fix cup/bowl on the edges but would account for slope.  Thoughts?

Roxy-3D, 2017-04-25:
> Now working on keeping G29 P2 from trying to manually probe off the circular bed.

No!  Please stop!   G29 P2 is going away.   If you give me tell tomorrow afternoon...  I'll post some code so you can do a 'Smart Fill' and all those invalid mesh locations just go away.  Smart Fill is a G29 P3 command option.   And after issuing that command you will be ready to go straight into the G29 Mesh Validation Pattern command!

oldmcg, 2017-04-25:
I've integrated this code and am testing it now.

oldmcg, 2017-04-25:
@Roxy-3D, here is the bed report after P1 then again after new P3 (skipping P2).

Note there are still some NAN mesh locations after P3 on some corners and sides.

Curious why it would interpolate from the outside-moving-in instead of inside-moving-out because then it would have more valid neighbors to work from as it populated outward.

`Bed Topography Report:
(0,10)                                                                                                                (10,10)
(-70,70)                                                                                                            (70,70)
    .        .        .        .        .        .        .        .        .        .        .

    .        .        .        .        .        .        .        .        .        .        .

    .        .        .        .      1.201    1.173    1.250      .        .        .        .

    .        .      0.818    0.709    0.849    0.983    1.019    1.079    1.149      .        .

    .      0.242    0.400    0.511    0.648    0.751    0.827    0.869    0.929    0.928      .

    .      0.029    0.167    0.323    0.403  [ 0.507]   0.629    0.663    0.718    0.733      .

    .     -0.199   -0.032    0.100    0.209    0.309    0.421    0.459    0.512    0.545      .

    .     -0.328   -0.227   -0.092   -0.041    0.101    0.181    0.248    0.298    0.370      .

    .     -0.561   -0.408   -0.317   -0.200   -0.153   -0.073    0.060    0.130    0.189      .

    .        .     -0.589   -0.509   -0.389   -0.319   -0.227   -0.140   -0.027      .        .

    .        .        .        .        .     -0.478      .        .        .        .        .
(-70,-70)                                                                                                                (70,-70)
(0,0)                                                                                                                   (10,0)

Bed Topography Report:
(0,10)                                                                                                                (10,10)
(-70,70)                                                                                                            (70,70)
    .        .        .        .        .        .        .        .        .        .        .

    .        .        .      1.741    1.552    1.363    1.481    1.599      .        .        .

    .      1.567    1.237    0.907    1.201    1.173    1.250    1.288    1.369    1.449      .

    .      0.455    0.818    0.709    0.849    0.983    1.019    1.079    1.149    1.122    1.122

    .      0.242    0.400    0.511    0.648    0.751    0.827    0.869    0.929    0.928    0.928

    .      0.029    0.167    0.323    0.403  [ 0.507]   0.629    0.663    0.718    0.733    0.748

    .     -0.199   -0.032    0.100    0.209    0.309    0.421    0.459    0.512    0.545    0.579

    .     -0.328   -0.227   -0.092   -0.041    0.101    0.181    0.248    0.298    0.370    0.441

    .     -0.561   -0.408   -0.317   -0.200   -0.153   -0.073    0.060    0.130    0.189    0.248

 -0.533   -0.561   -0.589   -0.509   -0.389   -0.319   -0.227   -0.140   -0.027    0.189    0.405

    .     -0.561   -0.589   -0.509   -0.389   -0.478   -0.227   -0.140   -0.027    0.087      .
(-70,-70)                                                                                                                (70,-70)
(0,0)                                                                                                                   (10,0)`

Roxy-3D, 2017-04-25:
You can do an extra G29 P3 to add another perimeter of extrapolated values.  It won't hurt anything to do G29 P3's until all values are 'valid'   There are going to be a lot of locations a Delta can never get to and having what ever value stored there won't matter.

> Curious why it would interpolate from the outside-moving-in instead of inside-moving-out because then it would have more valid neighbors to work from as it populated outward.

It is trying to fix the outside edge.   And it only does that if it finds two known values next to an invalid location.

Be sure to save your Mesh and then we are ready to see how badly the G26 P C O3.5 works.

oldmcg, 2017-04-25:
Ok, I had to run 5 iterations of P3 to fill in all the points.  When I ran G26 P C O3.5 it heated bed and nozzle, then very very very slowly started lowering nozzle (already at origin) until it crashed into the bed at the origin.  I did not do a G29 P5 or P6 or a G28 before G26.

oldmcg, 2017-04-25:
Currently the ADJUST_DELTA line in prepare_kinematic_move_to only does BILINEAR:

    #define ADJUST_DELTA(V) \
      if (planner.abl_enabled) { \
        const float zadj = bilinear_z_offset(V); \
        delta[A_AXIS] += zadj; \
        delta[B_AXIS] += zadj; \
        delta[C_AXIS] += zadj; \
      }

I will look for how UBL hooks into cartesian and see what I can do.

Roxy-3D, 2017-04-25:
You do not need to do a G29 P5 or G29 P6      Your mesh is filled in...

You do need to do a G28 prior to doing a G26

Roxy-3D, 2017-04-25:
> I will look for how UBL hooks into cartesian and see what I can do.

The G29 apparently can hook into the Cartesian coordinates well enough to probe the bed!   We may have problems when we try to edit the mesh with G29 P4.   But, Yes!  See if you can find what needs to shift.   Also...   Looking and probe_pt() might be good because that is being called by G29 and is working for you.

oldmcg, 2017-04-25:
Hmmm, ubl_line_to_destination is where all the cartesian mesh compensation is done by breaking the move into little pieces.  With delta/scara the moves are already broken into little pieces, so instead of dividing into ubl mesh boundary moves, each tiny delta segment should just query the proper target z-height of its destination from ubl.  No need for ubl to chop up the move, just provide z answers for each delta segment.  I'll start looking at this sometime tonight or tomorrow.

Roxy-3D, 2017-04-25:
Yes...   I think I agree.    What does the Bi-Linear leveling do when it is making a move for a Delta?   It maybe we can just use that code instead of the Cartesian version ????

oldmcg, 2017-04-25:
G29 is just doing xy moves separately from z moves.  The z moves during G29 are not mesh-compensated (mesh is disabled, not that delta would have noticed), just looking for z probe or manual z, so none of the mesh-compensated z moves are utilized during G29, at least not that I've used yet.

oldmcg, 2017-04-25:
Bilinear does just what I described -- it calls bilinear_z_offset with an XYZ destination and gets back a z adjustment for that destination.  It then adds that adjustment to the raw tower heights of the destination.  It does this for each delta segment.

Roxy-3D, 2017-04-25:
@oldmcg    Can you .ZIP up your current progress and post the file?    I'll merge your Delta stuff into RCBugFix.    No hurry...   But when it is convienient...   If we have Delta's partially working...  Other people might jump in and help.

Roxy-3D, 2017-04-26:
@oldmcg    RCBugFix now has Smart fill, Mesh Tilting (both 3-Point and Grid) along with changes to eleminate the data corruption problem that has been haunting us.    

It might be good to cross your changes over to the latest RCBugFix.

oldmcg, 2017-04-26:
Thanks for the heads up.  I will rebase to latest RCBugFix tomorrow.  I have 80% of the UBL delta movement worked out now, will get it all zipped up to you tomorrow after I rebase it.

[1 more reply]

closed 2017-11-01 by thinkyhead