MarlinFirmware/Marlin #6242: M105 Not Working?

tjay31584, 2017-04-04:
Hi all, using BugFix and  when I uncomment

>#define SHOW_TEMP_ADC_VALUES

I don't get any return after issuing M105.  In fact, I don't get ANYTHING showing in the log in Repetier, almost as if I didn't even send anything?  No acknowledgement of it being sent, nothing.

Here is a list of the log while I typed each M code numerically:

>16:40:58.686 : echo:DEBUG:INFO,ERRORS
16:40:58.686 : echo:Active Extruder: 0
16:42:01.464 : N39 M5*28
16:48:20.757 : N165 M5*36
16:48:25.701 : N168 M16*27
16:48:29.244 : N170 M14*16
16:48:31.487 : N172 M100*38
16:48:35.617 : N174 M101*33
16:48:37.704 : N176 M102*32
16:48:39.735 : N177 M103*32
16:48:41.913 : N179 M104*41
16:48:47.949 : N183 M106*46

It completely skips M105 like I didn't even type it.  Any ideas?  Is this an issue on the Repetier side (as if it's not sending it) or the Marlin side?  I don't know enough about this stuff yet.

I'm trying to calibrate my hot end temperature because my Fluke thermocouple is telling me I'm ~20 degrees C off from the actual temps.  Thanks!

Bob-the-Kuhn, 2017-04-05:
It sent the M105 command.  Repetier is just hiding it from you.

Repetier is sending M105 commands about every 3 seconds.  The results show up in the status bar on the bottom.

Most people find the M105 commands annoying so there's an option to remove it from the log you see. It defaults to not displaying the M105 commands.

Under CONFIG then PRIINTER SETTINGs you'll find the option on this page.
[image]

tjay31584, 2017-04-05:
Thanks so much!  I figured it was something I was doing wrong with the way it was hidden completely.  I appreciate the answer!

closed 2017-04-05 by tjay31584