MarlinFirmware/Marlin #6243: Firmware retraction troubles

billyd60, 2017-04-05:
Using RCbugfix from december and firmware retraction. I get small gaps on restart after a retract pretty consistently. I was able to prove this by forcing each layer to start at the same x-y and then all the gaps lined up.  The gaps do not appear anywhere else on the print. Using an E3DV6 extruder on a Taz 5. 

I've tried different retraction amounts, doesn't seem to effect the problem.

What can I do to get rid of these gaps?

I also have lin advance enabled and K=110.

Sebastianv650, 2017-04-05:
A TAZ 5, thats nice than I can confirm at least the K value of 110 :)
I would strongly recommend to move to current RCBugFix first to rule out all the small bugs since then.

Do we speak about real gaps, or is the perimeter loop closed but with a noticeable "dent" in it? Maybe a picture would help to provide a uniform understanding here.
If the gap still hapens with current RCBugFix, try to disable LIN_ADVANCE for a first test and post the configuration.h file as a attachment here. I will check and test print if necessary.

billyd60, 2017-04-05:
If I begin each layer leading into a convex curve, I get gaps in the curve. If I lead the layers into a straight I get a dent just after the beginning of the extrusion. It seems to be mitigated somewhat if I lead into a concave curve, although it may be just harder to see.

I can do pictures tonight.

billyd60, 2017-04-05:
How high should my retraction speed be? I've tried 1800 and 2100 and 2100 seems better. Should I go higher?

Sebastianv650, 2017-04-05:
I'm using FW retract with:
0.5mm retract
40mm/s retract speed
8mm/s prime speed

Maybe your prime speed is too high?

billyd60, 2017-04-05:
I will double check tonight

billyd60, 2017-04-05:
I forgot I can't control the speed from the slicer lol. G10 and G11 have no arguments. Silly. I can be really dumb sometimes. It's scary.

So for all prints I was using .5mm retract, 40 and 8. Everything else is zero

Anyway here's the pictures. First is with the retracts going into the convex curve (gaps) and the second and third is going into a straight line extrusion. (dent)

[image]
[image]
[image]

billyd60, 2017-04-05:
Here's the gcode for the print (this one is retracts leading into the straight extrusion.

[drak_split_rev_revB_wASP.zip](https://github.com/MarlinFirmware/Marlin/files/900946/drak_split_rev_revB_wASP.zip)

billyd60, 2017-04-05:
Ok this is an odd thing I just noticed. The print is actually two pieces printed side by side (in the Y direction.

At 7.4mm high in Z (.2mm  layer height) the other piece shown below suddenly stops the bad retraction behavior! Crazy (but it continues on the first piece!)

[image]

billyd60, 2017-04-05:
That was actually different gcode though with the retracts going into the convex curve so that is attached below:

[drak_split_rev_revB_wASP_retconv.zip](https://github.com/MarlinFirmware/Marlin/files/900988/drak_split_rev_revB_wASP_retconv.zip)

Sebastianv650, 2017-04-06:
Are you sure your filament spool isn't tangled, the hobbed bolt is clean and the nozzle is not partialy blocked?
Basicaly it can't be due to retraction because when it hops to the next layer or part to part, your print is always starting with the inner perimeter. Therefore if this is due to retraction, you won't see the error.
There is no retraction when the printer moves from inner to outer perimeter, check the gcode in gcode.ws.

I would recommend some further investigations:
- check filament on spool, hobbed bolt and extrusion in free air to be clean and steady
- update to latest BugFix to eleminate possible bugs since december & check again
- print without retract with and without LIN_ADVANCE enabled, compare

Sebastianv650, 2017-04-06:
If I would have the defect print, I would count to the exact layer and read to the specific gcode lines, then I would cut the print along the defect and have a look at it with my 20x magnifier. From the pictures only, I can only tell there is sparse extrusion which I only had due to tangled filament on my TAZ 5. But there are more options, and to eleminate them I recommend the list above.
I'm an engeneer, no magician :-D

Up to now I can't believe the retract theorie because an error during retraction wouldn't be visible in the outer perimeter. There are also no strange lines of gcode in the layers I checked at this locations, no extreme speeds, no too dense gcode points, no wired E moves. Therefore if it triggers a bug inside LIN_ADVANCE (which is the only option I'm afraid could cause such things beside a mechanical problem), it should be fixed in current RCBugFix. I'm not too sure what I did to advance since december..

Generaly, I know two kinds of extrusion error leading to the sparse lines we see in the pictures: A not uniform flow (due to whatever reason) in the layer itself, or a missing perimeter line (due to whatever flow reason..) below the error we see. To get the answer of that could also help.

billyd60, 2017-04-06:
Where do I get the latest rcbugfix? When I look it still says December 16. Sorry I am pretty ignorant of github. Also can I use my existing .h files or do I have to redo them from scratch?

Sebastianv650, 2017-04-06:
https://github.com/MarlinFirmware/Marlin/tree/RCBugFix
Klick clone or download, download zip.

Some of the configuration options could have slightly changed, to compare them side by side Notepad++ with the compare plugin is very useful.

billyd60, 2017-04-07:
Thanks Sebastian switching to the latest build fixed it. Wow fantastic. Thanks also for the tip on notepad++ and the compare plugin. A godsend.

It's still printing now but when it finishes I will post a photo. Big difference.

billyd60, 2017-04-07:
Here it is. One little gap near the top but this is a fairly low infill print and I think that might be the issue with that spot. But you can see the clear improvement.

I suppose I should back off on the acceleration a bit (and maybe the speed too) to cut back on the ringing.  You don't see it in most colors, but for black I should turn it way down in the future because it shows every little defect.

[image]

Sebastianv650, 2017-04-07:
Nice to hear it worked!

billyd60, 2017-04-07:
Thanks for your help!

closed 2017-04-07 by billyd60