MarlinFirmware/Marlin #6244: Extruder has stopped rotating

gunnir, 2017-04-05:
My extruder has mysteriously stopped functioning, and whilst printing will no longer feed plastic nor will it extrude on command in Pronterface...

This comes after updating the firmware and also installing an inductive proximity probe.

I've checked the motor and its wire, they are both still functioning. The stepper drivers are still functioning; have swapped and checked. I've checked the pins from the last software update, they seem to be the same in the new patch as in the old software. I've reloaded the old software and the extruder again doesn't seem to be working...

This points me to thinking that I've done something with my auto-levelling, or the way I've plugged things in on my board for my inductive proximity probe, that has caused it to stop functioning? Fairly perplexed so any help would be appreciated! I'm uploading two pictures of my board and also the config.h and config_adv files. 

I've also noticed my fans (hot-end and stepper) have stopped working (extrusion-nozzle fan still functions when running M106), but they're working fine in the forerunner firmware, so not sure if that's related? I've annotated the pictures but Preview has a nasty habit of saving them locally and not to the image, so let me know if you can't see the annotations on the JPGs
[My config.zip](https://github.com/MarlinFirmware/Marlin/files/894897/My.config.zip)

Bob-the-Kuhn, 2017-04-05:
Most of the inductive & capacitive sensors use 12V.  That can cause all kinds of havoc if wired wrong.

Do things work again when it is removed?

The only wiring diagram I've seen for these sensors use a high impedance resistor network to interface to the endstop input.  If that's what you're using then disable the internal pullup for that input.  That can make the sensor appear to be dead.

I didn't see anything obviously bad in your configuration files.  I did notice that they are not for the latest version of Marlin.  If you use config files from an older version you can have really strange behavior.

What are the two versions of Marlin that you switched between?

Is the extruder stepper the only stepper that isn't working?  If yes then swap the cable AT THE CONTROLLER with one of the other steppers and see what happens.  I had a socket go bad on my controller.  Drove me crazy trying to figure it out.

Going back to the old firmware and the fans work again but the extruder still doesn't.

The fan issue could be because of a mismatch between the config files and the firmware you're using.

What controller are you running?  I'd like to see if there could be a blown fuse causing the extruder issue.

gunnir, 2017-04-11:
@Bob-the-Kuhn Thanks for the advice Bob! I've managed to resolve both issues. First, the extruder motor will now rotate. Short term solution: I switched the pins for E0 and E1 (I have ramps 1.4) in pins.h and plugged the motor and stepper driver into the spare connector, it now works! Long term solution: new board. I tried switching/resoldering the spare Z connector with E0 but still not working, so some other component on the board must have fried... :(

As for the fans, I managed to resolve this by manually adding their pins to the config.h (see attached pictures). Got the hint from here: http://forums.reprap.org/read.php?146,742898. Previously in my older firmware they were in an overarching pin.h file which contained multiple definitions for differing boards. The newer firmware has individual header files for each board; somewhere along the line it must have got lost!

So... Fans are working, extruder working too, thank you! But something @Bob-the-Kuhn mentioned made me a little worried. I removed the newly installed inductive proximity probe, because as you correctly highlighted, it requires a voltage different to the board and was potentially playing havoc. Hopefully that's not the case, as now the board is working. I've uploaded a picture of the wiring of my inductive proximity probe (NPN Normal Open LJ18A3-8-Z/BX model). It seemed to function post install, and I could run G29. Repeat tests of running G29 indicated a low standard deviation around the data points too - so, accurate too!

If you notice anything whack with the wiring, my new pins.h or config.h, let me know!
[New config.zip](https://github.com/MarlinFirmware/Marlin/files/911799/New.config.zip)

Bob-the-Kuhn, 2017-04-11:
Glad to hear things are working.

I looked at your interface circuit.  If the internal pullup is on the low side of it's range then your logic low signal could be a little high. To be on the safe side you should have the internal pullup disabled.  The following will do that.  I've just commented out two lines.
```
// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
// fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
#define ENDSTOPPULLUP_XMAX
#define ENDSTOPPULLUP_YMAX
#define ENDSTOPPULLUP_ZMAX
#define ENDSTOPPULLUP_XMIN
#define ENDSTOPPULLUP_YMIN
//#define ENDSTOPPULLUP_ZMIN
#define ENDSTOPPULLUP_ZMIN_PROBE
#endif
```

Some inductive sensors will start to trigger closer to the bed when you're near the edge.  I suggest doing a few paper tests near the edges and compare that to ones near the center.

closed 2017-04-11 by thinkyhead