MarlinFirmware/Marlin #6246: Marlin Github Master Class • Livestream •

thinkyhead, 2017-04-05:
I'll be hosting a "master class" on using git in conjunction with the Marlin Firmware project. I propose to broadcast starting on Wednesday, April 5 starting at 9:30pm CDT (GMT-5), for as long as there are useful things to demonstrate.

emartinez167, 2017-04-05:
Will you record it? Cannot attend at that time.

Regards/Saludos,

Ernesto.

thinkyhead, 2017-04-05:
Yes, it will be streamed on YouTube, and published immediately afterwards.

Tannoo, 2017-04-05:
Hopfully, you can show more than just a few commits to squash.And how to open and resolve the conflicts that arise from an attempted squash.
How to rese from that.
Also how to avoid local repo corruption when something doesn't go right.

emartinez167, 2017-04-05:
Cool! Thanks a lot!

thinkyhead, 2017-04-05:
@Tannoo All of the above.

apballard, 2017-04-05:
@thinkyhead very excited about this! Thanks!!

Roxy-3D, 2017-04-05:
I think the Git developers should do a 5 minute video explaining why they developed something that is 100% non-intuitive.

thinkyhead, 2017-04-05:
The desktop app basic interface simply isn't for managers of projects.

Tannoo, 2017-04-05:
But, you know...
Gnome is better at a gui for linux than git desktop is for git.
Hell, Windows 3.x was a better gui than git desktop. 
There's always room for improvement. Shunning it off doesn't make it better.
Just sayin'.

thinkyhead, 2017-04-05:
It is early days for that app. I'm sure it could do better. In fact there are other front-ends for git that are more full-featured – "SmartGitHg" for example. But they come with a learning curve for their interface and method of representing the history too.

Once you have a good picture of "what a branch is" it starts to fall into place. On thing I have found is that it's absolutely imperative to have some way to run `git rebase -i`.

Roxy-3D, 2017-04-05:
> But, you know...
> Gnome is better at a gui for linux than git desktop is for git.
> Hell, Windows 3.x was a better gui than git desktop.
> There's always room for improvement. Shunning it off doesn't make it better.
> Just sayin'.

From a slightly different perspective...  Supposedly Git helps you develop your software and make it everything it can be.   How 'bout the Git Developers use their software to make GitHub-DeskTop everything it can be???   And at the top of the list is 'Easy and Obvious to Use'.

thinkyhead, 2017-04-05:
Insofar as what it does, it's kind of easy and obvious. It only allows these basic operations:
- Check out
- Sync / Push / Pull
- Commit selected file(s)
- …and partial-commit selected lines
- Undo last commit
- Diff two commits
- Rename branch
- Merge from your `origin/[default]` branch
- Merge from a branch you select (upstream or local)
- Delete `origin/[current]` branch
- Submit a Pull Request 
- Go to Pull Request

No interactive rebase possible, which means you have to use another means, such as the CLI.

Roxy-3D, 2017-04-05:
> But that's the same with all version control systems. A human has to decide what's best.

I'm going to watch a good sci-fi movie...  But Git has been giving a bunch of us 'Conflicts' when there is nothing even different.   I went so far as to delete everything in that area of the file and try to check it in.   No Go!   It's a conflict.     Come On....    And listen...  It is phylisophically wrong for it to edit my source files and put all kinds of things like <<<<<<<<HEAD in it.   Leave my source files alone!

Tannoo, 2017-04-05:
Yes. I agree. When rebasing (merging commits from upstream), sometimes there are conflicts. Git deskstop will show me what files are conflicting.
 Maybe I just missed them when trying to squash, but I didn't see the files it was complaining about.

thinkyhead, 2017-04-05:
Some git front-ends exist that will hide the `<<<<HEAD`, `=====`, and `>>>> comment` markers. But generally, we are expected to resolve commits using something which is a text editor at its base, and these markings clearly indicate the positions of the conflicts. It's important to be able to follow the git logic as to why some things are included in these conflict blocks. Sometimes they are not part of the changes, but only part of the context.

In the live-stream I will perform several examples of conflict resolution and explain what's going on at each point.

It is very much a process requiring attention to detail. When we run up against many conflicts that are hard to resolve, that can be annoying. It means we have to look carefully at all of them. Corrections in indentation are the most annoying, because then it's hard to resolve the ordering. In those cases, you need abort the merge, use a diff tool to decide what the up-to-date code in those blocks should be, and then when you re-do the merge/rebase and hit that conflict again, paste in the block more easily resolved elsewhere.

Of course under some circumstances, the preponderance of conflicts is too many to deal with. In that case it's easier to "start over" and rebuild the changes on the latest code. I will also go over that process in detail.

thinkyhead, 2017-04-05:
If anyone is in the mood for a teaser, I made a [56 minute intro video](https://www.youtube.com/watch?v=L2YvUr1dbCk&feature=youtu.be) today. Sorry there are clicks and typing noises. (The laptop microphone is sensitive.)

apballard, 2017-04-05:
@Roxy-3D <<<<<<<<HEAD also drives me crazy. A couple days ago I learnt about git merge tool and integrated winmerge. So far I'm lovin it. Makes things easier for me.

Bob-the-Kuhn, 2017-04-05:
@thinkyhead - the class notes you're preparing - after I get them all bound up, will they fit on my kitchen table?  ;-)

teemuatlut, 2017-04-05:
Will you be focusing more on the git command line or the GUI?

If I can suggest a few brief topics
Comparing files between branches or commits with "diff branch1..branch2 -- file.h"
Comparing logs with log branch1..branch2.
Setting up difftool and mergetool
Undo changes with checkout -- * and reset HEAD *
Commit --amend
Push -f

thinkyhead, 2017-04-06:
The stream is at: https://www.youtube.com/watch?v=b-2ngF9e6oI

psavva, 2017-04-07:
I use TortoiseSVN at work, and I love it!
i know they have released ToirtiseGIT but haven't given it a try yet....  As soon as I get back from Mexico in 2 weeks, I'll give it a try and let you know if it makes life easy :)

Roxy-3D, 2017-04-07:
> i know they have released ToirtiseGIT but haven't given it a try yet.... As soon as I get back from Mexico in 2 weeks, I'll give it a try and let you know if it makes life easy :)

I'm open to suggestions.  If you think ToirtiseGIT is better than GitHub-Desktop, I'll switch.  GitHub-Desktop is not at all intuitive.

thinkyhead, 2017-04-11:
I hope you found the linked videos informative, even though they are a bit scattered. I recently got a newer laptop that probably has a better microphone. And I'm ordering a condenser microphone and headphones so I can do these videos properly and be heard.

closed 2017-04-11 by thinkyhead