MarlinFirmware/Marlin #6247: Home offset

angellindo, 2017-04-05:
I'm having an issue setting the home offset.

If I try to set it with M428 after positioning with G1  X13 Y30 Z2 I'm  getting **MSG_ERR_M428_TOO_FAR**

Then I try setting the offset with M206 X13 Y30 Z2 but that is setting the endstops positions to those numbers.

So if I try to mesh level with G29 S1 it goes to x23 y40 because of the mesh offset padding which is ok, but it actually just moved to x10 y10 so the nozzle is not in the print area.

How do you correctly set the home offset? or is there a bug with m428 and m206 for home offsets?

I tried with RC8 and with RCBugFix branches.

Here's my config file:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/895407/Configuration.txt)

wookie666, 2017-04-05:
Cant you set your M206 X-13 Y-30? This will mean that when you want it to go to x0 y0 it will move 13mm in x and 30 mm in y
Is this not correct.

angellindo, 2017-04-05:
@wookie666 yeah that worked thx but in the end had to change

```c++
#define X_MIN_POS -13
#define Y_MIN_POS -30
#define Z_MIN_POS -2
```

Because manual bed level required those vars and was not using offset of manual_[xyz]_pos

angellindo, 2017-04-05:
@wookie666  also had to do this because of the negative numbers

```c++
  #define MESH_MIN_X (0 + MESH_INSET) //#define MESH_MIN_X (X_MIN_POS + MESH_INSET)
  #define MESH_MAX_X (X_MAX_POS - (MESH_INSET))
  #define MESH_MIN_Y (0 + MESH_INSET) //#define MESH_MIN_Y (Y_MIN_POS + MESH_INSET)
  #define MESH_MAX_Y (Y_MAX_POS - (MESH_INSET))
```

closed 2017-10-31 by thinkyhead