MarlinFirmware/Marlin #6249: Some SD Cards not working (class 10)

spilz87, 2017-04-05:
hello,
I spend a lot of time to understand why some SD cards work well and some other don't.
I took 2 SD cards 8GB SDHC, formated with the same computer withe the same parameters (FAT32).
the first one is a class 4, the second one is a class 10.
I put the same gcode file.
both are hardware detected : LCD show "SD card insered" (or something like this (mine is in french))
but when I bo in the menu : 
- card 1 (class 4) : "print form sd card", that works :)
- card 2 (class 10) : "No card", the card is not reconized :(

it seems that class 10 cards use a high speed bus and class2/4/6 cards use a normal speed bus.

but the problem is now it's difficult to find somethnig else than class 10 :(

Does someone has the same problem ?
Does someone has an other explaination about it ? or other informations ?

how to update sd card driver on marlin to use class 10 ?
or maybe it's an hardware issue with level converter not speed emought for high speed ?

thanks for your help

BillyBob54, 2017-04-05:
I wonder if this is related to the same issue discribed in #6128 ?  I have experienced the same "No Card" problem from time-to -time, but have always been able to correct it by removing/reinserting the SD card. I am not entirely certain of the calss SD cards I have- one  16G and pretty new, gives me problems( see #6128 )the other a 128M older card that works fine)

spilz87, 2017-04-05:
Actually that makes sense :
Old one (128M) is class 2 or 4 or 6.
New one (16G) is maybe class 10
(You can check it, it the value in a cercle on the card)
Let me know

BillyBob54, 2017-04-05:
Yep -16G is class 10.  Odd thing is, I am printing from that SD card as I write this.  LCD recognized it no problem.  The only time I have any real issues is if I try to use the 16G SD (class 10) AND connect via USB with Repetier Host- everything goes wonkie and marlin goes into a constant rest mode.  As everything (at least for me) is working OK, it appears that Marlin is playing nice with the 16G, so I suspect that there must be some sort of weird interaction between Repetier Host and Marlin. By the way, I am using the 2 April RCBugFix, Ramps 1.4 and Repetier Host 1.6.2 and arduino IDE 1.8.2

closed 2017-10-31 by thinkyhead