MarlinFirmware/Marlin #6250: Reprap full graphic controller Blue screen

tjp2579, 2017-04-05:
I have set marlin to reprap full graphic controller and I still have a blue screen anyone help would be greatly appreciated.

tjp2579, 2017-04-05:
Ok I got a display now but it just says Marlin on the face of my LCD shouldn't I get a control screen

Tannoo, 2017-04-05:
Make sure your 5vdc is good.

tjp2579, 2017-04-05:
what my power supply for the lcd

Tannoo, 2017-04-05:
Well, the 5v comes from a volage regulator on your control board.

What motherboard do you have?

Tannoo, 2017-04-05:
Your endstops are a good place to check for 5vdc.

tjp2579, 2017-04-05:
ya im getting 5vdc to my endstop

Bob-the-Kuhn, 2017-04-05:
When using the USB port, is the controller responsive?  

I'm wondering if the controller is hung which might explain why the LCD is stuck at the startup screen.

If the controller is hung try changing TEMP_SENSOR_0 to 998.  The controller will hang if a sensor reading is way too high or way too low.

tjp2579, 2017-04-06:
Thanks for all the help I got it now

Bob-the-Kuhn, 2017-04-06:
What did it take to get it running properly?

tjp2579, 2017-04-06:
Me to send it back and but a day Vinci pro 1.0 lol

closed 2017-10-31 by thinkyhead