MarlinFirmware/Marlin #6252: BLTouch Menu items not always available

mperdue, 2017-04-05:
The menu items for the BLTouch probe need to be in the menu all the time if a BLTouch probe is enabled. Sometimes Reset is near the top of the menu and Test is near the bottom. However, whenever I do a test the reset is removed from the menu so I have to reboot to stop the test.  It wouldn't bother me if there was a BLTouch item that was in the main menu and then a sub-menu with reset, test, deploy and retract available. That would allow you to test all the functionality using the LCD and dial.

Roxy-3D, 2017-04-05:
That might be a worth while option.   It would helpful to anybody with a BL-Touch.

mperdue, 2017-04-06:
It would be helpful to have the commands grouped in one place. Currently, there is sometimes a menu item to reset the BLTouch near the top of the menu. This seems to be when the printer thinks the BLTouch is in an error state and needs to be reset. Near the bottom of the menu is an entry to do a self test. One problem is that if you select the self test, the printer isn't in error state so the reset entry goes away. At that point there is no way to end the self test other that rebooting or sending the command from a connected computer. The logic needs to be sorted to just add a sub menu with all four commands associated with the BLTouch available all the time.

GrevanGER, 2017-04-16:
Got my BlTouch up and running just yesterday. 
I am using a Ramps 1.4 on an Arduino Mega 2560. 

From what I can say so far, there is no such problem with 'jumping' menus for the BLTouch.
The only time when the Reset option shows up in the main menu is, when I hold on to the pin until the BLTouch goes in to error state. And as far as I can tell, this was intended by the guy who wrote the code for that option. 

But there was just something missing to make my printer independed in terms of, "Not having to deploy and stow via G Code". So I went ahead and added a few lines of code myself in order to use the allready preconditioned "DEPLOY" and "STOW" states from the Conditionals_LCD.h. 

Code added to "language_en.h" at line 378:
`#ifndef MSG_BLTOUCH_DEPLOY
  #define MSG_BLTOUCH_DEPLOY                  "BLTouch Deploy"
#endif 
#ifndef MSG_BLTOUCH_STOW
 #define MSG_BLTOUCH_STOW                     "BLTouch Stow"
#endif`

Code added to "ultralcd.cpp" at line 1279:
`      MENU_ITEM(gcode, MSG_BLTOUCH_DEPLOY, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_DEPLOY)));
      MENU_ITEM(gcode, MSG_BLTOUCH_STOW, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_STOW)));`

mperdue, 2017-04-16:
The problem is when you select "BLTouch Self Test" from the "Prepare" menu. This will start the self test but then there's no way to stop it. You have to cycle power to stop the self test. I'll add your changes and see where that leaves things.

mperdue, 2017-04-16:
After looking at your code, it's pretty much exactly what I had done myself. It will work for now, but I think that ultimately it would be better to have a submenu with the BLTouch commands in it. The reset can still popup in the existing menu as needed for clearing an error state but also having reset available in a submenu along with deploy, stow and selftest would, IMHO, be a better solution. I've looked briefly at the way the preheat submenus are implemented but I'm not sure I'm following it all yet.

GrevanGER, 2017-04-17:
Well, I did take a quick look in to how the menu work. 
Got all the options I have thrown in to the Prepare menu now in a submenu, located in the main menu. 
[image]
[image]

This is how I did it: 

in ultrslcd.cpp add the following code at line 125:
`  #if ENABLED(BLTOUCH)
  static void bltouch_menu();
  #endif`

In ultralcd.cpp add the following code at line 585:
`    #if ENABLED(BLTOUCH)
        MENU_ITEM(submenu, MSG_BLTOUCH, bltouch_menu);
    #endif // BLTouch `

In ultralcd.cpp add the following code at line 725:
`  /**
   *
   * "BLTouch" submenu
   *
   */
  static void bltouch_menu() {
   START_MENU();
   //
   // ^ Main
   //
    MENU_BACK(MSG_MAIN);
    
    MENU_ITEM(gcode, MSG_BLTOUCH_SELFTEST, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_SELFTEST)));
    MENU_ITEM(gcode, MSG_BLTOUCH_DEPLOY, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_DEPLOY)));
    MENU_ITEM(gcode, MSG_BLTOUCH_STOW, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_STOW)));
   END_MENU();
    }`

In language_en.h add the following code at line 375:
`#ifndef MSG_BLTOUCH
  #define MSG_BLTOUCH                         "BlTouch"
#endif`

After compiling, arduino pointed a "endif without if" out in the ultralcd.cpp, right after the following line. 
#endif // ULTIPANEL
So I deleted the marked solo endif and that made the changes running. 

I am not 100% sure that I have listed everything I have done here. That's the problem with working down the road I guess. ;) 
But if you are running in to any problems, please let me know and I will try to help.

Roxy-3D, 2017-04-17:
@Bob-the-Kuhn    What do you think?  This looks very helpful to people with BL-Touch probes.

Bob-the-Kuhn, 2017-04-17:
I like the idea & the layout.

We should also have reset in the submenu.

mperdue, 2017-04-17:
I agree with adding reset back but otherwise you've done exactly what I was hoping to see.

Roxy-3D, 2017-04-17:
> We should also have reset in the submenu.

Yes!  Agreed!   And if we were to do that, perhaps we should consider putting Reset at the top of the list just because that is very likely to be the most used function in the sub-menu.

GrevanGER, 2017-04-18:
Well, now that I think about, Reset is a nice feature once you get your probe in to error mode. ^^
Didn't think about that when I just threw out the Reset function from the main menu. 
But hey, in the end I am just a copy cat who was 'coding' on the shoulders of others.

thinkyhead, 2017-04-26:
Looks good. Guess I'll take care of this one.

thinkyhead, 2017-04-28:
Added to #6491

closed 2017-04-28 by thinkyhead