MarlinFirmware/Marlin #6257: Thermistor Problems - Workaround Possible?

tjay31584, 2017-04-06:
Hi all,

To start, this problem is not the fault of Marlin firmware.  I am simply wondering if there is a way to tell the firmware which thermistor to look at for which use.  Specifically, JP7 shows thermistor 0, thermistor 1, and thermistor 2 as inputs.  Typically, thermistor 0 is hot end 1, thermistor 1 is the bed, and thermistor 2 is the second hot end if applicable.  I want to define thermistor 2 to be my hot end 1, not thermistor 0.  Here's why:

Today when I went to preheat for printing I got a minimum temperature alarm.  I looked at my LCD and saw 4 degrees for my hot end.  I assumed something was wrong with the thermistor or wiring, so I checked them both out.  Everything appears fine.  The bed was reading normally, and since both are using the same table, I swapped the leads at the board.

The thermistor for the bed was reading 21 degrees (room temp) when plugged into the bed port.  When switched to the hot end port, I get 4 degrees.

The thermistor for the hot end was reading 4 degrees, but when switched to the bed port, I get 21 degrees as expected.

Basically, this tells me there is nothing wrong with either thermistor, since they both read the proper temperature when plugged into the bed port.  So, I checked the power out from both sets of thermistor pins.  The hot end measures 4.970 VDC, the bed measures 4.999 VDC.

I am thinking this voltage difference is the cause of the problem, since it's basing it's resistance on the return voltage.  Agreed?

So, if I can define thermistor input 2 to be my hot end, I should be able to connect it there and see if my theory holds true (that my thermistor 0 input is "bad").  Additionally, I would be able to continue using my printer until I can get it repaired.

Tannoo, 2017-04-06:
In the pin_(board).h file, you can assign what Arduino pin each thermistor is on.

Bob-the-Kuhn, 2017-04-06:
TEMP_0_PIN is used for the first extruder/heater, TEMP_1_PIN is used for the second extruder/heater, ...

TEMP_BED_PIN is used for the bed.

These can be assigned to any convenient analog input.

Open the pins_YOUR_BOARD.h file and find the line with TEMP_0_PIN in it.  It'll say something like **#define TEMP_0_PIN 3** where 3 is the analog input 3.  Analog inputs are numbered from 0 through 15 on a 2560.  Just change it to any convenient analog input.  

There are no changes to any other files.

The one gotcha is you need to pick one that has a 4.7k pullup on it.  If one isn't available then you'll need to connect one to the analog in.

tjay31584, 2017-04-06:
Thanks!  I am still fairly new to all this stuff, but that makes perfect sense.

I did some testing with my meter and it seems one of my 4.7K resistors has gone bad?  I am getting less voltage out of one than the other two that are for the three thermistor circuits, and that's the one that goes to T0.  Input voltage is identical on all of them, so I suspect that's my problem.  I'm going to try and swap it to the other pins and see if that saves me.  Thanks again!

tjay31584, 2017-04-06:
I was able to change the pins in the pin file and got it working.  I think I need to replace this board possibly.  It has been doing some weird things, and it makes me wonder if there's more than one problem with it.  Thanks again!

closed 2017-04-06 by tjay31584