MarlinFirmware/Marlin #6259: UBL G29 P1 probing outside the bed on Y axis

akselii, 2017-04-06:
Hello,
I got my  Y_MIN_POS -30,  and Y_MAX_POS 230 and when doing G29 W, im getting mesh min y -30 and mesh max y as 230, is this right?

My print area is as correct as it can be, after g28 and moving the nozzle around to the corners it matches up perfectly, however, when doing G29 P1 it tries to probe under the y min, so that the probe is outside of the bed.

 If i set the UBL_MESH_MIN_Y 30 And UBL_MESH_MAX_Y -30, that seems to give me the right y max and min when checking G29 W. With that done, i can get a successful probe.
Is this intended operation? 

Also, the probe accurately probes on the X axis correctly, as seen when it probes the max x of the bed, where the nozzle would be while printing.

I think i can make a workable mesh with the UBL_MESH_MIN and MAX values set but this just doesnt seem right to me

Also, some of this info might be inaccurate thanks to me being at work and not being able to check my configs.

Any ideas?

Bob-the-Kuhn, 2017-04-06:
The travel area of the nozzle is set by these defines:
```
  #define X_MIN_POS -20       // G28 sets X position to this value after homing (- direction)
  #define Y_MIN_POS -20        // G28 sets Y position to this value after homing (- direction)
  #define Z_MIN_POS 0        // G28 sets Z position to this value after homing (- direction)
  #define X_MAX_POS 415     // G28 sets Z position to this value after homing (+ direction)
  #define Y_MAX_POS 520     // G28 sets Z position to this value after homing (+ direction)
  #define Z_MAX_POS 455     // G28 sets Z position to this value after homing (+ direction)
```

The location & size of the bed are set by these defines.  If not defined then the above defines are used.
```
    #define UBL_MESH_MIN_X 0
    #define UBL_MESH_MAX_X 395
    #define UBL_MESH_MIN_Y 0
    #define UBL_MESH_MAX_Y 500
```
The travel area is used when determining if a point on the bed is probeable.

Roxy-3D, 2017-04-06:
> If i set the UBL_MESH_MIN_Y 30 And UBL_MESH_MAX_Y -30, 

Max can't be less than Min !!!!

akselii, 2017-04-06:
I'm at home and trying again.

I'm sorry about the 30 and -30, I meant I add or subtract 30 to get the correct mesh size
And wondering why UBL tries to probe outside the bed when it knows the probe offsets

These are my UBL_MESH_MIN and MAXses: 

#define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
#define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
#define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET + 30)
#define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET) - 30)

And these are my XYZ MIN MAX:

#define X_MIN_POS 0
#define Y_MIN_POS -30
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 230
#define Z_MAX_POS 200

With these values G29 W gives me this:

16:31:25.863 : X:75.00 Y:103.00 Z:10.30 E:0.00 Count X:6000 Y:8240 Z:16480
16:31:25.878 : Unified Bed Leveling System Version 1.00 Inactive.
16:31:25.931 : No Mesh Loaded.
16:31:25.994 : g29_correction_fade_height : 10.00
16:31:25.994 : z_offset: 0.000000
16:31:26.548 : X-Axis Mesh Points at: 1.0  23.0  45.0  67.0  89.0  111.0  133.0  155.0  177.0  199.0
16:31:27.063 : Y-Axis Mesh Points at: 1.0  23.0  45.0  67.0  89.0  111.0  133.0  155.0  177.0  199.0
16:31:27.063 : Kill pin on :41  state:1
16:31:27.115 : ubl_state_at_invocation :0
16:31:27.115 : ubl_state_recursion_chk :0
16:31:27.178 : Free EEPROM space starts at: 0x0020
16:31:27.178 : end of EEPROM              : 0x0FFF
16:31:27.232 : sizeof(ubl) :  1
16:31:27.232 : z_value[][] size: 400
16:31:27.278 : EEPROM free for UBL: 0x0FDF
16:31:27.332 : EEPROM can hold 10 meshes.
16:31:27.394 : sizeof(ubl.state) : 64
16:31:27.394 : GRID_MAX_POINTS_X  10
16:31:27.448 : GRID_MAX_POINTS_Y  10
16:31:27.448 : UBL_MESH_MIN_X         1
16:31:27.495 : UBL_MESH_MIN_Y         1
16:31:27.495 : UBL_MESH_MAX_X         199
16:31:27.563 : UBL_MESH_MAX_Y         199
16:31:27.563 : MESH_X_DIST        22.000000
16:31:27.563 : MESH_Y_DIST        22.000000
16:31:27.613 : Unified Bed Leveling sanity checks passed.

Am I good to go?

Thanks!

Roxy-3D, 2017-04-06:
If the probe is trying to go off the bed...  You have your Bed dimensions and/or X_PROBE_OFFSET_FROM_EXTRUDER or Y_PROBE_OFFSET_FROM_EXTRUDER in error.    Just to see how you are centered, you can set UBL_MESH_INSET to 50  and start lowering it until the probe wants to go off the bed.    That will help you figure out what number you have wrong.

akselii, 2017-04-06:
My probe offsets are pretty spot on, If I do a home it centers the probe at center of the bed, as does G29 P1 M, it shows me where its probing and the probe does appear to be at the right place

Roxy-3D, 2017-04-06:
What is the coordinate of where the probe tries to leave the bed?

akselii, 2017-04-06:
It was 30mm on the -Y, not sure of the X. It seemed like it ignored my Y_MIN_POS -30 and just went on going. I think the mesh bed size was min y of -30 and max y of 230, so what I wrote on x y min max

Roxy-3D, 2017-04-06:
Can we set Y_MIN_POS to 0 and see what it does?

akselii, 2017-04-06:
Just tried it out, it goes over the min, even the whole nozzle is out of bed. 
These settings:

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 230
#define Z_MAX_POS 200

#define UBL_MESH_INSET 1

#if ENABLED(AUTO_BED_LEVELING_UBL)
  #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET) // +30
  #define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET)) // -30
#endif  

They don't work, my beds 0Y position is not at the endstop, its 30cm away from it

These seem to give me a working mesh:

#define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET + 30)
#define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET)- 30 )

#define Y_MIN_POS -30
#define Y_MAX_POS 230

Roxy-3D, 2017-04-06:
With the symmetrical numbers, it would seem you should use 
```cpp
#define UBL_MESH_INSET 30
#define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
#define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET))
```

akselii, 2017-04-06:
Is it a problem if I set the UBL_MESH_MIN_Y and MAX to 30?

Roxy-3D, 2017-04-06:
Yes.   The mesh exists between the two.  Setting them equal means the mesh has no size.

akselii, 2017-04-06:
Again, sorry. I meant would these be a problem: 

#define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET + 30)
#define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET)- 30 )

Roxy-3D, 2017-04-06:
No...   You should be able to adjust the values in Configuration_adv.h    But...  Don't do that just yet.
Lets start with this:  And depending upon where things are...  It will be much more clear and easy to shift the numbers.    Once everything is rational...  We can shift the numbers.   But first, lets get everything on the bed.
```cpp
#define UBL_MESH_INSET 30
#define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
#define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET))
#define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
#define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
```

akselii, 2017-04-06:
It seems that i was getting a mesh going when i was just probing with the +-30 on the UBL_MESH_MIN/MAX, with the numbers being from 0,0 to 199, 199.

I am going to retry with the inset 30 option right now.

akselii, 2017-04-06:
So G29 P1 was able to get a mesh done, its missing 30mm from the left and right now tho
https://pastebin.com/raw/iwtRETE2

Roxy-3D, 2017-04-06:
Your bed has a pretty significant tilt from front to back.   Left to Right is pretty good.   If you can, you should consider physically leveling your bed from front to back.

You can change
```cpp
#define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET-25)
#define UBL_MESH_MAX_X (X_MAX_POS - UBL_MESH_INSET+25)
```

akselii, 2017-04-06:
I really wonder why i shouldnt just do this thing:

#define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET + 30)
#define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET) - 30 )

It seemed to get the whole bed the probe could reach and leave the rest as dots on the left side

Roxy-3D, 2017-04-06:
You said "its missing 30mm from the left and right now tho"   That is typically the X-Axis.  

And I bumped the size down just to make sure you would not go off the bed.    It's your printer...  You are in a better situation to make the changes!   Do what makes sense!

When you get a G26 to run, please take a picture of the Validation Pattern, and we can continue.

jrmann1999, 2017-04-14:
It appears the code is not taking into account negative values for the offset when using UBL, I'm having the same issue with my Y_MIN set to -30.  G1 and G28 respect it (and stop before going off the bed).  G29 W prints a negative Y value which appears to be outside the bed.

jrmann1999, 2017-04-14:
Actually, it looks like it's double dipping.  I again have my Y_MIN_POS at -30, and just set the UBL_MESH_MIN_Y hard coded to 1 and it was spot on.  Doesn't appear it needs to calculate based on board dimensions, just needs to give the MESH_INSET offset and work from there.

akselii, 2017-04-14:
I recently broke my sensor mount due to it sticking to some warping ABS so i havent been able to play around with UBL

I was going to figure out how to deal with the negative values from the MIN_POS values but if what you (jrmann1999) are saying is right, it should be enough to just remove the MIN/MAX_POS from UBL_MESH_MIN/MAX?

This was the original issue i had with is, bug or intended?

jrmann1999, 2017-04-14:
I'm printing right now with UBL (working fantastic BTW) after having removed MIN/MAX POS from RCBugFix train of firmware.  I did a P1 level prior to it, and it stayed within bounds for everything.

closed 2017-11-01 by thinkyhead