MarlinFirmware/Marlin #6262: Should BL-Touch reset error state in middle of probing?

Roxy-3D, 2017-04-06:
@Bob-the-Kuhn    I was doing a grid probe session and reaching under the BL-Touch to trigger it.  And I put it in an error state by bumping the pin when it was trying to deploy.

Here is the question:   Should we try to reset the error state of the BL-Touch during the middle of a probing session?   I think the answer is "Yes!" but there may be a reason why that is a bad idea.   With a BL-Touch it doesn't matter so much because it is 'fail safe' and starts reporting triggered when in the error state.  But some of the knock off clones do not do that and if the probe goes into an error state, it is likely to drive the extruder into the bed.

Anyway...   I was surprised to see the BL-Touch not recover out of the error state until the next G28.   We know the formula to fix that.   And we can probably do it with no delay or performance impact in the normal case where the probe has not entered the error state.   Should we put some checks down in the run_z_probe() function???

Bob-the-Kuhn, 2017-04-07:
With the code I added today G29 bilinear and G29 P1 act about the same.

Both will skip to the next probe location if the BLTouch is in error and stays in error when the probe starts.

Looks like G29 bilinear reports the Z_deploy height while G29 P1 reports NAN.

Both will continue even though the host screen says that Stop has been called.  Is that the way it should be?

Roxy-3D, 2017-04-07:
I don't know that stop() or kill() makes sense.    I'm not sure what the 'right' answer is.    You don't want to operate with bad data.   So maybe that is the only option.    But it would be nice if the error state automatically got reset.

Technically...  I think I would argue NAN is more appropriate.   But I'm not sure it makes sense to change the bi-linear to be the same.

Bob-the-Kuhn, 2017-04-07:
As best I can tell the STOPPED state just means Marlin is no longer accepting G0, G1, G2 & G3 commands over the USB channel.

Duhhhh - All I had to do was read the STOP() header "Turn off heaters and stop the print in progress. After a stop the machine may be resumed with M999"

SOMETIMES I AMAZE MYSELF   👎

thinkyhead, 2017-04-11:
I recently learned the cool factoid that the `isnan` macro is this:
```cpp
#define isnan(X) (X != X)
```
because `NAN` doesn't equal itself. 🏋️

Roxy-3D, 2017-04-12:
> I recently learned the cool factoid that the isnan macro is (sometimes) this:
> 
> #define isnan(X) (X != X)
> because NAN doesn't equal itself. :weight_lifting_man:

I'm not an expert at IEEE Floating point.   But NAN is very special.  It is not a single value.   In our case we have 4 byte floats.  NAN is carefully chosen so that if the upper 2 or 3 bits are set, it is a NAN.  But the rest of the mantissa and exponent continue to function.   You can do floating point math involving a NAN and the numbers continue to propagate correctly (albeit...  They are now branded as 'NAN').    There are special cases where even though you can't trust the final number (because it is NAN), you still want to examine the exponent or mantissa for extra information.   And that is why:
```cpp
 #define isnan(X) (X != X)
```

closed 2017-10-31 by thinkyhead