MarlinFirmware/Marlin #6263: Understanding E steps/mm and E# in the firmware and gcode

kcheeeung, 2017-04-06:
Hi everybody again. I wanted to confirm with you all about what these concepts are. I wanted to use this knowledge to perform careful calculations because this I will use the use extruder motors to push down on a syringe, which will extrude some materials.

If I had for example:
```

/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 160, 160, 3200, 100 }

and 

G0 X00 Y0 Z0 **E00**
G0 X10 Y0 Z0 **E30**
```
The printer will extrude **10mm of filament** starting from X0 to X10.
And therefore, based on the E steps/mm, the extruder motor will spin/turn for:
**30mm x (100steps/mm) = 3000 steps**

Is this assumption correct? I will eventually use this concept and turn this into a way to turn extruder motor ### of steps or ### angle.

Bob-the-Kuhn, 2017-04-06:
You are correct.

kcheeeung, 2017-04-09:
Thanks again Bob.

What is the meaning of feedrates?
```

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 25, 25, 10, 5 }
```
Line 1 G1 X0 Y0 F1000
Line 2 G1 X10 Y0 E0
Line 3 G1 X20 Y0 E10

Is it that
From line 1 to 2: It will take **(60secs/1000mm)*(10mm) = 0.6** secs to travel from X0 to X10.
From line 2 to 3: It will take 0.6 secs to travel X10 to X20. The extrusion for X10 to X20 will be (E10-E0) **10mm filament*(1 secs/5mm filament) = 2 secs.**

Does the E# stepper automatically use the MAX_FEEDRATE of 5mm/s? And for this theoretical scenario, the extrusion would not be right because the XY movement is faster than the E# motor movement?

Bob-the-Kuhn, 2017-04-09:
The planner routine always makes sure that movements of the X, Y, Z & E axis all finish simultaneously.

In your case it's best to think of everything in units.  

The line 3 command says to move the X axis by 10 units and move the Extruder material by 10 units, both at a feed rate of 1000 units per minute.  The planner routine would first look to see if the 1000 units per minute violated the max feedrate settings for the two movements.  If there is a violation then the feedrate for each axis is adjusted so that there isn't a violation and both movements finish at the same time.

If you're also trying to predict how long a single movement command will take you'll also need to take into consideration that acceleration & jerk limits will extend the time by a little bit.

If you're trying to calculate how long a series of movements will take then you also need to consider the optimizer function.  It scans the queued up movement commands and adjusts the start and end accelerations for each command such that maximum speed is maintained as long as possible.

kcheeeung, 2017-04-10:
Quite interesting. Thanks so much for the detailed explanation :)

kcheeeung, 2017-04-10:
Hey Bob, one more question. 
Say for I have 800 steps/mm and the E max feedrate is 1mm/sec. It will take 1 sec to complete 800 steps. 

For a E max feedrate of 2mm/sec, it will take 0.5 sec to complete 800 steps?

Bob-the-Kuhn, 2017-04-10:
Correct

kcheeeung, 2017-04-18:
Hey Bob, I have a problem with the firmware that you may be able to help me solve. On the Azteeg X3 Pro board, I have 1/16 stepping set with the settings found on my Git. https://github.com/kcheeeung/MarlinRC8git

When I disable all microstepping by removing all the jumpers SD6128 drivers (for 1 full step), the motors make a really loud noise. I have Nema 17 17HS16 2004S1 motors. Is there something in the firmware that I must change in order to have no microstepping on the extruders? I tried setting the jumpers on 1/2 steps and it also didn't work.

http://www.omc-stepperonline.com/3d-printer-nema-17-stepper-motor-2a-45ncm64ozin-17hs162004s-p-16.html

Bob-the-Kuhn, 2017-04-18:
The Azteeg X3 Pro board doesn't support setting the microsteps via firmware.  Some cards do but not the Azteeg X3 Pro.

With all the jumpers removed you have no microstepping.

One of the benefits of microstepping is that the motors run smoother which means they are quieter.  

One of the drawbacks with microstepping is you loose power as you increase the number of microsteps.

Basically it's a tradeoff between noise & power,  You'll need to do some playing to see what makes your system happy.

closed 2017-04-28 by kcheeeung