MarlinFirmware/Marlin #6276: Arduino console says I cant reach a point in my printer

batata004, 2017-04-08:
I am using the latest version of MARLIN and the configs file attached.

I have a probe that makes use of a SERVO that has a switch at the tip. I believe all the configurations in the files attached are correct but when I try to upload the sketchup to my arduino mega the console says:

#error "Z_SAFE_HOMING_X_POINT can't be reached by the Z probe."

My probe is at the left/front of the hotend. I inverted the values displayed by M114 and saved in the files attached but for no reason I am getting this error message.

How can I fix it?
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/907830/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/907831/Configuration_adv.txt)

Bob-the-Kuhn, 2017-04-08:
X_PROBE_OFFSET_FROM_EXTRUDER -141
Y_PROBE_OFFSET_FROM_EXTRUDER -100  

These are way off.

You need to measure from the nozzle tip to the probe tip when setting these values.

Safe homing wants to put the **PROBE** at 85,85.  That means the nozzle has to be at 85 + X offset, 85 + Y offset.  -56, -15 isn't reachable.

batata004, 2017-04-09:
Hi @Bob-the-Kuhn I am doing this:

1) precise position the nozzle in a mark at the middle of my heated bed using pronterface;
2) then I send the code "G92 X0 Y0 Z0".
3) after that I precise position the tip of my servo arm (that has a switch in it) right at the same mark where previously I had put the nozzle;
4) I finally send the code "M114" to my printer and it displayed those values below:

X_PROBE_OFFSET_FROM_EXTRUDER -141
Y_PROBE_OFFSET_FROM_EXTRUDER -100

What am I doing wrong? The servo arm is around 7cm distnce from the nozzle in the X and 3cm in the Y. What is going on wrong with those values?

Roxy-3D, 2017-04-09:
> The servo arm is around 7cm distnce from the nozzle in the X     

You do mean 7mm from the nozzle, right?   You don't really mean 70mm, right?

Bob-the-Kuhn, 2017-04-09:
Something really strange is going on.  As best I can tell what you're reporting can't happen ... but it is.  

We really need to get your firmware & config file versions straightened out.  After that we can see if you're still getting those strange results.

After straightening out the above then my recommendation would be to do the following:
G28
move the probe to the center
M114
G92 X0 Y0 Z0
M114
move the nozzle to the center
M114
copy the results shown on the console and paste them into the reply

batata004, 2017-04-09:
@Bob-the-Kuhn  I think you got it! I was using the configuration.h and configuration_adv.h provided by @ghost (in the other thread that you are also helping me) and, because he didnt say anything, I assumed those configs files were supposed to work with the main flavor version of MARLIN -> https://github.com/MarlinFirmware/Marlin

So, all this time, I was trying to use a different configuration files than the marlin that I had downloaded. You nailed it, I have no idea how did you guess this was the problem but finally I compiled my configs files with the correct version of marlin and it worked perfectly, no errors at all.

Now I will follow exactly your commands and paste here the results when I get them done!

batata004, 2017-04-09:
IT WORKED PERFECTLY! I actually had a problem bringing the Z down below 0 (home) but @Bob-the-Kuhn 
 helped me in another thread so this problem is fixed. My OFFSETs are now much more consitent and I get no more errors while compiling!,

closed 2017-04-09 by batata004