MarlinFirmware/Marlin #6280: Bug when using LCD REPRAP_DISCOUNT_SMART_CONTROLLER

batata004, 2017-04-09:
Hi,

I am using the latest code provided by MARLIN. I downloaded it and opened it in ARDUINO IDE (latest). Then I uncommented the line 

#define REPRAP_DISCOUNT_SMART_CONTROLLER

When I click compile button, I see this error message:

```
sketch\ultralcd.cpp: In function 'void _lcd_move_xyz(const char*, AxisEnum)':

ultralcd.cpp:1414: error: 'min_software_endstops' was not declared in this scope

       float min = (soft_endstops_enabled && min_software_endstops) ? soft_endstop_min[axis] : current_position[axis] - 1000,

                                             ^

In file included from sketch\MarlinConfig.h:27:0,

                 from sketch\Marlin.h:36,

                 from sketch\ultralcd.h:26,

                 from sketch\ultralcd.cpp:23:

ultralcd.cpp:1431: error: 'max' was not declared in this scope

       if ((int32_t)encoderPosition > 0) NOMORE(current_position[axis], max);

                                                                        ^

sketch\macros.h:66:33: note: in definition of macro 'NOMORE'

 #define NOMORE(v,n) do{ if (v > n) v = n; }while(0)

                                 ^

exit status 1
'min_software_endstops' was not declared in this scop
```

Do you know how can I workaround this so I can use this LCD type?

Tannoo, 2017-04-09:
Is that all you setup was the display?

Tannoo, 2017-04-09:
>I am using the latest code provided by MARLIN.

Would that be the latest Release, RC or RCBugFix?
There are 3 "lastest" flavors.

Bob-the-Kuhn, 2017-04-09:
This is also being discussed in issue #6140 and may be the cause of issue #6276.

Right now it looks to me that he's probably using RC8 but the config files are from RCBugFix.  He's seeing some really strange behavior.

batata004, 2017-04-09:
I am sorry guys if I opened this thread related to the others. But I am completely lost so I decided to try to fix every problem I am having separetely, ok? This way I can fix one by one and make sure I am moving ahead to solve this prblem.

Could you kindly tell me why there are 3 flavors of the latest firmware? I thought the latest was just the one that sits here:

https://github.com/MarlinFirmware/Marlin

Why there are other "latest" like the one @Bob-the-Kuhn said at here:

https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

?

Which one should I use?

Tannoo, 2017-04-09:
Well, there is the official release:
https://github.com/MarlinFirmware/Marlin/tree/1.0.x

There is the release candidate (RC):
https://github.com/MarlinFirmware/Marlin

Then there is the release candidate bugfix (RCBugFix):
https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

The RCBugFix branch has all the latest and greatest feature and fixes going for the next release.
Most of us here are running this. This is the branch that @Bob-the-Kuhn mentioned.
Recommendation is to use this one.

The RC that you downloaded is good. But there are also LOTS of fixes and additions in RCBugFix since that RC was released.

batata004, 2017-04-09:
@Tannoo wow that's was very easy to understand! So I will stick with the RCBUGFIX version as I am not doing anything too serious, so I can afford a not too stable version.

Would you mind telling me how often are the fixes implemented in the RCBUGFIX commited to the main branch? For example, if one bug gets fixed today in the GCBUGFIX, how long will it take to this bug gets fixed in the main branche (official latest release)?

Bob-the-Kuhn, 2017-04-09:
Usually there isn't a schedule for creating an RC but currently we're aiming for a full 1.1.0 release around the end of April.

batata004, 2017-04-09:
Thanks, good to know it does not take too long!

closed 2017-04-09 by batata004