MarlinFirmware/Marlin #6288: Cant lower Z index anymore

batata004, 2017-04-09:
Hi,

I executed G28 and the homing worked perfectly. Now I want to get the Z lower (to make the tip of the nozzle touch the heated bed) but I cant. After sending G28 and waiting I tried sending G92 Z20 so I could lower the nozzle but it is not lowering. I can make it go up but not lower than the actual position.

My configuration file is attached. How can I solve this problem? I am using the latest version of MARLIN.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/908419/Configuration.txt)

batata004, 2017-04-09:
obs: I read in some other threads that G92 is not supposed to help to overcome this problem anymore and I read many suggestions around but many of them seem impractical (like sending a new firmware) cause if I am printing and I have a power disconnect and I am too far away from the 0 at Z I will have lots of trouble having to flash a new firmware only to overcome this problem;

Bob-the-Kuhn, 2017-04-09:
Make sure that the min_software_endstops is set to false.  If true then you won't be able to go below zero.

What I do is:
1. G28
2. move the nozzle to a convenient location
3. G30  ; do a single probe of the bed
4. move the nozzle to the height reported by G30 - that's where the probe is triggering
5. G92 Z0
6. carefully lower the head until a piece of paper just starts to drag on the nozzle
7. M114 ; grab the Z location and then subtract another 0.050" for the thickness of the paper
8. M851 Zxx  ; xx is the value you just calculated above

You'll probably have to fine tune the Z offset based on print results.

Don't forget to put the new Z offset back into your Configuration.h file.  You'll need to do the M851 until you either recompile or you save it to EEPROM.

batata004, 2017-04-09:
YOU ARE A KING!!! Indeed disabling min_software_endstops IT WORKED perfectly! It would be nice to only be able to disable min_software_endstops to Z axis cause when I was using the LCD right now I moved the X axis to far to the left and it hit the endstop and kept going going with very strange noises.

I wanted to disable min_software_endstops only to make the "paper calibration" but instead I almost destroyed my X belt!

Thanks man, I really appreciate your help!

closed 2017-04-09 by batata004

Bob-the-Kuhn, 2017-04-09:
In configuration_adv.h enable ENDSTOPS_ALWAYS_ON_DEFAULT.  That'll use the hardware endstops to keep from trying to go too far.

batata004, 2017-04-09:
So if I enable ENDSTOPS_ALWAYS_ON_DEFAULT and keep min_software_endstops  set to false, the endstops will always be respected?

batata004, 2017-04-09:
Sorry for the stupid comment and question I did, I just had to make a test here and you are right, with both settings the endstops are respected all the times and I still can move my printer below Z = 0!

Thanks thanks thanks!

Bob-the-Kuhn, 2017-04-09:
You're asking the same questions I did when I first started.  

Understanding 3D printing and understanding Marlin is an iterative process.  Seems like every time I go through the config files I find something that finally makes sense or something that I realize I don't really understand.

batata004, 2017-04-09:
Good to know others have been where I am! And good to know these people got really smart in the process! Thank you @Bob-the-Kuhn you made my weekend much more easier helping with so many questions!