MarlinFirmware/Marlin #6290: Firmware set values vs EPPROM values

batata004, 2017-04-09:
Hi,

I set some values inside the FIRMWARE code and uploaded them to the printer. Values like bed temperature, abs hotend temperature...

So after that I decided to use the LCD to change those values (with computer disconnected). 

My question is this: next time when I upload a new firmware will those epprom values be lost/reseted? I think so, then if those values are lost is there anyway to arduino IDE backups them before uploading the new sketch?

Bob-the-Kuhn, 2017-04-09:
If the EEPROM is enabled then the values are restored on power up and after a reset.

The EEPROM can literally wear out so all changes to it are done manually.

The EEPROM values override the values in the config files.   If you change something in your config files you'll need to do a M502 followed by a M500 to save it to EEPROM.

If you change something via the LCD or via a command like M851 then just do a M500 to save it.

Not all settings are saved in EEPROM.  

Compiling and uploading will not change the values in EEPROM.

The Arduino IDE does not backup anything.

One nasty is that sometimes, when switching to a newer version of Marlin, the EEPROM gets wiped out.  This happens because there's been an improvement in how/what is stored in EEPROM but there's never been an effort at backward compatibility.  It's **VERY** important that any changes made via the LCD or the host interface be copied back into your config files.

The UBL bed leveling system is an exception to the backward compatibility issue.  That data is stored in a separate section of the EEPROM.  The creators of the UBL system didn't want all the work it takes to get a finely tuned UBL grid/mesh to be thrown away when a new variable/feature is added to Marlin.

batata004, 2017-04-09:
Great, that's really good to know. I just didnt get one thing, in my config file (which is the default of the Marlin configuration) I have this:

//#define EEPROM_SETTINGS

You see, it's commented. So EEPROM should be disable right? But when I set some configuration using LCD (for example, ABS BED TEMPERATURE) and hit save, it saves. If I power down/up my printer the saved value is still there. Shouldnt it be gone?

Bob-the-Kuhn, 2017-04-09:
That's my understanding but .. I don't know much about the LCD system.  I much prefer the host interface.

batata004, 2017-04-09:
@Bob-the-Kuhn  could you please help me with one more question? I am trying to print my very first part right now. I used simplify3d to generate the gcode and I removed any configuration in the simplify3d regarding temperature. So after generating the GCODE I went to pronterface, clicked in the set temperature button to set temperature of heat bed and hotend and loaded gcode file. Then I clicked to start print. During the print the temperatures are droping, for some reason marlin is not holding the temperature that it was told to keep in pronterface.

I dont want to set temperature in the gcode cause sometimes I wanna try something different so I dont want to have to generate a new gcode just to increase 5 celsius degree the hotend. So how can I solve this problem?

Bob-the-Kuhn, 2017-04-09:
Does it keep dropping or does it drop a few degrees and stay there?

If it keeps dropping then use the M104 or M109 commands to set the extruder temperature yourself rather than use Pronterface.

If it's dropping a few degrees and staying there then it being controlled but not as well as you want.  Use the M303 auto tune command and see if the parameters it suggests work better.

```
/**
 * M303: PID relay autotune
 *
 *       S<temperature> sets the target temperature. (default 150C)
 *       E<extruder> (-1 for the bed) (default 0)
 *       C<cycles>
 *       U<bool> with a non-zero value will apply the result to current settings
 */
```

Roxy-3D, 2017-04-09:
> You see, it's commented. So EEPROM should be disable right? But when I set some configuration using LCD (for example, ABS BED TEMPERATURE) and hit save, it saves. If I power down/up my printer the saved value is still there. Shouldnt it be gone?

Can you verify that?   It should not be saved if EEPROM is turned off.   And in fact, the menu options should not be presented to the user either.

batata004, 2017-04-10:
I think I got the problem, anyway I am not sure if it's a malfunction/bug. I had EEPROM enabled cause I wanted to save some settings in the memory. Then I gave up and found this process too tedious and I would be better saving those configs in the firmware file itself. So I uploaded the new firmware (with EEPROM disabled) and started pronterface right away. I checked for some configurations (use gcodes) and the printer was telling me some data that was not in the firmware but in the memory that I had set previously.

I thought that was odd, so I powered everything off and on again. Them I clicked disconnect in pronterface and connected again. Then I asked for some configurations (using gcodes) and the printer was retrieving the exactly values from the firmware!

Maybe it's a pronterface problem that for some reason "cached" the response from the printer, I dont know. But after @Bob-the-Kuhn  told me to disable EEPROM I did that and after power off the printer it worked.

Anyway, this problem is solved, just need to unplug and plug power cord AND usb cable and it worked just fine, excatly as @Bob-the-Kuhn  said!

closed 2017-04-10 by batata004

batata004, 2017-04-11:
My friend @Bob-the-Kuhn  would you mind recommending a good and active forum where I can try to get some questions answered? I tried http://forums.reprap.org but it's not an active forum, I made a few questions and some of them dont even have an answer for days. I know it's a community and I am not demanding and answer, but is there a good forum where I can ask and have a good probability of gettins helped?

I just printed the "boat" that people usually print to calibrate the printer and I would like to show the picture and ask some advise how I can improve.

Bob-the-Kuhn, 2017-04-11:
They need some old retired people with lots of time on their hands.

Sorry, I don't have a recommendation.  I haven't asked any questions on those forums.  I'm having more fun with the firmware than I am with the printer itself so I'm pretty low on the learning curve on the printing part.

batata004, 2017-04-11:
thanks for your honesty :) BTW you are all doing a great job with Marlin, it's a great project and it works amazingnly well! It's head to head with major players in 3d printing.

thinkyhead, 2017-04-13:
The forums on Thingiverse seem to get a lot of traction. Sometimes even too much.

batata004, 2017-04-13:
Thnks I will take a look at it!