MarlinFirmware/Marlin #6291: Delta homing problem

iNherjer, 2017-04-09:
Hi 
When I send the G28 and the effector is not in the center of the printer (say x stays 5 cm above y and z) , the uppermost arm hits the endstop and a second later the firmware seems to crash. When all 3 arms are at the (nearly) same high, homing works normal.

RC8 from dec16

iNherjer, 2017-04-10:
Solved the problem my self, the stock homing speed was to fast, lowered from 200*60 to 50*60 and it works now like it should. Thanks

closed 2017-04-10 by iNherjer

thinkyhead, 2017-04-11:
When feedrates are too high for Marlin to keep up, the "watchdog" ends up getting invoked after a 4 second delay, and this reboots Marlin. If you see rebooting issues, too many steps-per-second is usually the culprit. If you reduce the number of micro-steps on your Z axis you can get a higher feedrate. If the number of micro-steps is greater than 16 then you can probably afford it.