MarlinFirmware/Marlin #6292: ISR stack frame usage

Roxy-3D, 2017-04-10:
It used to be that the ISR stack frame needed well less than 64 bytes of space.   Right now, it is around 200 bytes.    Do we really need the stack frame to be so large?   It is cutting into the 'free memory' pool pretty dramatically.

I'm not sure if this is because of recursive interrupt nesting.   Or because of something else like enabling Serial Transmitter interrupts.   I haven't dug down deep enough to know the specifics yet.   But I would like people to be aware the ISR routines are getting a little bit piggish about the stack frame.   (I wish we had a 32-Bit processor running!!!)

mperdue, 2017-04-10:
I'm waiting for 32-bit as well. :-)

Roxy-3D, 2017-04-10:
If anybody is interested in exploring the amount of memory (stack frame) used by the various ISR's...   You turn on 
```cpp
#define M100_FREE_MEMORY_WATCHER
``` 
in your Configuration.h file and give it a M100 I  (to initialize it) at startup.    At any time you can ask (to this point in time) how much free memory was available with a M100 F command.

I don't know how to interpret the various stack frames.   I thought AVR was 'little endian'.  But it sure looks to me like the return addresses pushed on the stack are 'big endian'.    If I could parse the stack frame that is visible when you do a M100 D command, I could figure out exactly what ISR was calling what and how much local data that routine set aside.   It is probably just one ISR that is guilty of chewing up all that memory...  But I don't know how to figure it out from looking at the M100 D dump.

thinkyhead, 2017-04-13:
It probably uses several stack-based variables with long lifespans. It pre-calculates a lot of values to save on computation to do its work. It is a complex function. There is no recursion, and I believe it calls no external functions.

thinkyhead, 2017-04-13:
The Stepper and Temperature ISRs are pretty straightforward. They don't use a lot of local stack-based variables and don't call any complex functions. However, both the Stepper ISR and Temperature ISR have been made more flexible about being interrupted so that the Serial UART can do its work whenever it wants. This seems to have improved the reliability of serial communication a lot, with an apparent elimination of the checksum errors that used to be very common.

So, I guess the downside is that it's is possible for more interrupt stack frames to accumulate, but never more than 3.

closed 2018-02-26 by thinkyhead