MarlinFirmware/Marlin #6294: Advice with layer height

batata004, 2017-04-10:
I dont know if this page is the correct place to ask this question but I would like some advice: I followed @Bob-the-Kuhn orientation how to find the correct offsets and set them in my marlin firmware. I use a servo probe and it looks everything works fine.

When I tried to find the correct offsets, in one of the last steps I had to put the nozzle as close as possible to the bed and try to pass a paper under it and check if there is some pressure. After that, the instruction says, I need to move the Z axis 0.1mm above that and set that as my ideal position.

So I ask you this: when I am using 0.3mm layer height, can I keep the same offsets or should I make all the procedures to find the correct offsets and at the end move the Z axis 0.3mm after it touches de paper?

StefanIGit, 2017-04-10:
As of my understanding and experience this is not the right place to ask this, this here is about possible problems with the firmware. 
Your question is more of a general 3d printing understanding question which I would ask in forums like reprap forum.
Any way this offset defines the starting point aka 0 on the z axis. When changing the layer height in your slicer like 0.1 to 0.3 you do not have to change the offset (z 0 point). The slicer knows the layer height and just calculated the correct needed z move.   Like printing 1 or 456mm height 0 is always 0 but layer 1@0.1 would be z 0.1 and layer 1@0.3 would be z0.3.
So it moves z to 0.1, 0.2, 0.3... Or 0.3, 0.6, 0.9...  Hope this helps.

batata004, 2017-04-10:
Thanks that makes sense @StefanIGit . Sorry for asking it here, I will use reprap forum next time. I didnt know marlin would move the nozzle to the specific layer height before printingir!

closed 2017-04-10 by batata004

StefanIGit, 2017-04-10:
Hi, it's not Marlin how is doing the moving.... Ok yes it is, but the cause of it is the slicer which is telling Marlin to do so by making g-code out a the 3D model if you want learn more look here http://reprap.org/wiki/G-code#G0_.26_G1:_Move

batata004, 2017-04-10:
Alright it looks right, the gcode generator probably is the one that takes care of that. I will read that link :) Thanks.