MarlinFirmware/Marlin #6298: Z-babystepping doesn't work

Kaibob2, 2017-04-10:
The Z-babysteps don't work anymore when i upgraded from [this](https://github.com/MarlinFirmware/Marlin/commit/fe2ea2906f356acf359c0eb27b21adae9bc6ea82) RCBugfix version to the [most recent one](https://github.com/MarlinFirmware/Marlin/commit/dad907b9233d2ca3ee27b5c4613559a71e1b2dd2)

X and Y babysteps are working. Enabling or disabling `DOUBLECLICK_FOR_Z_BABYSTEPPING` doesn't make a change.

I surely merged the old config files correctly with WinMerge respecting all new changes etc.

Here's my config
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/910940/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/910941/Configuration.txt)

simon-jouet, 2017-04-11:
HI Kaibob2,

Same here, I have pretty much the latest (f8408036f2d3f74483eb485c423a2af17bd8ae38) and babystepping stopped working. Might it be something related to the changes brought by UBL?

Kaibob2, 2017-04-11:
@simon-jouet Hi Simon, i can only confirm, that the Z-Babystepping doesn't work. X and Y Babysteping is working fine as before.

@thinkyhead I copied the working babystep code part from the version i used before to the most recent one and then it works again as it did before. Unfortunately i don't understand what exactly has changed in the new code.
In detail:
In stepper.cpp i replaced
```
#if ENABLED(BABYSTEPPING)

  #define CYCLES_EATEN_BY_BABYSTEP 60

  #define _ENABLE(axis) enable_## axis()
  #define _READ_DIR(AXIS) AXIS ##_DIR_READ
  #define _INVERT_DIR(AXIS) INVERT_## AXIS ##_DIR
  #define _APPLY_DIR(AXIS, INVERT) AXIS ##_APPLY_DIR(INVERT, true)

  #if STEP_PULSE_CYCLES > CYCLES_EATEN_BY_BABYSTEP
    #define _SAVE_START (pulse_start = TCNT0)
    #define _PULSE_WAIT while ((uint32_t)(TCNT0 - pulse_start) < STEP_PULSE_CYCLES - CYCLES_EATEN_BY_BABYSTEP) { /* nada */ }
  #else
    #define _SAVE_START NOOP
    #define _PULSE_WAIT NOOP
  #endif

  #define START_BABYSTEP_AXIS(AXIS, INVERT) { \
      old_dir = _READ_DIR(AXIS); \
      _SAVE_START; \
      _APPLY_DIR(AXIS, _INVERT_DIR(AXIS)^direction^INVERT); \
      _APPLY_STEP(AXIS)(!_INVERT_STEP_PIN(AXIS), true); \
    }

  #define STOP_BABYSTEP_AXIS(AXIS) { \
      _PULSE_WAIT; \
      _APPLY_STEP(AXIS)(_INVERT_STEP_PIN(AXIS), true); \
      _APPLY_DIR(AXIS, old_dir); \
    }

  // MUST ONLY BE CALLED BY AN ISR,
  // No other ISR should ever interrupt this!
  void Stepper::babystep(const AxisEnum axis, const bool direction) {
    cli();
    uint8_t old_dir;
    #if STEP_PULSE_CYCLES > CYCLES_EATEN_BY_BABYSTEP
      uint32_t pulse_start;
    #endif

    switch (axis) {

      case X_AXIS:
        _ENABLE(x);
        START_BABYSTEP_AXIS(X, false);
        STOP_BABYSTEP_AXIS(X);
        break;

      case Y_AXIS:
        _ENABLE(y);
        START_BABYSTEP_AXIS(Y, false);
        STOP_BABYSTEP_AXIS(Y);
        break;

      case Z_AXIS: {

        #if DISABLED(DELTA)

          _ENABLE(z);
          START_BABYSTEP_AXIS(Z, BABYSTEP_INVERT_Z);
          STOP_BABYSTEP_AXIS(Z);

        #else // DELTA

          bool z_direction = direction ^ BABYSTEP_INVERT_Z;

          enable_x();
          enable_y();
          enable_z();
          uint8_t old_x_dir_pin = X_DIR_READ,
                  old_y_dir_pin = Y_DIR_READ,
                  old_z_dir_pin = Z_DIR_READ;
          //setup new step
          X_DIR_WRITE(INVERT_X_DIR ^ z_direction);
          Y_DIR_WRITE(INVERT_Y_DIR ^ z_direction);
          Z_DIR_WRITE(INVERT_Z_DIR ^ z_direction);
          //perform step
          #if STEP_PULSE_CYCLES > CYCLES_EATEN_BY_BABYSTEP
            pulse_start = TCNT0;
          #endif
          X_STEP_WRITE(!INVERT_X_STEP_PIN);
          Y_STEP_WRITE(!INVERT_Y_STEP_PIN);
          Z_STEP_WRITE(!INVERT_Z_STEP_PIN);
          #if STEP_PULSE_CYCLES > CYCLES_EATEN_BY_BABYSTEP
            while ((uint32_t)(TCNT0 - pulse_start) < STEP_PULSE_CYCLES - CYCLES_EATEN_BY_BABYSTEP) { /* nada */ }
          #endif
          X_STEP_WRITE(INVERT_X_STEP_PIN);
          Y_STEP_WRITE(INVERT_Y_STEP_PIN);
          Z_STEP_WRITE(INVERT_Z_STEP_PIN);
          //get old pin state back.
          X_DIR_WRITE(old_x_dir_pin);
          Y_DIR_WRITE(old_y_dir_pin);
          Z_DIR_WRITE(old_z_dir_pin);

        #endif

      } break;

      default: break;
    }
    sei();
  }

#endif // BABYSTEPPING
```
with
```
#if ENABLED(BABYSTEPPING)

  #define _ENABLE(axis) enable_## axis()
  #define _READ_DIR(AXIS) AXIS ##_DIR_READ
  #define _INVERT_DIR(AXIS) INVERT_## AXIS ##_DIR
  #define _APPLY_DIR(AXIS, INVERT) AXIS ##_APPLY_DIR(INVERT, true)

  #define BABYSTEP_AXIS(axis, AXIS, INVERT) { \
      _ENABLE(axis); \
      uint8_t old_pin = _READ_DIR(AXIS); \
      _APPLY_DIR(AXIS, _INVERT_DIR(AXIS)^direction^INVERT); \
      _APPLY_STEP(AXIS)(!_INVERT_STEP_PIN(AXIS), true); \
      delayMicroseconds(2); \
      _APPLY_STEP(AXIS)(_INVERT_STEP_PIN(AXIS), true); \
      _APPLY_DIR(AXIS, old_pin); \
    }

  // MUST ONLY BE CALLED BY AN ISR,
  // No other ISR should ever interrupt this!
  void Stepper::babystep(const AxisEnum axis, const bool direction) {

    switch (axis) {

      case X_AXIS:
        BABYSTEP_AXIS(x, X, false);
        break;

      case Y_AXIS:
        BABYSTEP_AXIS(y, Y, false);
        break;

      case Z_AXIS: {

        #if DISABLED(DELTA)

          BABYSTEP_AXIS(z, Z, BABYSTEP_INVERT_Z);

        #else // DELTA

          bool z_direction = direction ^ BABYSTEP_INVERT_Z;

          enable_x();
          enable_y();
          enable_z();
          uint8_t old_x_dir_pin = X_DIR_READ,
                  old_y_dir_pin = Y_DIR_READ,
                  old_z_dir_pin = Z_DIR_READ;
          //setup new step
          X_DIR_WRITE(INVERT_X_DIR ^ z_direction);
          Y_DIR_WRITE(INVERT_Y_DIR ^ z_direction);
          Z_DIR_WRITE(INVERT_Z_DIR ^ z_direction);
          //perform step
          X_STEP_WRITE(!INVERT_X_STEP_PIN);
          Y_STEP_WRITE(!INVERT_Y_STEP_PIN);
          Z_STEP_WRITE(!INVERT_Z_STEP_PIN);
          delayMicroseconds(2);
          X_STEP_WRITE(INVERT_X_STEP_PIN);
          Y_STEP_WRITE(INVERT_Y_STEP_PIN);
          Z_STEP_WRITE(INVERT_Z_STEP_PIN);
          //get old pin state back.
          X_DIR_WRITE(old_x_dir_pin);
          Y_DIR_WRITE(old_y_dir_pin);
          Z_DIR_WRITE(old_z_dir_pin);

        #endif

      } break;

      default: break;
    }
  }

#endif //BABYSTEPPING
```

I think this will give you a hint what might be the cause.

thinkyhead, 2017-04-11:
Awesome! I wish everyone would provide as much detail. I'm sure it'll be solved momentarily.

Kaibob2, 2017-04-11:
I do what i can. I'm always glad when i can contribute.

thinkyhead, 2017-04-11:
So, it stands out that the non-working code includes handling for `MINIMUM_STEP_PULSE` while the older code gets away with a call to `delayMicroseconds(2)` between pulse start and pulse stop.

I see that the newer code will not produce an adequate length pulse unless the `MINIMUM_STEP_PULSE` stuff is activated. In fact, I have a PR fixing that feature at #6309, so I will try restoring the `delayMicroseconds(2)` call when that `MINIMUM_STEP_PULSE` isn't in use, then you can test the code in that PR.

Kaibob2, 2017-04-11:
@thinkyhead I downloaded your [Fix minimum pulse fork](https://github.com/thinkyhead/Marlin/tree/rc_fix_minimum_pulse) and only copied my 2 configuration files into it, compiled and uploaded.

Now, when i start a print the steppers start whisteling and after a few seconds the printer resets.
So, not really an improivement :)

thinkyhead, 2017-04-11:
It's a work-in-progress, and I've just patched it further. It probably doesn't fix the reset issue, however. Let me give it some testing and I'll let you know when it's safe to use. Please post your configuration files to ensure that I see what you're seeing.

Kaibob2, 2017-04-11:
@thinkyhead I was a little to fast. I just downloaded your fork again, but nothing changed. When i start the print the steppers get powered, reset, powered, reset, powerd, reset, Main screen. 
My configuration files are in the first post,

thinkyhead, 2017-04-11:
Yep, sure enough, that branch needs a little work! I'm running it on my i3 at the moment, and hopefully will figure it out soon. I have to meet with my client in a little while, but I will try to have this patched later today.

Kaibob2, 2017-04-11:
OK, i will test it then.
For the moment i will fix it like i mentioned above by exchanging the code snippet back to the working version.

thinkyhead, 2017-04-11:
All better!

It turns out that this will trigger the `#error` line:
```cpp
#define SMALL_VALUE 10
#define LARGE_VALUE 5
#define DIFF_VALUE (LARGE_VALUE - (SMALL_VALUE))
#if DIFF_VALUE > 2000
  #error "Yep! A 'negative' value is considered 'positive' by the preprocessor."
#endif
```
Who knew?

Roxy-3D, 2017-04-11:
That seems wrong.

thinkyhead, 2017-04-11:
Ah, here's why:
```cpp
#define CYCLES_PER_MICROSECOND (F_CPU / 1000000UL) // 16 or 20
```
…needs to be changed to…
```cpp
#define CYCLES_PER_MICROSECOND (F_CPU / 1000000UL) // 16 or 20
```
Subtle stuff, this C++.

Roxy-3D, 2017-04-11:
Those two lines are identical.

Kaibob2, 2017-04-11:
Not really. 1000000UL vs 1000000L

Roxy-3D, 2017-04-11:
No...  I just cut and paste the post and got the two lines right on top of each other.   I think those two #define's are the same.

Kaibob2, 2017-04-11:
@thinkyhead I can confirm, that with 6a9b008 XY and Z Babystepping is working!! Perfect!

thinkyhead, 2017-04-11:
> I think those two #define's are the same.

A page reload will convince you otherwise. ;-)

> The meaning of the other stuff in your PR isn't clear to me.

The other stuff fixes `MINIMUM_STEPPER_PULSE`, in case anyone needs it.

simon-jouet, 2017-04-11:
Thanks @thinkyhead, I finish my print and I will test this out, but guessing I had exactly the same issues as @Kaibob2 and it works for him, shouldn't be a problem on this end either.

thinkyhead, 2017-04-11:
I'm pretty confident we've got this one solved. With `MINIMUM_STEPPER_PULSE` left at zero it merely restores the original 2µs delay between stepper pulse start and stop.

Please re-open this issue if the problem crops up again.

PR #6309 (2017-04-11): Fix MINIMUM_STEPPER_PULSE calculations
Pointed out by @loos-br at https://github.com/MarlinFirmware/Marlin/commit/eabff30e75429fe1db715c46c123fde678e689e6#commitcomment-21669327 …
And may also fix #6298.

**Background:** I forgot to include the 8x prescaler for `TCNT0` when I wrote `MINIMUM_STEPPER_PULSE`. As a result, pulses using this feature will have been too long.

**Solution:** This PR fixes calculations to use the prescaler when determining the required delay.

In addition…
- Added a macro to perform small delays (up to ~1µs) using `NOP` commands.
- Use the NOP-based delay macro if only [... cut]
closed 2017-04-11 by thinkyhead