MarlinFirmware/Marlin #6299: Printer restarts after autolevel

Ashikaru, 2017-04-10:
I am working on a printer of my own design for a senior design project. The printer is being run via USB at a baud rate of 115200 using an AT Mega 2560/RAMPS 1.4 with DRV8825 stepper drives. The error also occurs when running the printer off of an SD card.

Last night I was able to get the printer to run 5 times in a row, but only if I completely shut it down and disconnected the power/USB first. If I didn't do that then the printer would reset after autoleveling or it would run through all of the gcode in about 3 seconds causing the printer to go crazy and crash into the end stops/maximum position. This morning I moved the printer onto my desk to start working on it again and it will no longer run even after a full power cycle which makes it seem like an issue with the wiring maybe? I just dont understand though since I've gone through the wiring several times and have crimped and recrimped the connectors multiple times as well as plugging the wires directly into the board without the Molex plugs to ensure that there was a good connection.

I've been dealing with this issue for the last 2 weeks almost and am running out of ideas. I can upload the config.h file or a short video of the issue if anyone would like to take a look. Any and all help would be very appreciated.

Bob-the-Kuhn, 2017-04-10:
Please provide the following and we'll at least get you to the point where the heads not crashing into the stops.
- Which flavor of Marlin are you running?  1.02.x, RC8, RCBugFix from 15 March, ...
- Do you have hardware endstops and are they working?
- A diagram of how you attached the inductive sensor to your machine.  Most of these run at 12V.  If the 12V gets to the wrong places then all kinds of nasty/strange things can happen.
- Have you calibrated/verified your steppers (if you command a 100mm move, do you get a 100 mm move)?
- Your configuration.h file
- Your Configuration_adv.h file

The files & diagram can be ZIP'd and then dropped on your reply.

Ashikaru, 2017-04-10:
I'm running Marlin 1.1.0-RC8. I can home each axis just fine on its own, I'm using a mechanical switch on X and Y with an inductive probe on Z. I setup a voltage divider on the probe so that the output was ~4.5V or so which was verified with a multimeter. It triggers at the exact same point each time with no isues.

I have not calibrated my X/Y/Z steps beyond getting rough values from the reprap calculator/Prusa website.I'm in the right ballpark already though (20mm cube measures 20.05x19.75x19.95) E steps have been calibrated. 

I'm getting slightly more success with having EEPROM disabled and running an M502 in the starting gcode of the printer, but it still doesn't work every single time. I'll get the config files uploaded for you to check out.

Thank you for your help!

Ashikaru, 2017-04-10:
I guess I'm stupid and can't upload .zip files to Github for whatever reason, but here's the files on dropbox: https://www.dropbox.com/s/29hiim7rulr17c4/Configuration.zip?dl=0

So far the M502 line seems to make the printer much more reliable. I'm still trying to get it to fail again though. I'm not sure why the printer would automatically clear the memory after a print stopped though. I've got no issues with my other 2 printers.

Bob-the-Kuhn, 2017-04-10:
Glad to hear things are going better.

GitHub can be downright cantankerous.  When it's cooperating it likes ZIP, TXT and photos and that's about it.

The only unusual thing I saw in the config files was the steps per unit for the Z axis.  I'm jealous!

The software endstops should have kept the head from grinding away on the stops.  You may want to have the hardware endstops always enabled.  Maybe even add X_MAX & Y_MAX endstops.

With the EEPROM disabled, M502 has no effect.  With EEPROM enabled you need to do a M502 followed by a M500 whenever you change your configuration.  There's been a LOT of hair pulling because of forgetting to do the M502, M500.

I can believe you had repeated attempts to go beyond the stops when executing the slicer output.  Marlin will continue to execute the gcode file even though it's run into an endstop.

Have you checked that the bed size the slicer's using matches your printer?

----

What do you mean by `clear the memory` ?

----

If you continue to have problems after enabling the EEPROM and doing the M502, M500 then I suggest going to [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).  Lots of fixes since RC8 and some new features.

The down side is you'll have to manually transfer your machine specific settings over to the new config files.

thinkyhead, 2017-04-11:
> There's been a LOT of hair pulling because of forgetting to do the `M502`, `M500`.

I assume this is due to the eternal cause: It's easy to forget that simply re-flashing won't use your new config settings until you do `M502` and `M500`.

I believe the current checks for matching version, EEPROM data length, and valid checksum are catching those cases where the EEPROM layout has changed (and thus the default settings _**are**_ in effect.)

knobik, 2017-04-13:
@Ashikaru: Are you using a aluminium hotbed? Check if the hotend is touching the hotbed and possibly causing a short in the circuit. Overloading your power supply / arduino and resetting it.

closed 2017-10-31 by thinkyhead