MarlinFirmware/Marlin #6303: Only CORE_XY and XZ supported for homing

markfrencken, 2017-04-10:
In the rc 8 and also in the code for the latest build, the homing update function only supports two of the 6 possible CORE configurations.
This is hard to debug since the M commands to verify the end stops will correctly report the status, but the software will not act on the EOS min sensors (the software incorrectly checks for the MAX EOS swensors when travelling in the - direction for CORE YX for example).
This bug will crash the machine during homing !!

I checked the code so I'm pretty sure the bug is for real, but just to be clear (and to get my printer working) I changed the setup to COREXY and verified this config works for homing.

Bob-the-Kuhn, 2017-04-10:
In [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix) we made a change to the endstop logic about a month ago.  Prior to that CoreYX was being treated the same as CoreXY, ...    That made a CoreYX user happy.  Shouldn't have affected the CoreXY operation.

Have you looked at RCBugFix built in the last month?  Is it doing anything strange?

If there's a bug in RCBugFix we'll get it fixed **quickly**.

One of the problems we have is a lack of testers with Core machines.  We've been known to make changes to fix one type only to find out later we broke another type.

markfrencken, 2017-04-11:
Hi,

 

If you check the code   in void Endstops::update() in the module endstops.cpp ( it is essentially the same for RC8 and bugfix releases),

You will notice that the CORE printers except COREXY and COREYZ are not handled. This will make the routine check the MAX endstops for moves towards the min endstops, even if thay (the MAX endstops) are not installed/enabled.

 

The code to READ the endstops is functioning as it is supposed to, nevertheless.

 

This seems a bug to me.

 

Regards,

 

Mark Frencken

Bob-the-Kuhn, 2017-04-11:
There may well be bug(s).  The problem is testing.  Anything we do will just be an educated guess until we have a way to test the other four configurations .

We'd **REALLY** love to be able to test.  Ideas?

markfrencken, 2017-04-11:
Of course I can test.I could even try some configs myself, if you (or anyone) could point me to, for starters, the definition of the global vars so I do not have to use the code as documentation. I do that on a regular basis at work already 😉

 

Regards,

 

Mark

Bob-the-Kuhn, 2017-04-11:
What, you want ease of use?!!!??   Where do you get such strange ideas?  ;-)

What CORExx configuration(s) can you test?

I don't know what you mean by "definition of the global vars".  

The two configuration files are the first level of documentation.  The new [Wiki](http://marlinfw.org/) is another source of documentation.  Unfortunately the code is the only real documentation of the software and of a lot of the options.

Ignore the certificate warning when going to the Wiki.

What kind of help can I provide?   Normally I just put print statements in all the branches  the code should & shouldn't take.

markfrencken, 2017-04-12:
Hi, I can test this, I could rewire nu setup dit different core config and test honing.
Another approach is to decide what config to support and what not. There is no conceptual difference between a corexy and a coreyx printer but they require a different approach in software. (There is, of course, a conceptual difference between corexy  and corexz so they should remain. )
Another approach, that I believe covers some requests from different long standing feature requests, is to implement a 2-transformation matrix to go from logical to physical axis. That could cover the conceptually alike configurations while offering a software solution for non-otho axis.
Please indicate what direction you want to approach and what the project roadmap looks like.
As said before, I can easily test core homing but I can not test printing in all configurations.
Regards,
Mark

Verzonden vanaf mijn Samsung Galaxy-smartphone.

Bob-the-Kuhn, 2017-04-13:
I'm finally seeing the light.  There's not one CORExx where the endstop detection is 100% correct.

I know the XY and YX were working at one time.  That's the danger when you have too many cooks.

Before I tackle the endstops I need to know if the axis movement logic is correct.  Maybe the endstop logic is correct but the XYZ movements are what's screwed up.

On as many core configurations as convenient, please check that the XYZ movements  are correct.

By any chance can you point me to the movement equations for the various core configurations?  All I can find are the COREXY & COREYX equations.

If you like to see funny/horrible stuff then copy the attached file.  It's a version of endstops.cpp with lots of print statements added. 
[endstops.zip](https://github.com/MarlinFirmware/Marlin/files/918265/endstops.zip)

The setup I used is:

- ENDSTOP_INTERRUPTS_FEATURE disabled - .  This feature only checks endstops after one has been triggered.  With it disabled the endstops are checked every time a step pulse is issued.
- ENDSTOPS_ALWAYS_ON_DEFAULT enabled - then you don't need to do G28.  G0 commands will be checked
- change your xx_ENDSTOP_INVERTING settings to the wrong way.  This'll keep the display from filling up with useless print statements

markfrencken, 2017-04-13:
Hi,
I will perform the tests you request BUT there is the principal question again.
When I look at the endstop and encoder c files I start to understand the logic. There are 6 corexx types to choose from which are mapped on 3 core_is_xx types. For each core_is_xx type the config on an existing printer can be made to work by changing motor rotation directions and still end up with an orthogonal right turning config, only the origin changes.
To get the endstop working in all intended config we need to clearly define the intended a,b to x,y rotation matrix. This is needed since the endstop logic differentiates between the 2 possibilities of one core_is_xx implementation by looking at the relative direction of the a,b motors. This defines the position of the origin with respect to the pulley s but it is a choice of implementation, not a physical necessity.
So I can write down the config and check how and when endstop are triggered. This should then result in a choice of implementation for all future users of corexy printers (or one should accept a different origin).
I personally think a rotation matrix approach would give more freedom of design.
Will report back with results in a few days.
Regards,Mark

Verzonden vanaf mijn Samsung Galaxy-smartphone.

Bob-the-Kuhn, 2017-04-14:
[endstops corrected.zip](https://github.com/MarlinFirmware/Marlin/files/921189/endstops.corrected.zip)

Here's what I think is the corrected endstop logic.  It now checks the correct endstops and only the correct endstops.

Please do your testing with this updated code.

In the ZIP you'll find the file I've just put on a PR #6335 plus a couple more with different levels of debug prints.  The full debug is useful for checking that the logic is operating correctly.  The direction only one just reports which endstop is being checked.

Also in the file is the logic chain I used when deciding how the compares should be done.

Bob-the-Kuhn, 2017-04-14:
The fixes were just merged into RCBugFix

closed 2017-11-01 by thinkyhead