MarlinFirmware/Marlin #6310: RCBugFix Heater 4 broken

kcheeeung, 2017-04-11:
@thinkyhead I believe you forgot to include a few lines in **thermistortables.h** for the RCBugFix, as the current version does not compile right.

https://github.com/MarlinFirmware/Marlin/pull/6264/files?w=1#diff-208fc9ab75ffd757bceef9d68e6fecfcR171

Bob-the-Kuhn, 2017-04-13:
The M43 code does some unusual memory operations that depend on the compiler to convert between different types of data.

It looks like your compiler and/or your IDE extension have the type conversion flags set to a more conservative level than what most people are using.

What are you using to compile the code?  If it's Arduino I'll want to know the Arduino revision number and the setting of` Board:` under the tools menu.

kcheeeung, 2017-04-14:
Hey Bob, I'm just using the Arduino IDE 1.8.1 to compile. I was just curious what that meant, as I am still able to upload and flash the firmware to the board

Bob-the-Kuhn, 2017-04-15:
I didn't notice those were warnings.  Errors would have killed the compile.

I see those warnings fly by also.

closed 2017-04-16 by kcheeeung

thinkyhead, 2017-04-17:
Seems to have gotten patched along the way. There sure are many details to support an additional extruder.