MarlinFirmware/Marlin #6312: Frequently getting TERMAL RUNAWAY error

batata004, 2017-04-11:
Hi,

I am trying to print some parts and usuaaly at the middle of the printing I get the error TERMAL RUNAWAY and I have to turn off/on the printer and I cant resume the print after that!

So I ask you: is there some debugger/log where I can see what went wrong (if it was the hotend of the heated bed)?

And I also ask you if there is someway to resume the print after that?

I have no idea what is causing this, maybe some configuration can be relaxed?

Bob-the-Kuhn, 2017-04-11:
Repetier Host and Pronterface both will graph the temperatures of the bed and the extruder.  That should give you a good idea of what is going on.

THERMAL RUNAWAY is when the temperature drops below the limits set in the configuration file.

HEATING FAILED is when Marlin fails to increase the temperature enough.

There's advice in Configuration_adv.h on how to solve these issues.

You may have to dial back the speed of the fan(s).

batata004, 2017-04-11:
I was printing with SD card :( I was not using Pronterface nor any software like that. I thought Marlin would save in the epprom some data do help debug when this error happens, but as I understand I think this is not the case.

I took a look as you said in the configuration_adv.h file and you are right, I can learn a lot from there. I decided to change WATCH_TEMP_PERIOD from 20 to 25 in hope that I never more get this error.

But it got me thinking: as I read from the comments inside the _adv file the "timer" starts ticking after the target temperature is hit. Ok, suppose the target temperature is 230 and it gets hit right now. Then in the next minute it falls to 229, after 60 seconds it falls to 228 and so on (one degree falling every minute). In my understanding the thermal protection should never fire cause the "fall" of temperature was within the WATCH_TEMP_INCREASE (in my code it's set to 2).

Sebastianv650, 2017-04-11:
What marlin version are you running? For example rc8 has some bugs related to this. Rcbugfix should be ok.

thinkyhead, 2017-04-11:
Watch what happens to your temperature when the fan is turned on. If it cools the nozzle down too much, you may want to reduce the fan speed.

If your nozzle or bed is just taking a very long time to heat up and you're getting errors before it reaches temperature, try increasing the `WATCH_TEMP_PERIOD` by 10 or 20 seconds.

If your temperature is simply unstable when it gets close to the target temperature, you should run `M303 E0 S200 U` to auto-tune the PID parameters for your nozzle. Save the results to EEPROM with `M500` and copy the values in the log output to the appropriate place in your `Configuration.h` file (around line 320).

Do the same for your heated bed with `M303 E-1 S60 U`. Save the results to EEPROM with `M500` and copy the values in the log output to the appropriate place in your `Configuration.h` file (around line 370).

batata004, 2017-04-11:
@Sebastianv650  I am using the latest stable Marlin. I dont use fan in the printed part, I have only a fan in the extruder (in the cold region of it).

The error I get is during print, not at the start. It's at the middle.

I never heard of the auto tune to hotend and heated bed! I will do that right now and save the values in the firmware. I am confident this will solve most of my problems!

@thinkyhead thanks a lot man, I will come back and tell you if it solved! I will also relax some confgs like WATCH_TEMP_PERIOD!

batata004, 2017-04-11:
Hi @thinkyhead I did exactly what you said in pronterface but after about 3 minutes (the hotend got hot a few times and finally it came cold) the printer finished and didnt tell me the PID coefficientes! I dont have epprom enabled in my marlin, so how can I see the PID result after the calibration?

batata004, 2017-04-12:
@thinkyhead  I did exactly what you asked but a problem happened -> https://github.com/MarlinFirmware/Marlin/issues/6316

closed 2017-04-12 by batata004