MarlinFirmware/Marlin #6314: new idea: Detect stepper issues

stigjoergensen, 2017-04-11:
Hi

A new idea...

We have fail safe on heating... if heating isnt fast enough we stop

On compile time we have detection about impossible configs.

It would be nice if we could have the same detection on movements as we do on heating failed...

the idea is to monitor how long it takes to do a movement vs the length of movement, if threshold exeeded then either pause or abort print. as it indicates that the stepper driver have entered thermal protection.

Sebastianv650, 2017-04-11:
Marlin only sends a step signal at a given frequency to the stepper driver. There is no way to get a feedback, it's open loop.

JustAnother1, 2017-04-11:
The steppers are currently used in an open-loop-setup. That means that marlin toggles the Step and Dir and Enable pins of the pololu modules and expects the stepper to move. If the stepper can not move, due to over heat under voltage, mechanical blocking or whatever then Marlin can not detect that. There is no feedback signal going back to the firmware. Nothing that can be read to detect over heat or the other issues.

I agree that it would be good if the printer could tell the user that "somethings wrong with the X axis" instead of producing a failed print.

One way to solve this could be the trinamic steppers. They have a status register that can be read (using SPI). In that register they have flags for over heating, short to gnd, and the force actually needed to move the motor. I just recently saw a pull request that allowed for homing without end switches by reading this information. It is here:https://github.com/MarlinFirmware/Marlin/pull/6229

Maybe @teemuatlut should take a look?

stigjoergensen, 2017-04-11:
I do understand that there is no direct way to knowing that the stepper
have stalled or something.

However, i still think its possible somehow. what i have noticed on a mates
printer is the movements starts to slow down (physically), but the print
actually completes (ugly does to cooking the filament) but it does all the
right movements...

So how is this possible?

I get the Step, direction and enable - but how does marlin then do 300
steps?

stigjoergensen, 2017-04-11:
Ohhh come to think of it...

Perhaps its pronterface that should know this, because when the printer
slows down we see "busy processing" in the log window of pronterface...

2017-04-11 21:50 GMT+02:00 Stig Joergensen <stig.joergensen@clearsky.dk>:

Sebastianv650, 2017-04-11:
>> I get the Step, direction and enable - but how does marlin then do 300
steps?

By toggling the step pin 300 times at a specific frequency. When your steppers "slow down" which can only be missed steps in reality marlin would need the exact same time to toggle the pin 300 times than without the missed steps.

JustAnother1, 2017-04-11:
If the slowdown is due to missed steps, then that would be noticeable in the print. The print would fail or at least look ugly (Shifted Layers, wrong dimensions,..)

Some slicers actively slow down the print in some stages, to get a better first layer or too let the printed part cool down. So this might be intentional of the slicer and your friends printer is just bad at printing at a low speed.

If steppers overheat or have other malfunctions they just miss steps as  Sebastianv650 said. There is no failure mode that would make the steppers move slower.

You might want to gather more information on this effect. logs, pictures of the printed object, a video, whatever you can get to look for what is happening. 

If the printer uses marlin then Marlin does not know that there is a problem with those steppers.

thinkyhead, 2017-04-11:
The feature you're looking for requires Trinamic stepper drivers, such as the TMC2130 or TMC5130, connected through an SPI interface. These drivers can detect and report overheating, and even when a stepper is prevented from moving, such as when an axis reaches its limit. Thus, these drivers can substitute for end-stops.

thinkyhead, 2017-04-11:
> "busy processing" in the log window of pronterface

It doesn't indicate a problem. These messages are merely hints to the host software that Marlin is still alive, but its current operation is lengthy. You will see this if you send enough long, slow G1 moves that you fill up the command buffer.

teemuatlut, 2017-04-11:
About the Trinamic drivers.
Yes you can read the stallGuard values that indicate the level of load but it wouldn't be very suitable for precisely this. There's an indication that the load was too great and you missed steps, but you have no way of knowing how many steps. This would be the crucial part.

However, and I did look into this, there's another Trinamic way. They have a technology called DC step. The way it works is that you send pulses to the driver and then after the move you check if there were steps missed. If true, step some more. Then check again until the driver says there were no missed steps. This would be perfect for 3Dprinting, as we'd have close to closed-loop setup without (much) greater cost. You'd also have the benefit of not needing acceleration curve at all because if the acceleration was too great, you'd just step some more in the end. The driver would always move at maximum speed. But the reason I didn't look further into this was that this only works in fullstep mode. Bummer.

stigjoergensen, 2017-04-11:
The problem is most likely thermal protection of the stepper driver...

There is no oddities in the gcode, and after turning down the current of
the steppers there is a lot less "busy" in the output window, which is why
i think its related... but that dont add up with lost steps as it seems the
print completes "fine", no missing steps, no shifted layers - only ugly
blobs where the head is not moving..

and it typically happens on long travel moves.

we first thought it was UART speed related, but the same thing happened
from the SD print...  it was a pain to find it was the thermal protection..
i have thought about getting a video , but it only happens after 3-4 hours
of print, its not something that happens at the same time, or initially -
it takes a good amount of time to show its ugly head.

and yes, he is running marlin bugfix.

JustAnother1, 2017-04-11:
@teemuatlut  I think it would already be an improvement if we could detect that we have lost steps and cancel the print.

I was also thinking about using the stallGuard values to keep the motor in a safe area to avoid loosing steps. so basically dynamically adopting the motor current to the actual load with a safety margin.

An issue I see with DC Step is also that we move up to 4 steppers synchronously. If only one of these steppers has to take some action to adopt to the load it will probably "fall behind" and then compensate. I don't know how many steps this deviation might be but it changes the intended path that the printer should follow.

The sensOStep is interesting. Is is open Hardware? Support for closed loop steppers would be a good feature to have. But as with other features I'm pretty sure that the AVR will not be up for that challenge. But once 32bit-marlin comes along that would be a feature we should have.

teemuatlut, 2017-04-11:
> I was also thinking about using the stallGuard values to keep the motor in a safe area to avoid loosing steps. So basically dynamically adopting the motor current to the actual load with a safety margin. We could then also pause the print (or lower the speed) if the load increases so much that we can not raise the current anymore.
This is basically what [coolStep](https://www.youtube.com/watch?v=Prw7wNa20Gk) does. Except that the purpose is to run cooler and more efficiently. But for joining stallGuard and current control at least.
As far as current control goes, in my recent PR there was automatic current control in relation to driver temperature. Marlin would raise the current until a prewarn was triggered and then lower it until it was cleared.

I realized that too with DC step and I'm not sure if it's doable in any case. The compensation would have to happen at the end of every step ISR.

I doubt sensOstep is open and I think the steppers that Trinamic sells are fairly expensive.
But I'd love to see position feedback in some form. Linear encoders maybe?

JustAnother1, 2017-04-11:
> I doubt sensOstep is open and I think the steppers that Trinamic sells are fairly expensive.
But I'd love to see position feedback in some form. Linear encoders maybe?

I think the encoders themselves are out of the scope of Marlin. It would be nice if there would be open hardware encoders, maybe there are? Implementing an interface for encoders and working with the date would be in the focus of Marlin. Ideally many different encoders would work with marlin. But again too much work for an AVR.

I did some test with Trinamic steppers connected to a Cortex-M CPU. And the real limit regarding speed are the limits of the chip(max Step rate, max SPI clock). So checking after each step would be possible (as long as the chips SPI if fast enough). But checking it once per millisecond is a peace of cake for the cortex-m.

fiveangle, 2017-04-12:
Support for some kind of cheap optical rotary encoder just for E would be a huge boon since we could then outfit super lightweight and torquey geared D.C. motors as direct drive extruders, reducing moving mass to Bowden levels of intertia *dream*

but I digress…

bobc, 2017-04-12:
Obviously Trinamic methods are very specific to their chips. For general
use, Mechaduino and uStepper have closed loop Nema stepper motor solutions
using encoders, but they are not cheap.

I've been looking at using DC motor + encoder based on Misan's work, I
think it looks very promising and the cost would be comparable to current
stepper solutions (including motor, gearbox, encoder and driver) but
requires development work in electronics,mechanical and software areas.

closed 2018-02-06 by thinkyhead

juergmaier, 2018-08-10:
I am a bit late but what about adding an encoder to the X/Y axis motors? For each layer we could position at a known position an if encoder values do not match we could cancel the print?

JustAnother1, 2018-08-10:
@juergmaier That is possible. And you wouldn't even need to cancel the print. Making the step again might yield the wanted result.

As far as I can tell there are two issues with your Idea:
1.)
Encoders cost money.
In the last years the price for printers has gone down. And the users seem the want the cheapest printer, not the best nor the most relieable. So adding encoders brings the price up and therefore doesn't help to get users.

2.)
Encoders would make the stepper control a closed loop system.
Currently the stepper control is a open loop system. Using encoders would make it closed loop. For details see here: https://en.wikipedia.org/wiki/Control_theory
The problem for marlin is that this change would cause the need for a big section of the code to be rewritten. It would need so many changes that thaut would then probably be Marlin 3.0

So if you want to get better printers by using better methods and better components, then better not hold your breath for it. Because that is not the direction that we are currently moving towards.

But that just might be me being frustrated. I was begged to not comment here anymore for that reason. I'm sorry for my violation of that request.

juergmaier, 2018-08-11:
I am printing usually individual parts and some parts just need special
parameters to get successfully printed. So its not that I would expect a
step to be retried as I usually lose steps when the extruder runs into
raised up material. Continuing with printing does not make sense in these
cases. It would just be nice to have the print aborted whenever I lose
steps. Encoders can be found for around 10 Euros - not that expensive?