MarlinFirmware/Marlin #6316: Possible some bug with marlin

batata004, 2017-04-12:
Hi,

I executed for the second time the command (yes, I want just one cycle)

M303 E0 S222 U C1

And it displays always KP KD KI at 0. Is this right? The return is below:

READ:  bias: 86 d: 86 min: 218.33 max: 228.13PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
READ: #define  DEFAULT_Kp 0.00
READ: #define  DEFAULT_Ki 0.00
READ: #define  DEFAULT_Kd 0.00

When I execute the command I see the temperature arises and them it passes a little the right point and after that pronterface displays the above information.

Kaibob2, 2017-04-12:
What is the `U` parameter for? Normally the call for PID autotune is M303 E0 S200 C8, or in your case M303 E0 S222 C1
By the way, one cycle doesn't make much sense as it has no chance to cool down a little and heat up again. I would suggest to run at least 5 cycles: M303 E0 S222 C5

teemuatlut, 2017-04-12:
U applies the resulting settings after the autotune. [source](http://marlinfw.org/docs/gcode/M303.html)

Kaibob2, 2017-04-12:
Cool. Didn't know that.

batata004, 2017-04-12:
Thanks, I will use more than 1 U!

closed 2017-04-12 by batata004