MarlinFirmware/Marlin #6317: RC8 Bugfix and delta print issue

wookie666, 2017-04-12:
Trying to print the attached Zip file.

The parts do fit on the bed, and nothing is outside the max radius of the parts. Yet my delta will do the beginning wipe that i have coded, Then start the print, It does approx one line of the print and before it can change direction, it starts to slowly lift z until it hits the max endstops, but keeps going. 

I cant see anything in the code that would cause the issue, so im putting it out here to test.

I have tested other files before and after this one, and they work fine. So it makes me think something is wrong with this file.

any idea?
[Marlin-parts issue.zip](https://github.com/MarlinFirmware/Marlin/files/915139/Marlin-parts.issue.zip)

Roxy-3D, 2017-04-12:
So...  I don't know...  But when a Delta start's raising the carriages until they hit the Max endstops...  It is usually because it is told to go off the bed.    Try scaling the part to 50% and see if that works ok.   Because if it does...   My bet is you are going off the bed.

wookie666, 2017-04-12:
I'll have to check. That would be my next guess.

I'll reduce to 50 percent then go from there until it hits

wookie666, 2017-04-12:
it has to be a bug or an icorrect setting somewhere. because it fits at 80 percent and works fine, but at 100, it should still fit. i have tested everything and cant figure out why it doesn't fit.

wookie666, 2017-04-12:
I think it might have something to do with the changes made to autoleveling recently. Because the Probing options use position_is_reachable and so do the movement options when printing to check if the location is reachabel. could me having   #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 20)
  #define LEFT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
  #define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
  #define FRONT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
  #define BACK_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS

have something to do with it? As that probable radius reduces my entire print area by 20mm for autolevelling, but it shouldn't do anything for printing. if i have it smaller than 20, then i get the autoleveling bilinear probe points try to probe outside the bed circle

I also have this as my travel limits.
.

wookie666, 2017-04-12:
And this is set as

```
 // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
  #define DELTA_PRINTABLE_RADIUS 116.0
```

wookie666, 2017-04-12:
So i can confirm that it doesn't have an issue if i dont use auto leveling, if i leave it off, then it runs the code no problem. As soon as i turn it on, thats when i have the problem. can someone tell me why?
My probe points are a 9 x 9 grid. With the printable radius being 116 and the grid being 20mm smaller than that at 96, only because otherwise it will try and probe just outside the bed on some points.

oldmcg, 2017-04-12:
I also had problems with bilinear going outside the printable_radius during probing.  The current leveling code that limits probeable_radius to 20 less than printable_radius is kind of presumptive.  A square probeable grid that will completely fit within the circular printable_radius has sides of .7071 (sqrt(2)) times the radius (not 20 less than the radius), so for 116 radius that would only be 67.88mm probeable square grid.  I'm still looking through the code, but trying to extend UBL and bilinear to fully probe the circular bed without going outside it.

wookie666, 2017-04-13:
>  I also had problems with bilinear going outside the printable_radius during probing. The current leveling code that limits probeable_radius to 20 less than printable_radius is kind of presumptive. A square probeable grid that will completely fit within the circular printable_radius has sides of .7071 (sqrt(2)) times the diameter (not 20 less than the radius), so for 116 radius (232 diameter) that would only be 135mm probeable square grid. I'm still looking through the code, but trying to extend UBL and bilinear to fully probe the circular bed without going outside it.

The included Bilinear does prevent the probes outside the bed area from actually probing by completely skipping them, But i did find that sometimes it would still hit when i had it at printable radius -5.

It just needs to be fixed so that it can still print outside the autolevel grid. I think the parts that are not hit shouldn't be skipped, they should be automatically filled in with the nearest probe point value.

closed 2017-10-31 by thinkyhead