MarlinFirmware/Marlin #6321: [DUAL_X_CARRIAGE COREXY combination]

evanvruggink, 2017-04-12:
Hello,

Once i found out about the dual x carriage support and saw how the BCN3D Sigma used it in their printer i started designing a new 3D printer concept. For me the biggest flaw of the BCN3D Sigma is the 2 motors on the y-axis driving the extruders. This adds a high amount of weight which really decreases the dynamic properties of the system and therefore the print quality. That is why i came up with a concept which kinematics looks like a COREXY, but then with 2 Independent moving extruders. 

What happens is that 2 stepper motors drive the 2 extruders carriages independently with belts running through the moving y-axis, but the steppers are fixed to the frame. However, when the Y-axis moves and the x-stepper motors stand still, the 2 extruders will move to the right for y-axis moving up and to the left for y-axis moving down.

Is it possible to have two independent moving extruder x-carriages (the same as already implemented) and having y-axis movement resulting in the same negative x-axis motor carriage movement, so the carriages stay in the same place with y-axis movement.

I saw some possibilities in the code to adjust the COREXY to only oppose movement in the y-direction, but it is described that DUAL_X_CARRIAGE and COREXY cannot be used together. Also i'm not an expert on coding which does not help.

I hope the concept and feature request is clear, i can add some pictures for clarity if needed.

With kinds regards

thinkyhead, 2018-02-06:
Interesting concept. Sorry this issue got buried. Any progress in experimenting or designing?

thinkyhead, 2018-11-05:
You understand correctly. No one has yet submitted CoreXYU support to the project.

boelle, 2019-03-12:
@thinkyhead i think this one should also have the feature request label

boelle, 2019-06-22:
wrong label on this one

InsanityAutomation, 2019-06-22:
Looks interesting. Id do it if I could get one in my hands...

boelle, 2019-06-22:
@evanvruggink what @InsanityAutomation said  ^^^

Zuru1, 2020-09-03:
If one wants to give a go at this feature implementation, is it a correct assumption that one can use this  PR #19235 as a starting stone?