MarlinFirmware/Marlin #6322: 12 axis simultaneous control/interpolation (xyz,abc,ijk,uvw)

benw-ideas, 2017-04-12:
I'm working on a project that needs 12 axes of simultaneous/interpolated motion.  I looked at off-the-shelf commercial multi-axis motion control systems from ABB, Delta-Tau, Gail, Aerotech, etc, and those all cost a bundle!  I saw @thinkyhead added 5-extruder capability for a recent medical/syringe type application for an individual.  Is it possible to add a bunch of other axes capable of full interpolated/simultaneous motion to Marlin? 

I'd love to be able to write Gcode commands like: G1 X0.023 Y2.223 Z4.332 A3.225 B3.676 C6.349 I9.492 J8.504 K8.345 U6.337 V9.343 W5.296 F600.00 (Where F is feedrate of course)  and send them to the Arduino using a g-code sender (maybe cut and past my gcode into the edit window of Repetier and send it that way)

I also have no problems using jumpers directly from the Arduino to stepper drivers on a breadboard, just as long as I know the pins for each axis's step/direction signals.  (Because there isn't an off-the-shelf arduino shield that drives 12 steppers.)
 
I'm really good with  #writing G-code and all of the inverse-kinematics for my application... but I know nothing about programming in general or the workings of Marlin.  Any help would be super-appreciated!

Thanks in advance!

Bob-the-Kuhn, 2017-04-12:
My first reaction was "There's someone out there even crazier than me!" but then I remembered the mixing extruder function/feature.

With this enabled all the extruders move proportionally  I don't think you can have one moving in reverse from the others - at least in the firmware.  You could wire up the extruder steppers so that one always was reversed from the others.

Here's the info I could find on the mixing function.  See if it will do what you want.
```
/**
 * "Mixing Extruder"
 *   - Adds a new code, M165, to set the current mix factors.
 *   - Extends the stepping routines to move multiple steppers in proportion to the mix.
 *   - Optional support for Repetier Host M163, M164, and virtual extruder.
 *   - This implementation supports only a single extruder.
 *   - Enable DIRECT_MIXING_IN_G1 for Pia Taubert's reference implementation
 */
//#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
#endif

  /**
   * M163: Set a single mix factor for a mixing extruder
   *       This is called "weight" by some systems.
   *
   *   S[index]   The channel index to set
   *   P[float]   The mix value
   *
   */
    /**
     * M164: Store the current mix factors as a virtual tool.
     *
     *   S[index]   The virtual tool to store
     *
     *
    /**
     * M165: Set multiple mix factors for a mixing extruder.
     *       Factors that are left out will be set to 0.
     *       All factors together must add up to 1.0.
     *
     *   A[factor] Mix factor for extruder stepper 1
     *   B[factor] Mix factor for extruder stepper 2
     *   C[factor] Mix factor for extruder stepper 3
     *   D[factor] Mix factor for extruder stepper 4
     *   H[factor] Mix factor for extruder stepper 5
     *   I[factor] Mix factor for extruder stepper 6
     *
     */
```

We're in the process of doing a major release so it may take a while before someone responds on the feasibility of adding more extruders.

----

If the M163, M164 & M165 method of specifying extruder stepper movements doesn't work for you then we probably have some big problems.  Both the host software and Marlin are expecting to see a certain set of arguments with the G1 command.  Going to 12 axis would exhaust the list of acceptable arguments.

True 12 axis control would require a lot of work to implement.  Sure hope the M163, M164 & M165 method is acceptable.

benw-ideas, 2017-04-12:
My project is two 6-axis systems moving at the same time, but in different paths, and their motions need to be coordinated with respect to each other (but their paths are not the same as each other).  

Is there a way to add more arguments to the list of what the G1 command can accept?

thinkyhead, 2017-04-12:
You can add to the `gcode_get_destination` function to interpret more axes for movement commands. It's also used for `G2`, `G3`, `G5`, and `G38`.

Now, as to the actual handling of an arbitrary number of steppers, all acting simultaneously, would require a huge amount of changes. Even adding support for one additional extruder, as done recently, required changes across several files.

The other problem you'll quickly hit is that 8-bit AVR processors running at 16 or 20MHz just can't manage that number of simultaneous axes at once at any reasonable speed. You will need to use a 32-bit STM or ARM-based board, or you will need multiple 8-bit boards.

bobc, 2017-04-13:
> My project is two 6-axis systems moving at the same time, but in
different paths, and their motions need to be coordinated with respect to
each other (but their paths are not the same as each other).

If you have different paths then you need to think about how acceleration
limits etc are calculated. I also think a 32 bit platform would be a better
bet. I can't think of any printer firmwares that support 12 axes, they are
pretty much optimised to 3 + extruder. Smoothieware supports 6 I think.
That might be a better place to start.

​I did some work to support variable number of axes for firmware running on
R2C2 board (LPC1759), I haven't touched that code for about 4 years though.
I did foolishly once offer to write 6-axis firmware if anyone built a
6-axis machine, but no one ever took up the offer :)

closed 2017-11-01 by thinkyhead

anthonychukh, 2019-08-12:
I'm working on a similar project with the azteeg pro controller with 7 axis. I'm replicating the XYZ function to accommodate ABC but it'd be nice if each axis could be parametrically added. adding more variables mean It was running low on memory. I'm curious to know what approach you ended up achieving the 12 axis result