MarlinFirmware/Marlin #6323: [Question] Send M81 at end of file

Kaibob2, 2017-04-12:
Is there a way to implement an option to select "kill after file from SD is finished" via LCD. 
Background is, that I quite often start a long print which ends somewhen at night. Most of the time I forget to uncomment the M81 in my end script in S3D. When the print is running for 3 hours and has 6 more to go it would be nice to have an option to kill when finished.

Bob-the-Kuhn, 2017-04-12:
Most slicers allow you to customize the end of print commands.  Can you make the change you want in S3D?

Kaibob2, 2017-04-12:
@Bob-the-Kuhn Ehh, like i said 

> Most of the time I forget to uncomment the M81 in my end script in S3D.

If so everything is fine, but i have the M81 always command commented (disabled) in my end script because for shorter prints it's not necessary for the printer to kill itself. 
The option i'm talking about is for the case, that a user (me) decides that the printer shall kill itself after finishing while the printjob is already running. Modifying the end script wouldn't helb much at this time.

Bob-the-Kuhn, 2017-04-12:
What do you mean by "kill itself"?  Are you wanting to have it turn off the power supply?

The short answer is there would have to be custom code added to the ultralcd.cpp file to add another menu item.  The BLTOUCH self test basically sends a single command.  You could model the new code after that one.  Do a search for M280 and that'll pop up examples.

As best I can tell you should be able to send the M81 command at anytime after the print starts.  It'll wait until all the movements are finished before executing.  Hmmm ...  I don't know if switching extruders within the print would result in the print being killed at that point.

thinkyhead, 2017-04-12:
Sounds like the kind of thing that would be added to a "print jobs" feature, but Marlin is weak on the concept of a "print job" outside of SD printing. There is a print timer which, for host printing, takes cues from the heating commands to decide when to start and stop. So, any "do this at the end of the print job" feature should augment the print job timer.

Kaibob2, 2017-04-13:
I did it myself and, i'm a little surprised about myself, it works perfect. Can you @Bob-the-Kuhn @thinkyhead please review what a did there #6328

Kaibob2, 2017-07-26:
I guess this won't be merged anymore. I'll close here

closed 2017-07-26 by Kaibob2